espressif/esp-idf #3507: 蓝牙Mesh如何兼容其他厂家的蓝牙Mesh模块？

xiaodingding, 2019-05-21:
如题，我们想要了解，如何将蓝牙Mesh模块兼容其他厂家的蓝牙Mesh模块。
 我们的测试方法如下：
1.使用的芯片模组为：ESP32-WROOM-32模块，NRF52832模块。
2.一块ESP32-WROOM-32芯片充当provisioner角色，一块ESP32-WROOM-32芯片充当node角色。
3.一块NRF52832芯片充当node角色。
4.给provisioner角色和Node角色的ESP32模块上电，模块能快速的组网和进行通信。
5.给provisioner角色的ESP32和Node角色的NRF52832上电，provisioner不能搜索到Nrf52832，也不能正常组网。
6.关闭provisioner，重启所有Node，打开手机APP NRF Mesh软件，能正常搜索带充当Node角色的NRF52832，和充当Node节点的ESP32。使用nRF Mesh软件连接和配置ESP32和Nrf52832皆能正常连接和配置。
7.esp-idf版本：esp-ble-mesh-v0.6.1。

WCCWCC, 2019-05-23:
Hi xiaodingding,
  你可以把 provisioner 中设置 uuid match 的代码注释掉试一下 `esp_ble_mesh_provisioner_set_dev_uuid_match`。
  我们提供的 ble_mesh_provisioner 的 demo 会匹配未配网设备的 UUID 的值。如果UUID匹配不成功会不会进行设备配网的。

zhanlej, 2019-06-04:
@WCCWCC 
Hi WCCWCC,
我想实现只允许特定的provisioner才可以将设备加入到ble mesh网络中的功能，比如说我们开发的ble mesh设备只能通过我们开发的APP进行配置，其他APP无法配置。
据我了解``static oob``应该可以实现这个功能，比如设备和provisioner必须设置相同的static oob，这样就可以实现双向认证了。目前这也仅仅是我的猜测，不知道是否正确。
因此我在``ble_mesh_node``这个demo中做了如下修改：
```cpp
/* Disable OOB security for SILabs Android app */
static esp_ble_mesh_prov_t provision = {
    .uuid = dev_uuid,
#if 0
    .output_size = 4,
    .output_actions = ESP_BLE_MESH_DISPLAY_NUMBER,
    .input_actions = ESP_BLE_MESH_PUSH,
    .input_size = 4,
#else
    // .output_size = 0,
    // .output_actions = 0,
#endif
    // .oob_pub_key = 1,
    .static_val = {0x01,0x02,0x03,0x04},
    .static_val_len = 4,
};
```
编译并烧写后发现使用nRF mesh依旧可以直接将设备配置到Mesh网络中，使用``ble_mesh_provisioner`` demo配置设备现象也一样，感觉static oob好像根本没用，请问这种情况是我操作的问题还是你们的SDK中还没有实现static oob功能？

WCCWCC, 2019-06-21:
Hi zhanlej ,
     
    1. 如果你设置了 static oob 的值，表示 node 端的 oob 的值是可用的。
    2. provisioner 如果使用的是 static oob 的验证方式，它将检查入网设备的oob的值，如果值不正确是不会进行配网的。
    3. provisioner 如果使用的是 no oob 的验证方式，它将不会检测入网的设备的oob的值，而是直接配置设备入网。

Alvin1Zhang, 2019-07-17:
@xiaodingding Thanks for reporting the issue. Feel free to reopen the issue. Thanks.

closed 2019-07-17 by Alvin1Zhang