espressif/esp-idf #3508: esp_http_client fails to get content

tonyp7, 2019-05-21:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version v3
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v3.2
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:  Windows
- Power Supply: USB

## Problem Description
esp_http_client does not get body of the response on a given site.

Running the example under protocols/esp_http_client/esp_http_client_example.c; I simply changed the URL under void http_rest() from http://httpbin.org/get to http://api.mclk.org/time.

### Expected Behavior

200 response with content

### Actual Behavior

200 response with content length=-1

### Code to reproduce this issue

under protocols/esp_http_client/esp_http_client_example.c, replace in void http_rest()
```C
esp_http_client_config_t config = {
        //.url = "http://httpbin.org/get",
	.url = "http://api.mclk.org/time",
```

## Debug Logs

```
I (3210) HTTP_CLIENT: Connected to AP, begin http example
I (3730) HTTP_CLIENT: HTTP GET Status = 200, content_length = -1
```

## Other info:

-  **basic protocols/http_request example reads https://api.mclk.org/time correctly!**
- POSTMAN (or any other webservice testing tool) reads the content correctly

Seems like the esp_http_client doesn't like something; but I don't know what.

tonyp7, 2019-05-21:
Hello,

I found the issue. If the server doesn't respond with a Content-Length header the client fails to extract properly the body.

I have modified the api to force the output of this header and now the client works properly.

I believe this should not be the behavior. Behaviour should be: in case of no content-length header try to extract the body of the response anyway -- like a typical client/browser does.

Cheers

jitin17, 2019-05-22:
@tonyp7 In `protocols/esp_http_client/esp_http_client_example.c` you will find a HTTP event handling function `_http_event_handler`. Under `_http_event_handler` in case `HTTP_EVENT_ON_DATA`, you should get body data even if content-length is -1.

 I suggest having a print statement under `HTTP_EVENT_ON_DATA` to verify whether you receive body data or not.

tonyp7, 2019-05-23:
@jitin17 you are correct. 

I am not sure if content length should show as -1 still; or maybe the function should be renamed from esp_http_client_get_content_length to esp_http_client_get_content_length_**header**.

On that same vein esp_http_client_set_post_field should really be esp_http_client_set_body.

closed 2019-05-23 by tonyp7