espressif/esp-idf #3509: [RMT]some remote controllers not working

keesus, 2019-05-21:
## Environment

- Development Kit:  ESP32-Wrover-Kit
- Module or chip used: ESP32-WROVER-B
- Build System: CMake

i'm using [ESP32_IR_Remote.c](https://github.com/Darryl-Scott/ESP32-RMT-Library-IR-code-RAW).

I can copy IR raw datas from some remote controllers and save it and transmit it as well.

But for some remote controllers it reads raw data fine but when transmit it doesn't perform.

I found some points to make an adjustment here, 

  rmt_config_t config;
  config.rmt_mode = RMT_MODE_RX;
  config.channel = (rmt_channel_t)rmtport;
  config.gpio_num = (gpio_num_t)gpionum;
  gpio_pullup_en((gpio_num_t)gpionum);
  config.mem_block_num = 7; //how many memory blocks 64 x N (0-7)
  config.rx_config.filter_en = 1;
  config.rx_config.filter_ticks_thresh = 100; // 80000000/100 -> 800000 / 100 = 8000  = 125us
  config.rx_config.idle_threshold = TICK_10_US * 100 * 20; // originally TICK_10_US * 100 * 20
  config.clk_div = CLK_DIV;
  ESP_ERROR_CHECK(rmt_config(&config));
  ESP_ERROR_CHECK(rmt_driver_install(config.channel, 5000, 0));
  rmt_get_ringbuf_handle(config.channel, &ringBuf);
  rmt_rx_start(config.channel, 1);

and I tried some values but nothing changed.

Can anybody help me out ?

negativekelvin, 2019-05-21:
You should move this issue to the third party code repo or post in the forum.  I don't think this is directly related to esp-idf code.

It can depend on the protocol the remote uses https://globalcache.zendesk.com/hc/en-us/articles/114094165172-Tech-Tip-IR-Control-with-RC6-coded-IR-signals or the carrier frequency expected by the receiving device.

koobest, 2019-05-21:
Hi, @keesus
The information you provided can not help us debug this issue. Can you provide RMT signal waveforms?

thanks !!

keesus, 2019-05-29:
@koobest Hi, I copied by receiver and the data would be [8850,4650,400,750,400,750,400,1850,400,750,400,750,400,700,400,750,400,700,400,1850,400,1850,400,750,400,1850,400,1850,400,1850,400,1850,400,1850,400,750,400,750,400,1850,400,750,400,750,400,750,400,1850,400,1850,400,1850,400,1850,400,750,400,1850,400,1850,400,1850,400,750,400,750,400]

after I transmit the data to some air conditioner, nothing happens.
what makes me confused is it's working well to other air conditioner, TV , other devices.
I researched many IR related documents but still have no idea.

negativekelvin, 2019-05-29:
It does look like basic nec code.  What is make and model of a/c?

keesus, 2019-05-29:
@negativekelvin 

Hi there.

it says TCL air conditiner.

koobest, 2019-07-19:
Hi, @keesus
Have you solved this issue? 

> after I transmit the data to some air conditioner, nothing happens.what makes me confused is it's working well to other air conditioner, TV , other devices.I researched many IR related documents but still have no idea

You mean when you send this sequence with ESP32, it is works to some devices, but not for other devices?

Is this problem caused by the incorrect carrier frequency? Or the accepted data is wrong, so the data sent is wrong?

thanks !!

Alvin1Zhang, 2019-08-12:
@keesus Thanks for reporting the issue, will close this issue due to short of feedback, please feel free to reopen the issue if it still exists. Thanks.

closed 2019-08-12 by Alvin1Zhang