espressif/esp-idf #3510: APSTA + HTTP + IP_FORWARD does not work as intended

MikeMazowski, 2019-05-21:
## Environment

- Development Kit:
2 x ESP32-DevKitC v4 + 1 x LOLIN D32
2 x ESP32-WROOM-32D + 1 x ESP32-WROOM-32
- IDF version v4.0-dev-225-g2f8b6cfc7
- Build System:             Make
- Compiler version       (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- Power Supply:            USB

## Problem Description

I have 3 devices in APSTA mode with static IP configuration, each connected to the next device (Device1 connected to Device2, Device2 connected to Device3).

Setting "#define IP_FORWARD 1" inside "\esp-idf\components\lwip\lwip\src\include\lwip\opt.h" shall enable device 1 to send HTTP GET request directly to device 3 by its IP, but upon sending a request from device 1 to device 3, it fails.

Debugging forwarding function "ip4_route_src_hook()" in ip4.c, I am able to see that in direction device1->device3, device 3 receives the correct source and destination IP addresses, yet still HTTP GET request fails. In direction device3->device1, device 1 does not see any signs of retransmission being attempted.

Static IP configuration:
Device, Gateway IP,         IP      , NETMASK
    1     ,192.168.4.1, 192.168.4.1, 255.255.255.0
    2     ,192.168.5.1, 192.168.5.1, 255.255.255.0
    3     ,192.168.6.1, 192.168.6.1, 255.255.255.0

### Expected Behavior

I: HTTP GET Status = 200, content_length = 96

### Actual Behavior

E: Connection failed, sock < 0

### Steps to repropduce

1. Start all devices in APSTA mode.
2. Start HTTP server and client on each device.
3. Connect device 2 to device 3 and device 1 to device 2.
4. Send HTTP GET request from device 1 to device 3's IP address.

negativekelvin, 2019-05-21:
Probably this is what happens:
When 4.1 sends to 6.1, address will match netmask for both interfaces but STA is default interface so packet will get to 6.1.  When 6.1 sends to 4.1, address will match netmask for both interfaces but AP is not default interface so packet will not be forwarded.  If 6.1 STA interface is down, packet may get to 5.1 but it would choose STA route again and be dropped.

negativekelvin, 2019-05-21:
STA is default IF so packet will get forwarded by 5.1 to 6.1 but not opposite.

MikeMazowski, 2019-05-21:
> STA is default IF so packet will get forwarded by 5.1 to 6.1 but not opposite.

Thank You for the reply.

Does that mean that in the case of an HTTP request from device 1 to device 3, it can send the request, but the response cannot be forwarded backwards?

negativekelvin, 2019-05-21:
Yes.  You can try choosing other netmasks and ips that might work.

MikeMazowski, 2019-05-21:
> Yes. You can try choosing other netmasks and ips that might work.

I have tried the following configuration, but it did not work.

Static IP configuration:
Device, Gateway IP, IP , NETMASK
1 ,192.168.5.1, 192.168.4.1, 255.255.255.0
2 ,192.168.7.1, 192.168.6.1, 255.255.255.0
3 ,192.168.9.1, 192.168.8.1, 255.255.255.0

From device1 to device2:
Destination 192.168.7.1 is failing, however destination 192.168.6.1 works fine.
From device1 to device3 all addresses fail, no retransmission evidence on device 3.

I presume that STA and AP need to be in different subnets, however I tried configuring AP and STA IP addresses separately, but it causes a crash in APSTA mode, when configuring STA.

negativekelvin, 2019-05-21:
D1 STA 192.168.4.2
D2 AP 192.168.4.1
D2 STA 192.168.5.2
D3 AP 192.168.5.1
assuming D3 AP is the default NETIF, it should work in this limited case
You should be able to set separate static ips

MikeMazowski, 2019-05-22:
> D1 STA 192.168.4.2
> D2 AP 192.168.4.1
> D2 STA 192.168.5.2
> D3 AP 192.168.5.1
> assuming D3 AP is the default NETIF, it should work in this limited case
> You should be able to set separate static ips

This is what I initially tried where D1 STA was assigned 192.168.5.2 via DHCP and D2 STA 192.168.6.2, but unfortunately it only achieved this "unidirectional" forwarding.

negativekelvin, 2019-05-22:
Was 192.168.4.2 set as 192.168.4.1's gateway etc.?

MikeMazowski, 2019-05-22:
> Was 192.168.4.2 set as 192.168.4.1's gateway etc.?

Device with AP gateway and IP address being 192.168.4.1 assigned IP 192.168.4.2 to the STA that connected to it.

negativekelvin, 2019-05-22:
Yes I see.  Not sure if they have to be each other's gateways though.  You may want to enable lwip IP_DEBUG

MikeMazowski, 2019-05-22:
Device 1:

ip4_output_if: st1
IP header:
+-------------------------------+
| 4 | 5 |  0x00 |        44     | (v, hl, tos, len)
+-------------------------------+
|       42      |000|       0   | (id, flags, offset)
+-------------------------------+
|  255  |    6  |    0x9481     | (ttl, proto, chksum)
+-------------------------------+
|   15  |   15  |    4  |    2  | (src)
+-------------------------------+
|   15  |   15  |    5  |    1  | (dest)
+-------------------------------+
ip4_output_if: call netif->output()

  Device 2:

ip_input: iphdr->dest 0x1050f0f netif->ip_addr 0x1040f0f (0x50f0f, 0x40f0f, 0x10
00000)
ip_input: iphdr->dest 0x1050f0f netif->ip_addr 0x2050f0f (0x50f0f, 0x50f0f, 0x10
00000)
ip_input: iphdr->dest 0x1050f0f netif->ip_addr 0x100007f (0xf, 0x7f, 0x1050f00)
ip4_input: packet not for us.
ip4_forward: forwarding packet to 15.15.5.1

  Device 3:

ip4_output_if: ap2
IP header:
+-------------------------------+
| 4 | 5 |  0x00 |        44     | (v, hl, tos, len)
+-------------------------------+
|        4      |000|       0   | (id, flags, offset)
+-------------------------------+
|  255  |    6  |    0x94a7     | (ttl, proto, chksum)
+-------------------------------+
|   15  |   15  |    5  |    1  | (src)
+-------------------------------+
|   15  |   15  |    4  |    2  | (dest)
+-------------------------------+
ip4_output_if: call netif->output()

negativekelvin, 2019-05-22:
So 5.1 tries to send to 4.2 by sending on AP interface but  device 2 doesn't see the packet, correct?  This  may be because of filtering/routing in wifi ap driver, unicast packets only sent to matching STA.

MikeMazowski, 2019-05-23:
> So 5.1 tries to send to 4.2 by sending on AP interface but device 2 doesn't see the packet, correct? I would say try it again with static IP and make the AP's gateway set as its STA.

Indeed, in backward direction device 2 doesn't see the packet. 
> make the AP's gateway set as its STA.
In such case it also fails forward direction.

I composed very simple code combining APSTA + HTTP server/client, perhaps it will be easier to get to the core of problem this way.

This code is ready to run with IDF, only thing probably needed is to set IP_FORWARD to 1 inside opt.h in lwip component.

To use examples, device 3 should be powered up first, then device 2, then device 1, so that they connect to one another during startup.

freakyxue, 2020-01-17:
@MikeMazowski I will debug this problem soon

freakyxue, 2020-06-10:
hi @MikeMazowski A - > b - > C, a creates TCP, C node listens, a sends syn, B successfully forwards the syn packet, and B correctly processes the syn packet, but in the ARP layer, C node cannot find the appropriate MAC address, and sends the packet to B. Because the application layer address is a, C does not know that B will forward the syn packet to him.

freakyxue, 2020-06-10:
According to LwIP code, TCP syn forwarding is not supported

freakyxue, 2021-01-06:
hi @MikeMazowski 
If the C node is in AP mode, it can handle the TCP syn packet correctly when it receives it. However, when sending the packet, because the destination address is not in the same network segment, the C node will send the packet to the default gateway, but the gateway is itself, so the packet will fail to forward

If the forwarding is successful, the C node should be set to sta mode

closed 2021-01-06 by Alvin1Zhang