espressif/esp-idf #3511: How to create the communication between stm32 and esp32 via IIC

Herwey, 2019-05-21:
I sent 8 bit data (0X56) from stm32, and I used "esp-idf/examples/peripherals/i2c/i2c_self_test" this demo, I had set the slave address " 0X56".  And connected the SCL and SDA between ESP32 and STM32. But the STM32 couldn't receive the ACK singal from ESP32, could you figure it out ?
Thanks in advance :D

koobest, 2019-05-21:
Hi, @Herwey 
please make sure that  the hardware connection is correct(GND). Are both STM32 and ESP32 powered by 3.3V?

thanks !!

Alvin1Zhang, 2019-06-11:
@Herwey Thanks for reporting the issue. Could you help share if any updates w.r.t. the questions by @koobest ? Thanks.

Alvin1Zhang, 2019-06-21:
@Herwey Thanks for reporting the issue. If you have further such questions, please help post on our [forum](https://esp32.com), thanks.

closed 2019-06-21 by Alvin1Zhang