espressif/esp-idf #3512: Error -76 from MbedTLS / errno 11 on network timeout

bearsh, 2019-05-21:
## Environment

- Development Kit: none
- Module or chip used:  NINA-W1
- IDF version: v3.3-beta3-34-g8c57aa024
- Build System: Make
- Compiler version:1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: external 3.3V and Battery

## Problem Description
I created a tls connection with `timeout_ms` set to 1000ms. I have a transmitter and a receiver task. If not data is received `esp_tls_conn_read()` returns -76 (`MBEDTLS_ERR_NET_RECV_FAILED`), errno is set to 11 after 1000ms. the connection keeps working fine.

### Expected Behavior
either a different return value (something like `_TIMEOUT` or `_EAGAIN`) or at least esp-tls should not log it as error (as it it kind of expected)

### Actual Behavior
the log suggests an error happened but it is rather expected behavior.

### Steps to repropduce
1. create new connection with `timeout_ms` set to something != 0
2. read connection e.g. in a loop, the error happens every `timeout_ms` if nothing gets received.

see also #1630 #434

JakubRakus, 2019-08-01:
Any progress in resolving this issue?

I'm trying to implement https ota using AWS S3 bucket but `esp_https_ota_begin()` always throws this error.

    I (24856) mbedtls: ssl_tls.c:8084 => handshake
    I (24856) mbedtls: ssl_cli.c:3510 client state: 0
    I (24856) mbedtls: ssl_tls.c:2755 => flush output
    I (24866) mbedtls: ssl_tls.c:2767 <= flush output
    I (24866) mbedtls: ssl_cli.c:3510 client state: 1
    I (24886) mbedtls: ssl_tls.c:2755 => flush output
    I (24886) mbedtls: ssl_tls.c:2767 <= flush output
    I (24886) mbedtls: ssl_cli.c:774 => write client hello
    I (24906) mbedtls: ssl_tls.c:3184 => write handshake message
    I (24906) mbedtls: ssl_tls.c:3343 => write record
    I (24916) mbedtls: ssl_tls.c:2755 => flush output
    I (24916) mbedtls: ssl_tls.c:2774 message length: 258, out_left: 258
    I (24926) mbedtls: ssl_tls.c:2779 ssl->f_send() returned 258 (-0xfffffefe)
    I (24926) mbedtls: ssl_tls.c:2807 <= flush output
    I (24936) mbedtls: ssl_tls.c:3476 <= write record
    I (24946) mbedtls: ssl_tls.c:3320 <= write handshake message
    I (24946) mbedtls: ssl_cli.c:1106 <= write client hello
    I (24956) mbedtls: ssl_cli.c:3510 client state: 2
    I (24966) mbedtls: ssl_tls.c:2755 => flush output
    I (24966) mbedtls: ssl_tls.c:2767 <= flush output
    I (24966) mbedtls: ssl_cli.c:1499 => parse server hello
    I (24976) mbedtls: ssl_tls.c:4311 => read record
    I (24996) mbedtls: ssl_tls.c:2536 => fetch input
    I (24996) mbedtls: ssl_tls.c:2697 in_left: 0, nb_want: 5
    I (25016) mbedtls: ssl_tls.c:2721 in_left: 0, nb_want: 5
    I (25016) mbedtls: ssl_tls.c:2722 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
    I (25016) mbedtls: ssl_tls.c:2742 <= fetch input
    I (25026) mbedtls: ssl_tls.c:2536 => fetch input
    I (25036) mbedtls: ssl_tls.c:2697 in_left: 5, nb_want: 92
    I (25056) mbedtls: ssl_tls.c:2721 in_left: 5, nb_want: 92
    I (25056) mbedtls: ssl_tls.c:2722 ssl->f_recv(_timeout)() returned 87 (-0xffffffa9)
    I (25056) mbedtls: ssl_tls.c:2742 <= fetch input
    I (25076) mbedtls: ssl_tls.c:4385 <= read record
    I (25076) mbedtls: ssl_cli.c:1789 server hello, total extension length: 11
    I (25076) mbedtls: ssl_cli.c:1978 <= parse server hello
    I (25096) mbedtls: ssl_cli.c:3510 client state: 3
    I (25096) mbedtls: ssl_tls.c:2755 => flush output
    I (25106) mbedtls: ssl_tls.c:2767 <= flush output
    I (25106) mbedtls: ssl_tls.c:5655 => parse certificate
    I (25106) mbedtls: ssl_tls.c:4311 => read record
    I (25126) mbedtls: ssl_tls.c:2536 => fetch input
    I (25126) mbedtls: ssl_tls.c:2697 in_left: 0, nb_want: 5
    I (25166) mbedtls: ssl_tls.c:2721 in_left: 0, nb_want: 5
    I (25166) mbedtls: ssl_tls.c:2722 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
    I (25166) mbedtls: ssl_tls.c:2742 <= fetch input
    I (25176) mbedtls: ssl_tls.c:2536 => fetch input
    I (25186) mbedtls: ssl_tls.c:2697 in_left: 5, nb_want: 3175
    I (25206) mbedtls: ssl_tls.c:2721 in_left: 5, nb_want: 3175
    I (25206) mbedtls: ssl_tls.c:2722 ssl->f_recv(_timeout)() returned 3170 (-0xfffff39e)
    I (25216) mbedtls: ssl_tls.c:2742 <= fetch input
    I (25256) mbedtls: ssl_tls.c:4385 <= read record
    I (25446) mbedtls: ssl_tls.c:5863 <= parse certificate
    I (25446) mbedtls: ssl_cli.c:3510 client state: 4
    I (25446) mbedtls: ssl_tls.c:2755 => flush output
    I (25456) mbedtls: ssl_tls.c:2767 <= flush output
    I (25466) mbedtls: ssl_cli.c:2336 => parse server key exchange
    I (25466) mbedtls: ssl_tls.c:4311 => read record
    I (25466) mbedtls: ssl_tls.c:2536 => fetch input
    I (25486) mbedtls: ssl_tls.c:2697 in_left: 0, nb_want: 5
    I (30486) mbedtls: ssl_tls.c:2722 ssl->f_recv(_timeout)() returned -76 (-0x004c)
    W (30486) mbedtls: ssl_tls.c:4973 mbedtls_ssl_fetch_input() returned -76 (-0x004c)
    W (30496) mbedtls: ssl_tls.c:4344 ssl_get_next_record() returned -76 (-0x004c)
    W (30516) mbedtls: ssl_cli.c:2381 mbedtls_ssl_read_record() returned -76 (-0x004c)
    I (30516) mbedtls: ssl_tls.c:8094 <= handshake
    E (30516) esp-tls: mbedtls_ssl_handshake returned -0x4c
    I (30526) esp-tls: Certificate verified.
    E (30526) esp-tls: Failed to open new connection
    E (30546) TRANS_SSL: Failed to open a new connection
    E (30556) HTTP_CLIENT: Connection failed, sock < 0
    E (30556) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
    E (30556) esp_https_ota: Failed to establish HTTP connection

Toolchain: esp32-2019r1, ESP-IDF: v4.0-dev-1381-g6fe853a2c

supreetd21, 2019-11-11:
Hi @JakubRakus, @bearsh,
Does your issue seem somewhat similar to this: espressif/esp-idf#3192? The fix for this is already available on the ESP IDF master branch now.

crippsJack, 2019-11-11:
@supreetd21 Hi, which merge request fixed this issue on the master branch?

supreetd21, 2019-11-12:
Hi @crippsJack,
You can find the commit here.
https://github.com/espressif/esp-idf/commit/622d6d550494f8c4a7a5e1a742d60c2a3c5bb4e4

closed 2019-12-27 by Alvin1Zhang