espressif/esp-idf #3518: documentation: please provide a readme.txt to all ino Examples + fully detailed comments in either source code

dsyleixa, 2019-05-22:
hello,
I appreciate your multiple Arduino examples, but it's hard to see what either example actually does by which requirements and by which functionalities, and the single code lines are mostly not understandable what they do in detail and what either passed argument does.

So I would like to request to provide 
a) a readme.txt to all ino Examples for a detailed description of this special example, and
b) fully detailed comments in either source code line.

krzychb, 2019-05-22:
HI @dsyleixa, do you mean examples in [libraries](https://github.com/espressif/arduino-esp32/tree/master/libraries) folder of **arduino-esp32** repository? Please point specific folder if you mean some other examples.

dsyleixa, 2019-05-22:
As to the comments to the source code lines I actually mean all the examples, both in the Arduino libraries and in the repo, 

Tbh, I actually assumed that the repo examples are the same as in the Arduino esp32 libraries folder and also identical to those which can be accessed via 
Arduino IDE -> File -> Examples -> Examples for Adafruit Feather ESP32

Nonetheless, in the repo they are arranged more clearly, and especially here the readme.txt files would make sense eventually.

krzychb, 2019-05-24:
Hi @dsyleixa, as this issue is related primarily to Arduino examples, please post it in arduino-esp32 repository. 

All esp-idf examples have readme.md. They also have comments to substitute what is covered in the readme. The comments may not be sufficient and please point specific examples, so respective developer can look Ito it.

Alvin1Zhang, 2019-07-19:
@dsyleixa Thanks for reporting the issue, would you help share if any updates for the issue? Thanks.

dsyleixa, 2019-07-19:
to make sure that we are talking about the same sets of examples:
which repo (ur)l is recommended by you for Arduino users who are looking for step-by-step tutorial examples?

krzychb, 2020-09-09:
> which repo (url) is recommended by you for Arduino users who are looking for step-by-step tutorial examples?

For Arduino users I would recommend to check under espressif.com > Ecosystem > IOT College that provides an extensive collection of learning resources including [Courses](https://www.espressif.com/en/ecosystem/iot-college/courses) and [Books](https://www.espressif.com/en/ecosystem/iot-college/books).

closed 2020-09-09 by krzychb

dsyleixa, 2020-09-09:
all courses I can find are with costs, but free-of-charge tutorials are required, as Arduino is open-source and free-of-charge, too!