espressif/esp-idf #3519: undefined reference to '_binary_certs_pem_start

Gohith, 2019-05-22:
Unable to get .elf file in build directory, saying undefined reference to '_binary_certs_pem_start'
Error:
~/esp/WorkSpace/build/main/libmain.a(FileName_main.o):(.literal.check_update_task+0x2c): undefined reference to `_binary_certs_pem_start'

chegewara, 2019-05-22:
Do you have proper certificate in your app?
As you can see in demo, you have this variable:
https://github.com/espressif/esp-idf/blob/master/examples/system/ota/native_ota_example/main/native_ota_example.c#L36
and here is certificate:
https://github.com/espressif/esp-idf/tree/master/examples/system/ota/native_ota_example/server_certs

Last thing you have to implement is here:
https://github.com/espressif/esp-idf/blob/master/examples/system/ota/native_ota_example/main/component.mk#L6

Gohith, 2019-05-23:
Hi,
       Thank you for your quick reply, I had followed the instructions as
in Espressif Github page and I copied the ca_cert.pem files on both example
directory and OTA code directory too. But, the error still persists even
though after updating component.mk file as there.
Thank You, Regards
Gohith

chegewara, 2019-05-23:
Hi,
what i meant the certificate file name has to match the name in `component.mk` and has to match this string:
`"_binary_ca_cert_pem_start"` <-- in this case file name should be `ca_cert.pem`.

So, basically what you have to do is to get certificate file name, replace dot in name with `_` and add leading `_binary_ ` and triling `_start` part. The same with end variable.

Gohith, 2019-05-23:
Hi,
       Very thanks to you, You saved the day!!.
        Its working now, looks small spell mistake.
With regards,
Gohith

mahavirj, 2019-05-23:
Closing, as problem is solved.

closed 2019-05-23 by mahavirj