espressif/esp-idf #3520: Make menuconfig fails due to weird dependency ('zconf.tab.o')

Ian-Re, 2019-05-22:
I seem to be getting a weird issue when trying to access menuconfig, make build, clean and flash work perfectly. I'm on the latest esp-idf build.

con3@con3-pc:~/Desktop/code/esp-idf-template$ make menuconfig
[  0%] Performing build step for 'mconf-idf'
make[4]: *** No rule to make target '/home/con3/esp/esp-idf/tools/kconfig/zconf.tab.c', needed by 'zconf.tab.o'.  Stop.
make[3]: *** [CMakeFiles/mconf-idf.dir/build.make:125: mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build] Error 2
make[2]: *** [CMakeFiles/Makefile2:1069: CMakeFiles/mconf-idf.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2210: CMakeFiles/menuconfig.dir/rule] Error 2
make: *** [Makefile:1009: menuconfig] Error 2

I have seem to other closed issues on this, but I think they were in reference to macos and I'm running linux, thus the fixes didn't seem to work.

Any help with this would be appreciated!

Dazza0, 2019-05-22:
Trying navigating to `$IDF_PATH/tools/kconfig` then running `make clean` and see if that resolves the issue.

Ian-Re, 2019-05-22:
Hi @Dazza0 ,

Thank you for the help!

Here's the output of the command :

```
~/esp/esp-idf/tools/kconfig$ make clean
rm -f qconf.moc .tmp_qtcheck .tmp_gtkcheck zconf.tab.c zconf.lex.c zconf.hash.c gconf.glade.h conf.o  zconf.tab.o mconf.o zconf.tab.o lxdialog/checklist.o lxdialog/util.o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o lxdialog/checklist.o lxdialog/util.o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o conf.d zconf.tab.d mconf.d zconf.tab.d lxdialog/checklist.d lxdialog/util.d lxdialog/inputbox.d lxdialog/textbox.d lxdialog/yesno.d lxdialog/menubox.d lxdialog/checklist.d lxdialog/util.d lxdialog/inputbox.d lxdialog/textbox.d lxdialog/yesno.d lxdialog/menubox.d conf-idf mconf-idf conf mconf
```

Here's me rerunning make menuconfig:
```

~/Desktop/esp-idf-template$ make menuconfig
[  0%] Creating directories for 'mconf-idf'
[  0%] No download step for 'mconf-idf'
[  0%] No patch step for 'mconf-idf'
[  0%] No update step for 'mconf-idf'
[  0%] No configure step for 'mconf-idf'
[  0%] Performing build step for 'mconf-idf'
make[4]: *** No rule to make target '/home/con3/esp/esp-idf/tools/kconfig/zconf.tab.c', needed by 'zconf.tab.o'.  Stop.
make[3]: *** [CMakeFiles/mconf-idf.dir/build.make:125: mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build] Error 2
make[2]: *** [CMakeFiles/Makefile2:1069: CMakeFiles/mconf-idf.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2210: CMakeFiles/menuconfig.dir/rule] Error 2
make: *** [Makefile:1009: menuconfig] Error 2

```

closed 2019-05-22 by Ian-Re

Dimfred, 2019-09-02:
Hi, I have the same problem. Restarting the terminal doesn't help. 
My log:
    make menuconfig
    [  0%] Performing build step for 'mconf-idf'
    make[4]: *** No rule to make target '/media/01_Data/01_Libraries/esp32- 
lib/tools/kconfig/zconf.tab.c', needed by 'zconf.tab.o'.  Stop.
    CMakeFiles/mconf-idf.dir/build.make:125: recipe for target 'mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build' failed
    make[3]: *** [mconf-idf-prefix/src/mconf-idf-stamp/mconf-idf-build] Error 2
    CMakeFiles/Makefile2:899: recipe for target 'CMakeFiles/mconf-idf.dir/all' failed
    make[2]: *** [CMakeFiles/mconf-idf.dir/all] Error 2
    CMakeFiles/Makefile2:2230: recipe for target 'CMakeFiles/menuconfig.dir/rule' failed
    make[1]: *** [CMakeFiles/menuconfig.dir/rule] Error 2
    Makefile:1035: recipe for target 'menuconfig' failed
    make: *** [menuconfig] Error 2

tonivega, 2020-02-16:
Future me : Install **gperf** package

Dazza0, 2020-02-18:
@tonivega Suggest installing the [ESP-IDF prerequisites](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup.html#install-prerequisites) while you're at it

benpeoples, 2020-08-10:
Just to add, I just ran into this on a project.
One project built fine, but menuconfig got this error.

Other project's menuconfig worked fine.

`idf.py fullclean` fixed it.