espressif/esp-idf #3521: esp-tls.c and CONFIG_MBEDTLS_DEBUG_LEVEL

WallaceWilliam, 2019-05-22:
please, replace in esp-tls.c line 331

```
#ifdef CONFIG_MBEDTLS_DEBUG
    mbedtls_esp_enable_debug_log(&tls->conf, 4);
#endif

```
to

```
#ifdef CONFIG_MBEDTLS_DEBUG
    mbedtls_esp_enable_debug_log(&tls->conf, CONFIG_MBEDTLS_DEBUG_LEVEL);
#endif
```

mahavirj, 2019-05-23:
@WallaceWilliam Any specific reason why you are recommending this? I do not see `MBEDTLS_DEBUG_LEVEL` as config option either.

negativekelvin, 2019-05-23:
I think he is requesting that the debug threshold be configurable and that includes adding the config option.

WallaceWilliam, 2019-05-23:
> I think he is requesting that the debug threshold be configurable and that includes adding the config option.

so much
if threshold = 4 a lot of debug information is issued

closed 2019-06-10 by igrr