espressif/esp-idf #3522: wishful: std::thread implementation for ESP32

dsyleixa, 2019-05-22:
wishful IMO: 
a std::thread implementation for ESP32, like meanwhile already available for Teensy (TeensyThread):
https://github.com/ftrias/TeensyThreads
or perhaps native C++ <thread>

igrr, 2019-05-22:
Please see https://github.com/espressif/esp-idf/tree/master/examples/system/cpp_pthread

dsyleixa, 2019-05-23:
> 
> 
> Please see https://github.com/espressif/esp-idf/tree/master/examples/system/cpp_pthread

how can this be used by Arduino IDE? I see no Arduino ino example unfortunately

igrr, 2019-05-23:
Hi @dsyleixa, for Arduino related issues, please use this tracker: https://github.com/espressif/arduino-esp32/issues

The code mentioned above should also be usable in Arduino, provided you include the same header files.

dsyleixa, 2019-05-23:
ok, but that is really unclearly ordered here, tbh...

Alvin1Zhang, 2021-12-31:
Thanks for reporting, let us tracker further in Arduino issue tracker https://github.com/espressif/arduino-esp32/issues.

closed 2021-12-31 by Alvin1Zhang