espressif/esp-idf #3524: GPIO set as OUTPUT will be set as High, PROBLEM

filzek, 2019-05-22:
Latest Master as of 05/22/2019

There is a problem when setting a GPIO, 

Once a default routine is set in ESP-IDF example: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/gpio

 gpio_config_t io_conf;
    //disable interrupt
    io_conf.intr_type = GPIO_PIN_INTR_DISABLE;
    //set as output mode
    io_conf.mode = GPIO_MODE_OUTPUT;
    //bit mask of the pins that you want to set,e.g.GPIO18/19
    io_conf.pin_bit_mask = GPIO_OUTPUT_PIN_SEL;
    //disable pull-down mode
    io_conf.pull_down_en = 0;
    //disable pull-up mode
    io_conf.pull_up_en = 0;
    //configure GPIO with the given settings
    gpio_config(&io_conf);

GPIO 18 / 19 will be on HIGH,

How to set and config the GPIO to NEVER goes to HIGH on setup? 

This is a problem.

chegewara, 2019-05-22:
`io_conf.pull_down_en = `1;`

filzek, 2019-05-22:
Ok, i also did find change gpio_matrix_out to false in the first statement also work, but it has change in the core gpio.

filzek, 2019-05-23:
> `io_conf.pull_down_en = 1;`

This doesnt work at all, will enable output as high state.

Unique way to enable a GPIO in a LOW state is changing the GPIO.C code in the function 

static esp_err_t gpio_output_enable(gpio_num_t gpio_num)
{
    GPIO_CHECK(GPIO_IS_VALID_OUTPUT_GPIO(gpio_num), "GPIO output gpio_num error", ESP_ERR_INVALID_ARG);

    if (gpio_num < 32) {
        GPIO.enable_w1ts = (0x1 << gpio_num);
    } else {
        GPIO.enable1_w1ts.data = (0x1 << (gpio_num - 32));
    }
    gpio_matrix_out(gpio_num, SIG_GPIO_OUT_IDX, false, false);
    //esp_restart();
    return ESP_OK;
}

CHANGE IT TO:
gpio_matrix_out(gpio_num, SIG_GPIO_OUT_IDX, true, false); 

So, expressif need to add a patch to fix this in the oficial SDK, the best is to add an extra field in the gpio_config_t as something like  
output_action = (HIGH / LOW)

So, once setting the GPIOS with the gpio_config(&io_conf); will add a real situation where a GPIO could be setup as an OUTPUT and can bring it HIGH or LOW state.

negativekelvin, 2019-05-23:
This is how is how uart avoid glitch 
https://github.com/espressif/esp-idf/blob/81ca1c01395f604972fbf141cfbe49764a746023/components/driver/uart.c#L598-L599

filzek, 2019-06-17:
Will close this issue as Espressif already know this and shall implement an option in the future SDK to allow the option to ENABLE the output mode in HIGH or LOW and the GPIO SETUP!

For now, the unique solution is to SETUP GPIO manually in all steps or change GPIO.C sdk file to always enable the output as LOW always.

projectgus, 2019-06-17:
Reopening this as there should be a way to set the initial output level when calling gpio_config(), and we don't support this yet.

filzek, 2019-06-25:
> Reopening this as there should be a way to set the initial output level when calling gpio_config(), and we don't support this yet.

@projectgus  Do you want that I add this option in all files related and send to you? I could do it as I already did to my own need.

projectgus, 2019-06-25:
@filzek You'd be very welcome to send a Github Pull Request that adds this field in the gpio_config_t and adds whatever code support is needed for it to work correctly. Check https://docs.espressif.com/projects/esp-idf/en/latest/contribute/index.html for more details about how best to do this.

Alvin1Zhang, 2020-01-19:
@filzek Thanks for reporting, would you please help share if any updates for the issue? Thanks.

filzek, 2021-01-05:
I will check about this if this was fixed, can test it tomorrow and post updates

Alvin1Zhang, 2021-01-26:
@filzek Thanks for the updates, would you please share if any progress of this issue? Thanks.

Alvin1Zhang, 2021-03-09:
@filzek Would you please help share any updates of this issue? Thanks.

ygoe, 2022-01-09:
Is this issue about all output GPIOs being HIGH at boot? That's what I see here. I have an ESP32-PICO and attached 8 small LEDs to different GPIOs. I want to use the LED PWM controller and control my room lights with this. Right now, I see all 8 LEDs flash up for an instant when booting the device. Imagine this with real bright room lights. It would feel like a lightbulb that just burnt out when switched on. I've had that effect when I missed a pull-down resistor on my AVR MCU that does the job now. Do I need those with the ESP32, too? Or is it a software issue?

I have code like this at the beginning of my app:

```c
gpio_reset_pin(LED1_GPIO);   // from blink example
gpio_set_level(LED1_GPIO, 0);   // from a comment in this issue
gpio_set_direction(LED1_GPIO, GPIO_MODE_OUTPUT);   // from blink example
```

Everything is working fine, but there's that highly annoying flash at boot.

closed 2022-04-29 by filzek