espressif/esp-idf #3525: Hot to set output level of GPIO32 - ESP32 Pico D4

wowaz, 2019-05-22:
Hi,

I just can't get the IO32 work as an output pin on the ESP32 Pico D4.

With the code below **gpio_get_level** returns the level defined with **gpio_set_level** but while measuring the pin it stays always low.

```C
uint64_t bitmask = (1ULL << GPIO_NUM_32);
gpio_config_t io_conf = {
        .intr_type = GPIO_PIN_INTR_DISABLE,
        .mode = GPIO_MODE_INPUT_OUTPUT,
        .pin_bit_mask = bitmask,
	.pull_down_en = GPIO_PULLDOWN_DISABLE,
	.pull_up_en = GPIO_PULLUP_DISABLE,
};
gpio_config(&io_conf);
gpio_set_level(GPIO_NUM_32, 1);
ESP_LOGI("Test", "Test %d", gpio_get_level(GPIO_NUM_32));
```

At first I though the connection was somewhere broken, but if the mode is set as input only **gpio_get_level** returns the levels correctly according to the applied source.

I also tried **gpio_pad_select_gpio** and **rtc_gpio_deinit** as descripted in #285 without success.

Any ideas what I'm doing wrong?

bhusang, 2019-05-22:
Are you using the chip directly or on an eval board?

IO32 | 12 | I/O | 32K_XP (32.768 kHz crystal oscillator input), ADC1_CH4, TOUCH9, RTC_GPIO9

The only I/O restrictions that are in the datasheet:
Note that the I/O GPIO pads are 0-19, 21-23, 25-27, 32-39, while the output GPIOs are 0-19, 21-23, 25-27, 32-33. GPIO pads 34-39 are input-only.

wowaz, 2019-05-22:
I'm using the chip directly, trying to use IO32 as SPI_CS with a 10k pull-down resistor.
With the same circuit everything works without a problem on IO25.

wowaz, 2019-08-13:
Well, after assembling another PCB everything worked just fine.
My fault, sorry.

closed 2019-08-13 by wowaz