espressif/esp-idf #3526: HOW TO USE LWIP_HOOK_UNKNOWN_ETH_PROTOCOL

yanzixiang, 2019-05-23:
HOW TO USE THIS ?

https://github.com/espressif/esp-lwip/blob/2.0.3-esp/src/netif/ethernet.c
```C
#ifdef LWIP_HOOK_UNKNOWN_ETH_PROTOCOL
      if(LWIP_HOOK_UNKNOWN_ETH_PROTOCOL(p, netif) == ERR_OK) {
        break;
      }
#endif
```

https://www.nongnu.org/lwip/2_1_x/group__lwip__opts__hooks.html#ga6cc92f5101dd7a0420e18e1f2101abf4

Can this be done by modify the eth_config_t defined here ?
https://github.com/espressif/arduino-esp32/blob/010a7c60f7000eb17c6333309c87156769848a0e/tools/sdk/include/ethernet/esp_eth.h
```c
/**
 * @brief ethernet configuration
 *
 */
typedef struct {
    eth_phy_base_t phy_addr;                                            /*!< PHY address (0~31) */
    eth_mode_t mac_mode;                                                /*!< MAC mode: only support RMII now */
    eth_clock_mode_t clock_mode;                                        /*!< external/internal clock mode selection */
    eth_tcpip_input_func tcpip_input;                                   /*!< tcpip input func */
    eth_phy_func phy_init;                                              /*!< phy init func */
    eth_phy_check_link_func phy_check_link;                             /*!< phy check link func */
    eth_phy_check_init_func phy_check_init;                             /*!< phy check init func */
    eth_phy_get_speed_mode_func phy_get_speed_mode;                     /*!< phy check init func */
    eth_phy_get_duplex_mode_func phy_get_duplex_mode;                   /*!< phy check init func */
    eth_gpio_config_func gpio_config;                                   /*!< gpio config func  */
    bool flow_ctrl_enable;                                              /*!< flag of flow ctrl enable */
    eth_phy_get_partner_pause_enable_func phy_get_partner_pause_enable; /*!< get partner pause enable */
    eth_phy_power_enable_func phy_power_enable;                         /*!< enable or disable phy power */
    uint32_t reset_timeout_ms;                                          /*!< timeout value for reset emac */
} eth_config_t;
```
And 
https://github.com/espressif/esp-idf/blob/6fd535c985f0019ffe8a924e62a57e29a9007a28/components/ethernet/emac_common.h
```c
struct emac_config_data {
    eth_phy_base_t phy_addr;
    eth_mode_t mac_mode;
    eth_clock_mode_t clock_mode;
    struct dma_extended_desc *dma_etx;
    uint32_t cur_tx;
    uint32_t dirty_tx;
    int32_t cnt_tx;
    struct dma_extended_desc *dma_erx;
    uint32_t cur_rx;
    uint32_t dirty_rx;
    int32_t cnt_rx;
    uint32_t rx_need_poll;
    bool phy_link_up;
    enum emac_runtime_status emac_status;
    uint8_t macaddr[6];
    eth_phy_func phy_init;
    eth_tcpip_input_func emac_tcpip_input;
    eth_gpio_config_func emac_gpio_config;
    eth_phy_check_link_func emac_phy_check_link;
    eth_phy_check_init_func emac_phy_check_init;
    eth_phy_get_speed_mode_func emac_phy_get_speed_mode;
    eth_phy_get_duplex_mode_func emac_phy_get_duplex_mode;
    bool emac_flow_ctrl_enable;
    bool emac_flow_ctrl_partner_support;
    eth_phy_get_partner_pause_enable_func emac_phy_get_partner_pause_enable;
    eth_phy_power_enable_func emac_phy_power_enable;
    uint32_t reset_timeout_ms;
    bool promiscuous_enable;
};
```

suda-morris, 2019-05-23:
Hi @yanzixiang 
One solution I can come up with is define the hook handler near [here](https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/include/lwipopts.h#L648).  e.g. 
```c
extern err_t eth_unknow_type_hook(struct pbuf* pbuf, struct netif* netif);

#define LWIP_HOOK_UNKNOWN_ETH_PROTOCOL eth_unknow_type_hook
```
`eth_unknow_type_hook` should be implemented in your project code. You don't have to modified the ethernet driver because the ethernet driver only process the regular ETHERNET type frame.

negativekelvin, 2019-05-23:
You have to #define LWIP_HOOK_FILENAME and inside that file #define LWIP_HOOK_UNKNOWN_ETH_PROTOCOL as your function.

yanzixiang, 2019-05-26:
> You have to #define LWIP_HOOK_FILENAME and inside that file #define LWIP_HOOK_UNKNOWN_ETH_PROTOCOL as your function.

It is here
https://github.com/espressif/esp-lwip/blob/18548d1e256f7651fb4da48dbd770556e0eca8ce/src/include/lwip/opt.h#L2422

yanzixiang, 2019-05-26:
https://github.com/espressif/esp-lwip/pull/7

Can someone help to have a look at this

freakyxue, 2019-11-13:
@yanzixiang 
Thanks for the contribution code, we will review

david-cermak, 2021-02-22:
@yanzixiang Thanks for raising this question and the PR in the lwip project. As mentioned [here](https://github.com/espressif/esp-lwip/pull/7), this hook, as well as any other lwip hooks, should be added to the port layers. Although we already have an internal IDF's lwip hook definition, I do not think we should include the `LWIP_HOOK_UNKNOWN_ETH_PROTOCOL` as a default or Kconfig defined function, since this seems to me like a corner case for a general IoT use case. This will not prevent users to define custom lwip hooks if needed, as @suda-morris suggested [above](https://github.com/espressif/esp-idf/issues/3526#issuecomment-495049535) (just note that we have to define [`ESP_IDF_LWIP_HOOK_FILENAME`](https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/include/lwip_default_hooks.h#L21-L23)  from  v4.3, see 9ffa9cce73)

Could you please close this issue if you agree with the proposed assessment or comment otherwise?

closed 2021-02-22 by yanzixiang