espressif/esp-idf #3527: How do I connect more than 3 SPI devices

amoghskulkarni, 2019-05-23:
`spi_master.c` driver is implemented in such a way that there is a hard limit on the maximum number of SPI devices that can be communicated with (as shown [here](https://github.com/espressif/esp-idf/blob/master/components/driver/spi_master.c#L152)).

1. Why this hard limit?
2. Is there a work around for connecting more than 3 SPI devices?

P.S. - I tried to manually change the macro value to more than 3, but it didn't change anything (besides not giving the error). I found [a reddit thread](https://www.reddit.com/r/esp32/comments/7shjpd/is_there_any_way_around_the_3_spi_device_limit/) that discusses this issue and a possible workaround, but I didn't understand how to use the workaround in my code.

nopnop2002, 2019-05-27:
Hello.
This is my idea, and I have not tried.

	spi_device_interface_config_t devcfg;
	memset( &devcfg, 0, sizeof( spi_device_interface_config_t ) );
	devcfg.clock_speed_hz = SPI_Frequency;
	devcfg.spics_io_num = -1;
	devcfg.queue_size = 1;
	spi_device_handle_t handle;
	ret = spi_bus_add_device( HSPI_HOST, &devcfg, &handle);

CS of SPI device #1 is GPIO_XX
CS of SPI device #2 is GPIO_YY
CS of SPI device #3 is GPIO_ZZ

You set GPIO_XX GPIO_YY GPIO_ZZ manualy.

ginkgm, 2019-05-30:
Hi all,
The hardware only has three CS signals, so that we set the limit to 3 in the early version.
If you want to support more chips, you can set ``NO_CS`` definition to greater value. And use software cs (controlled by ``gpio_set_level`` functions) in the pretrans and posttrans callbacks to activate corresponding slave.

Please take care of all cases where ``3`` or ``NO_CS`` are used in spi_master.c yourself. For example, you have to forbid the cs initialization when the device is 4 or larger:
spi_master.c:433
```c
    //Set CS pin, CS options
    if (dev_config->spics_io_num >= 0) {
        spicommon_cs_initialize(host, dev_config->spics_io_num, freecs, !(spihost[host]->flags&SPICOMMON_BUSFLAG_NATIVE_PINS));
    }
```

BTW, kindly notice: more slaves connected to the same SPI bus will cause the timing to be worse. The delay of all lines will be greater, which will limit your working frequency.

amoghskulkarni, 2019-06-01:
@ginkgm Thank you for the explanation. I am currently trying to get the software CS working for more than 3 devices, will be dropping any new development update/question on this issue. Please feel free to close the issue.

amoghskulkarni, 2019-06-02:
> If you want to support more chips, you can set `NO_CS` definition to greater value. And use software cs (controlled by `gpio_set_level` functions) in the pretrans and posttrans callbacks to activate corresponding slave.

@ginkgm I tried this and my 4th SPI device still wouldn't respond. My setup looks something like this -

```c++
void swcs_pre_cb (spi_transaction_t *trans){
    // reset the pin
    gpio_set_level(CS_PIN, 0);
    return;
}

void swcs_post_cb (spi_transaction_t *trans) {
    // set the pin
    gpio_set_level(CS_PIN, 1);
    return;
}

spi_device_interface_config_t dev_config;

dev_config.command_bits = 0;
dev_config.address_bits = 8;
dev_config.dummy_bits = 0;
dev_config.mode = 0;
dev_config.duty_cycle_pos = 128;  // default 128 = 50%/50% duty
dev_config.cs_ena_pretrans = 0;  // 0 not used
dev_config.cs_ena_posttrans = 0;  // 0 not used
dev_config.clock_speed_hz = 1000000;
dev_config.spics_io_num = -1; // SW cs 
dev_config.flags = 0;  // 0 not used
dev_config.queue_size = 1;
dev_config.pre_cb = swcs_pre_cb;
dev_config.post_cb = swcs_post_cb;

hspi.addDevice(&dev_config, &spi_handle);

gpio_set_direction(CS_PIN, GPIO_MODE_INPUT_OUTPUT);
PIN_FUNC_SELECT(CS_PIN, PIN_FUNC_GPIO);
```
The exact same configuration works for the first 3 devices with `NULL` callbacks, which brings me to following comment of yours -

> Please take care of all cases where `3` or `NO_CS` are used in spi_master.c yourself. For example, you have to forbid the cs initialization when the device is 4 or larger:
> spi_master.c:433

I could not determine which of the parts in `spi_master.c` need changes, because wherever `NO_CS` is being used, is used in the bus acquiring logic which my 4th device needs anyway.

amoghskulkarni, 2019-06-02:
P.S. I am taking care that the CS pins don't get initialized within `spi_master.c` driver functions by providing -1 to `spics_io_num` member of the config object.

o-marshmallow, 2022-01-13:
Closing this issue, if you are still countering this issue, please open a new ticket.

closed 2022-01-13 by o-marshmallow