espressif/esp-idf #3529: Heap space is sufficient, but malloc fails

GitOfLance, 2019-05-23:
### Environment
- Module or chip used: ESP32-WROOM-32, Chip is ESP32D0WDQ6 (revision 1)
- IDF version : v4.0-dev-207-g76da10a66
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

### Problem Description
I am going to convert a binary image data of `26400` Bytes into a string via `base64`, put it into `cJSON`, and transfer it through `MQTT`.
So I probably need space: `malloc 26400 * 4 / 3 + 4 = 35204` Bytes, but I have been unable to malloc success, I use `esp_get_free_heap_size()` and other functions to display the debugging information is as follows:
[image]
The corresponding code is as follows:
```
	needslen = image_len / 3 * 4 + 4;
	freelen = esp_get_free_heap_size();
	minimum = esp_get_minimum_free_heap_size();
	printf("heap_caps_get_free_size: %d\n", heap_caps_get_free_size(MALLOC_CAP_8BIT));
	printf("heap_caps_get_minimum_free_size: %d\n", heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT));
	printf("heap_caps_get_largest_free_block: %d\n", heap_caps_get_largest_free_block(MALLOC_CAP_8BIT));
		
	printf("base64: image len = %ld, needs %ld, free head %ld, can use minimum is %ld\n", image_len, needslen, freelen, minimum);
#if 1
	while (1)
	{
		userinfo->image = (char *)os_malloc(needslen + 4);
		if (userinfo->image)
		{
			ESP_LOGE(TAG, "malloc for base64 success");
			break;
		}
		else 
		{
			vTaskDelay(2000 / portTICK_PERIOD_MS);
			printf("<-->\n");
		}
	}
#endif	
```

So,
- Does my operation meet the requirements?
- How can I guarantee a successful application for large space through malloc? I used to fail when `malloc 26400Byte`.
- I don't quite understand the role of this function: `esp_get_minimum_free_heap_size()`. Can you explain it in detail?

chegewara, 2019-05-23:
Largest free is `17344`

GitOfLance, 2019-05-23:
So the maximum space that can actually be applied is based on the return value of `heap_caps_get_largest_free_block()`?

GitOfLance, 2019-05-23:
The description of this function is as follows:
```
/**
 * @brief Get the largest free block of memory able to be allocated with the given capabilities.
 *
 * Returns the largest value of ``s`` for which ``heap_caps_malloc(s, caps)`` will succeed.
 *
 * @param caps        Bitwise OR of MALLOC_CAP_* flags indicating the type
 *                    of memory
 *
 * @return Size of largest free block in bytes.
 */
size_t heap_caps_get_largest_free_block( uint32_t caps );
```
This function is for `heap_caps malloc(s, caps); `if I use `malloc`, do I need to pay attention to `heap_caps_get_minimum_free_size()`?

GitOfLance, 2019-05-23:
If the return value of `heap_caps_get_largest_free_block()` is true and reliable, then:
- What is the use of these functions?Like:`esp_get_free_heap_size()`、`esp_get_minimum_free_heap_size()`...
- Why is the value much smaller than the return value of the :`esp_get_free_heap_size()` and `esp_get_minimum_free_heap_size()`?

chegewara, 2019-05-23:
> So the maximum space that can actually be applied is based on the return value of `heap_caps_get_largest_free_block()`?

This is applied if you want to make single allocation. Buffer allocated with malloc cant be fragmented.

negativekelvin, 2019-05-23:
Because of heap fragmentation and/or segmentation.  You still have 39k free heap but you just can't use it all in 1 allocation.

GitOfLance, 2019-05-23:
You mean that allocating continuous space depends on `heap_caps_get_largest_free_block()`, and malloc allocates space because it can be discontinuous, so it does not depend on this function?
If my understanding is correct, then I should not pay attention to this function when I `malloc`. Why did I fail to apply for space through `malloc`? I just need so much space, whether it's continuous or separate.
Do I understand correctly? Is that what you want to say?@chegewara

negativekelvin, 2019-05-23:
malloc just ends up calling heap_caps_malloc and they both are limited to heap_caps_get_largest_free_block

GitOfLance, 2019-05-23:
@negativekelvin 
I see, but I remember when malloc applied for larger space, if there was not a large enough heap fragment, it would trigger defragmentation and splice all the fragments together, so theoretically, as long as the total space was enough, I could malloc. Now what I understand is that this function gets the total heap space size. This function is the size I can apply for. Why are these two different?

Can the problem be solved by adding on-chip DRAM? Or what you say about `stream processing`, but I don't understand this, can I say more about it?

negativekelvin, 2019-05-23:
> I remember when malloc applied for larger space, if there was not a large enough heap fragment, it would trigger defragmentation and splice all the fragments together, so theoretically, as long as the total space was enough, I could malloc

I don't think so, plus there are some fixed boundaries that segment heap and can't be moved

Sure, if you add a psram chip you will be able to store large buffers there.

Stream processing means don't base64-encode the image before you send it, do it while you are sending it

GitOfLance, 2019-05-23:
The fragmentation splicing of malloc function is the splicing of virtual addresses, not physical addresses, but it depends on the operating system, maybe ESP32 + FreeRTOS solution is different.
Stream processing or subcontracting is a good method, but to modify the interaction logic, so I first try to increase PSRAM.
Thank you very much for your answers. @negativekelvin @chegewara Thanks!

GitOfLance, 2019-05-28:
@negativekelvin 
I ported the program to the `ESP_WROVER_KIT` development board, but the function to get the stack size has not changed?
```
E (482) main: free head = 118372, minimum head = 118372
heap_caps_get_free_size: 118372
heap_caps_get_minimum_free_size: 118372
heap_caps_get_largest_free_block:8BIT 66412
heap_caps_get_largest_free_block:32BIT 66412
```
Are there any other settings on the development board?

negativekelvin, 2019-05-28:
https://github.com/espressif/esp-idf/blob/master/components/esp32/Kconfig#L23

GitOfLance, 2019-05-29:
@negativekelvin 
Thank you very much, the problem is solved.

Alvin1Zhang, 2019-05-30:
@GitOfLance Thanks for reporting the issue, and @negativekelvin thanks for the notes.

closed 2019-05-30 by Alvin1Zhang