espressif/esp-idf #3532: NimBLE - how to request pairing/encryption from peripheral without setting encrypted attribute permissions

bfriedkin, 2019-05-23:
Hello -

I believe that a BLE peripheral can only request that pairing/encryption be initiated. The BLE central must initiate the procedure.

With NimBLE a peripheral can configure the various SM options by setting the associated fields in the `ble_hs_cfg` structure. In my case I want LE 4.2 secure connections, MITM protection and DISPLAY_ONLY IO capabilities. I have set the 4.2 secure connection entry in the sdkconfig.defaults to Y. This is how I configure the options in my code:

```
ble_hs_cfg.sm_sc = 1;
ble_hs_cfg.sm_mitm = 1;
ble_hs_cfg.sm_io_cap = BLE_SM_IO_CAP_DISP_ONLY;
```

Unfortunately I have found that setting these SM options isn't enough to request an encrypted connection. When a central accesses my peripheral's services, no pairing is initiated and the attributes can be freely read/written. What seems to be additionally required is to set the `CHR_F_READ_ENC` flag in the characteristic declarations deployed in the services. 

On ESP32, as a peripheral, I can request an encrypted connection with MITM protection by calling `esp_ble_set_encryption()` when the central connects to the peripheral. Is there a way to request an encrypted connection with pairing from a peripheral using NimBLE without having to set the `CHR_F_READ_ENC` flag in the characteristic entries?

Thanks,
Brian

bfriedkin, 2019-05-23:
Hello -

A short update. I found the `ble_gap_security_initiate()` API and calling that from the peripheral code does mostly resolve the issue. I call the API from the `BLE_GAP_EVENT_CONNECT` event. When called, pairing is initiated by the central. I don't need to change any of the characteristic flags.

There is one failure case however. If I request a LE 4.2 secure connection from the app, no pairing is initiated and the characteristics can be freely accessed:

```
ble_hs_cfg.sm_sc = 1;
```

When I set this value to 0, pairing is initiated as expected:

```
ble_hs_cfg.sm_sc = 0;
```

Regards,
Brian

dhrishi, 2019-05-24:
@bfriedkin Yes, that's right. You can use the API `ble_gap_security_initiate()`.

> There is one failure case however. 

About this, it seems to be a regression due to a recent change. Can you please pull the latest `feature/nimble-preview` branch and test. Make sure that you remove `sdkconfig` file in your example and then recompile as the fix includes some changes in the configuration options.

bfriedkin, 2019-05-24:
@dhrishi -

Thank you for the quick follow-up. Indeed my branch was out of date. I see the recent fixes in the branch.

I pulled the latest `feature/nimble-preview` and unfortunately I am unable to successfully complete the firmware build:

```
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
/Users/brianfriedkin/esp32/esp-idf/components/bootloader_support/src/secure_boot_signatures.c:22:18: fatal error: uECC.h: No such file or directory
compilation terminated.
make[2]: *** [src/secure_boot_signatures.o] Error 1
```

I tried updating both my working copy and starting with a freshly cloned ESP-IDF. How do I get past this?

Regards,
Brian

dhrishi, 2019-05-24:
@bfriedkin It looks like you have switched from `master` branch. The thing is that `micro-ecc` component has been moved in bootloader directory in recent times. As the 'feature/nimble-preview` branch was forked before that, hence the error.
> How do I get past this?
On your `feature/nimble-preview` branch, check the status `git status | less` and just remove the untracked folders (the new bootloader subcomponent should be one of them) and give it a try.
Let me know if the issue still persists.

bfriedkin, 2019-05-24:
@dhrishi -

Removing the untracked folders allowed me to get through the hello_world build. Thank you for the tip. But when I try to build the `bleprph` example, it looks like the `ble_store_config.c` file isn't built?

```
LD build/bleprph.elf
/Users/brianfriedkin/esp32/esp-idf-nimble-new/examples/bluetooth/nimble/bleprph/build/main/libmain.a(main.o):(.literal.app_main+0x38): undefined reference to `ble_store_config_init'
/Users/brianfriedkin/esp32/esp-idf-nimble-new/examples/bluetooth/nimble/bleprph/build/main/libmain.a(main.o): In function `app_main':
/Users/brianfriedkin/esp32/esp-idf-nimble-new/examples/bluetooth/nimble/bleprph/main/main.c:375: undefined reference to `ble_store_config_init'
```

I'm not sure why this is happening.  Please let me know if you have any suggestions. I'll just remove the call from the example app(s) for now, so that I can move forward and retest the 4.2 secure connection + pairing issue.

Regards,
Brian

dhrishi, 2019-05-25:
@bfriedkin Did you update the NimBLE submodule `git submodule update --init` ?

bfriedkin, 2019-05-25:
@dhrishi -

I was able to confirm that the original issue is resolved after updating to the latest `nimble/feature-preview` branch. Thank you again for helping out.

I'll try starting with a fresh checkout again and follow-up on the missing symbol. But I typically do this:

```
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout feature/nimble-preview
git submodule update
```

I'll try the same but add the `--init` parameter.

Regards,
Brian

BTW: I think it would be useful if the `blecent` app supported the same pairing/encryption features found in the `bleprph` app. I am now working through the central side.

dhrishi, 2019-05-25:
@bfriedkin The changes on the central side should be similar. Or, may be, you can test bleprph by any BLE scanner/connect app on your mobile phone. Although, I will take your feedback and see what could be done.

Please close this if your original issue is resolved.

bfriedkin, 2019-05-25:
@dhrishi -

We don't need to test the peripheral side. That does seems to work well. We now need to test the central side, which is why I suggest providing a `blecent` implementation that parallels the `bleprph` implementation. The Bluedroid examples include `gatt_security_client` and `gatt_security_server` examples, which we used when bringing up our wrappers. That would be very helpful on NimBLE as well. Thank you for your consideration.

Regards,
Brian

closed 2019-05-25 by bfriedkin

redradist, 2021-01-27:
> @dhrishi -
>
> We don't need to test the peripheral side. That does seems to work well. We now need to test the central side, which is why I suggest providing a `blecent` implementation that parallels [...]

@bfriedkin Have you some good example for encryption of Ble attributes ?
I've recently faced with issue that even after pairing I do not see in Android Application available attributes with `ReadEnc` attribute enabled

bfriedkin, 2021-01-28:
@redradist -

Our work with NimBLE is part of the [Moddable SDK](https://github.com/Moddable-OpenSource/moddable). The primary programming language for development is JavaScript, so all our BLE [example apps](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/network/ble) are written in JavaScript, but the BLE bindings to the JavaScript API are written in C. It doesn't look like we published any example apps that support characteristic level encryption, but the feature does work for us when hosting services with encrypted attributes.

If you want to enable "read encrypted" permissions on a service characteristic, you'll need to set the flags field for the characteristic in the associated `ble_gatt_chr_def` structure to `BLE_GATT_CHR_F_READ_ENC | BLE_GATT_CHR_PROP_READ`. Here is an example generated by our build [tools](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/tools.md). Note that the flags value `514` corresponds to the two bit fields described above:

```
static const struct ble_gatt_svc_def gatt_svr_svcs[] = {
	{
		// Service 180F
		.type = BLE_GATT_SVC_TYPE_PRIMARY,
		.uuid = &service0_uuid.u,
		.characteristics = (struct ble_gatt_chr_def[])
		{
			{
				.uuid = &service0_chr0_uuid.u,
				.access_cb = gatt_svr_chr_static_value_access_cb,
				.flags = 514,
				.arg = NULL,
				.val_handle = &service0_chr0_handle,
			},
			{
				0,
			},
		}
	},
	{
		0,
	}
};
```

The Moddable SDK is open source and you can browse the code. The BLE server implementation can be found in the [modBLEServer.c](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/network/ble/nimble/modBLEServer.c) source. The `deployServices()` function deploys the BLE services defined by the host app.

Hope this helps.

Regards,
Brian

BoKTek, 2022-03-24:
Hello

I have a similary issue as you bfriedkin, but i can't figure it out. I have tried to use "ble_gap_security_initiate()" in Blecent but the BLE peripheral don't get any response. I don't exactly know where to put it.
I have also tried to change all the different settings in ble_hs_cfg.sm**** registers, but without any luck.
Have you got Blescent to work properly?
If it is working could you please tell me what you have done?

Best regards.
Bo

felixcollins, 2023-05-23:
We have similar issues. Our old device used the bluedroid port on esp32 and when the phone did a characteristic write to the device (auth and enc characteristic flags set), that would trigger the phone to initiate pairing. Now we are using nimble port and that no longer happens. I'm trying the ble_gap_security_initiate() if a new connection comes from an unknown (unbonded) phone. Is this the correct approach? It is not clear from the docs if ble_gap_security_initiate() should even be called on a server device. We are finding that sometimes it works and sometimes it fails and the connection is dropped. The phone sometimes uses a public address and sometimes uses a random address and this seems to have an impact.