espressif/esp-idf #3534: HTTP_CLIENT: Error parse url  AND  HTTP_CLIENT: Error while setting default configurations

Gohith, 2019-05-24:
Hello,
          Initially I tried the GitHub (espressif) OTA example code and its worked on esp32. After few times successfully ran, Now the following error's are displaying while connecting to HTTPS server. 

E (9844) HTTP_CLIENT: Error parse url  (HERE THE LOCAL HOST URL )
E (9854) HTTP_CLIENT: Error while setting default configurations
E (9864) native_ota_example: Failed to initialise HTTP connection
E (9864) native_ota_example: Exiting task due to fatal error...

[image]

jitin17, 2019-06-06:
@Gohith I suspect that your URL is incorrect. 
If you are hosting the OTA update firmware image on a local HTTPS server then your URL should be: `https://<server_host_ip>:<server_port>/<image_name.bin>`.

Gohith, 2019-06-06:
Thank you for concern, it was anyway resolved now.
It was url thing as you said.

jitin17, 2019-06-06:
@Gohith Thanks for your prompt reply. I am closing this issue.

closed 2019-06-06 by jitin17