espressif/esp-idf #3537: Periodically Wi-Fi disconnection in automatic light sleep mode with 3.2 esp-idf version

gal1990, 2019-05-24:
## Problem Description
In our project, we are using Amazon web services IoT platform and automatic light sleep mode (enabled power management and  tickless idle support) for low power consumption. Any time before the user code execution we are calling"pm_awake_lock" and  "pm_apb_freq_lock" acquire (esp_pm_lock_acquire) and when the user execution finished, we release (esp_pm_lock_release) them. 
Observed issue: Our device disconnects from the Wi-Fi network after  1-20 minutes and usually, this happens when it is running in idle (sleep)  mode. 

## Debug Logs

D (148327) sleep: RTC_PERIPH: AUTO(OFF), RTC_SLOW_MEM: ON, RTC_FAST_MEM: ON
D (148407) phy_init: wifi mac time delta: 96296
D (148427) sleep: RTC_PERIPH: AUTO(OFF), RTC_SLOW_MEM: ON, RTC_FAST_MEM: ON
D (180517) phy_init: wifi mac time delta: 31460303
I (180517) wifi: state: run -> init (6c0)
I (180517) wifi: pm stop, total sleep time: 173269964 us / 183433605 us

I (180517) wifi: n:2 0, o:2 1, ap:255 255, sta:2 1, prof:1
D (180527) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:CicadaWiFi, ssid_len:10, bssid:6c:3b:6b
:f9:cc:4f, reason:6
V (180537) event: enter default callback
V (180537) tcpip_adapter: check: local, if=0 fn=0x400e6190
0x400e6190: tcpip_adapter_down_api at C:/msys32/home/ati/esp/esp-idf/components/tcpip_adapter
/tcpip_adapter_lwip.c:1092

D (180547) tcpip_adapter: if0 start ip lost tmr: enter
D (180547) tcpip_adapter: if0 start ip lost tmr: interval=120
V (180557) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (180557) tcpip_adapter: check: remote, if=0 fn=0x400e6190
0x400e6190: tcpip_adapter_down_api at C:/msys32/home/ati/esp/esp-idf/components/tcpip_adapter
/tcpip_adapter_lwip.c:1092

V (180567) event: exit default callback
W (180567) WIFI: Disconnected with reason: 6
I (180697) wifi: n:2 1, o:2 0, ap:255 255, sta:2 1, prof:1
I (180697) wifi: state: init -> auth (b0)
I (180697) wifi: state: auth -> assoc (0)
I (180707) wifi: state: assoc -> run (10)
I (180717) wifi: connected with CicadaWiFi, channel 2
D (180717) nvs: nvs_set sta.chan 1 2
D (180727) nvs: nvs_set_blob sta.apinfo 700
I (180767) wifi: pm start, type: 1

D (180767) event: SYSTEM_EVENT_STA_CONNECTED, ssid:CicadaWiFi, ssid_len:10, bssid:6c:3b:6b:f9
:cc:4f, channel:2, authmode:3
V (180767) event: enter default callback
V (180767) tcpip_adapter: check: local, if=0 fn=0x400e609c
0x400e609c: tcpip_adapter_up_api at C:/msys32/home/ati/esp/esp-idf/components/tcpip_adapter/t
cpip_adapter_lwip.c:1092

V (180777) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (180787) tcpip_adapter: check: remote, if=0 fn=0x400e609c
0x400e609c: tcpip_adapter_up_api at C:/msys32/home/ati/esp/esp-idf/components/tcpip_adapter/t
cpip_adapter_lwip.c:1092

V (180787) tcpip_adapter: check: local, if=0 fn=0x400e64f4
0x400e64f4: tcpip_adapter_dhcpc_start_api at C:/msys32/home/ati/esp/esp-idf/components/tcpip_
adapter/tcpip_adapter_lwip.c:1092

D (180797) tcpip_adapter: dhcp client init ip/mask/gw to all-0
D (180797) tcpip_adapter: if0 start ip lost tmr: enter
D (180807) tcpip_adapter: if0 start ip lost tmr: already started
D (180807) tcpip_adapter: dhcp client start successfully
V (180817) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (180827) tcpip_adapter: check: remote, if=0 fn=0x400e64f4
0x400e64f4: tcpip_adapter_dhcpc_start_api at C:/msys32/home/ati/esp/esp-idf/components/tcpip_
adapter/tcpip_adapter_lwip.c:1092

V (180827) event: exit default callback
D (180927) sleep: RTC_PERIPH: AUTO(OFF), RTC_SLOW_MEM: ON, RTC_FAST_MEM: ON
D (180967) phy_init: wifi mac time delta: 65735
D (181017) phy_init: wifi mac time delta: 35673
D (181077) phy_init: wifi mac time delta: 6217
D (181087) sleep: RTC_PERIPH: AUTO(OFF), RTC_SLOW_MEM: ON, RTC_FAST_MEM: ON

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3216923/sdkconfig.txt)

negativekelvin, 2019-05-24:
**wifi mac time delta: 31460303**
Seems like a big delta

gal1990, 2019-05-27:
@negativekelvin  Do you have any idea what could be the problem? Why is the time delta big?

liuzfesp, 2019-11-15:
Hi @gal1990 we will fix the delta time too big issue soon. But I'm not sure it's related to WiFi disconnects.  Does the WiFi disconnects still exist?

liuzfesp, 2019-12-31:
Hi @gal1990 could you check whether this issue still exists?

gal1990, 2020-01-05:
Hi @liuzfesp we finished that project and I can't test if the issue still exists. Thank you for your answer!

Alvin1Zhang, 2020-01-20:
@gal1990 Thanks for sharing the updates.

closed 2020-01-20 by Alvin1Zhang

awilkie1, 2020-02-10:
Hi, i believe I'm having a similar issue to this. i have over 700 devices connected, over a few hours devices slowly disconnect and cannot reconnect. I believe its because the device is entering sleep mode but would like some confirmation.

ESP_ERROR_CHECK( esp_wifi_set_ps(WIFI_PS_NONE) );