espressif/esp-idf #3538: OTA Native Not Working

filzek, 2019-05-24:
Trying to run the Native OTA but the follow problem happen?

I (7006) native_ota_example: SHA-256 for the partition table: : ff5cefa5a144e3e5c89541eb1db05b7d725e18
5fb767e5e2ceb8095b8400fc8c
I (7034) native_ota_example: SHA-256 for bootloader: : c8006bc5c08870b73e36bbdfd5f42c57c20a2b19e37e7c5
ad84e40b6b3a8eb1a
I (7717) native_ota_example: SHA-256 for current firmware: : 89abdf7939c24f8355f1d552fc70b7857e0f002f2
7e271aa7cb2baf6ef71aa01
I (7719) native_ota_example: Starting OTA example...
I (7725) native_ota_example: Running partition type 0 subtype 16 (offset 0x00010000)
I (7733) native_ota_example: Connect to Wifi ! Start to Connect to Server....
I (9846) native_ota_example: Writing to partition subtype 17 at offset 0x190000
I (9847) native_ota_example: New firmware version: 16300003
I (9850) native_ota_example: Running firmware version: 16300002
I (14828) native_ota_example: esp_ota_begin succeeded
I (14857) native_ota_example: Connection closed,all data received
I (14857) native_ota_example: Total Write binary data length : 8598
I (14860) esp_image: segment 0: paddr=0x00190020 vaddr=0x3f400020 size=0x2c358 (181080) map
E (14997) esp_image: invalid segment length 0xffffffff
E (14997) native_ota_example: esp_ota_end failed!
E (14999) native_ota_example: Exiting task due to fatal error...

Got this error:  invalid segment length 0xffffffff

Any ideas why this behavior on the example?

filzek, 2019-05-24:
Find out that a OPENSSL running in Windows will break the operation, so, is a Windows problem with OPENSSL, running in a IIS will work fine.

closed 2019-05-24 by filzek

lakshmisivareddy, 2019-10-15:
Hi filzek 
i am using below command to run the Http server in windows 
winpty openssl s_server -WWW -key ca_key.pem -cert ca_cert.pem -port 8070
i am facing the same issue: invalid segment length 0xffffffff 
can you please provide the steps for running in iis