espressif/esp-idf #3544: ESP-IDF preemptive multthreading by std::thread for arduino-esp32 is NOT preemptive because of cooperative FreeRTOS scheduler

dsyleixa, 2019-05-25:
from a discussion at https://github.com/espressif/arduino-esp32/issues/2814#issuecomment-495917869  it turned out that ESP-IDF preemptive multthreading  by std::thread for arduino-esp32 is NOT preemptive because of a just cooperative FreeRTOS scheduler (IIUC).
Nonetheless, a preemptive multithreading is crucial.

I encounterd that issue by creating  2 threads with 1 of them having no sleeps in between.
But std::thread is preemptive by design, so a scheduler has to split threads into equal timeslices even without sleeps  in between (just for cooperative MT sleeps or yields have to be inserted).
But even when decreasing both thread prios to 1, always the maun loop() is totally blocked.

I first assumed it was an arduino-esp32 issue, but a contributor (atanisoft) stated:
> The FreeRTOS task scheduler is not a preemptive scheduler, it is a co-operative scheduler.
> Unless you switch to using arduino-esp32 as an IDF component you will have no control over the priority of the loopTask.
> arduino-esp32 does not provide ANY part of the scheduler, pthread, std::thread, etc. All of that is coming from ESP-IDF. 
> Feel free to open a bug against the ESP-IDF project for their consideration at implementing what you feel is missing there.

So IIUC, there is already a layer which is capable of preemptive MT, but not available for arduino-esp32
that is why I am now forwarding this issue to this repo, to get preemptive MT also available for arduino-esp32.

igrr, 2019-05-27:
Hi @dsyleixa,

FreeRTOS scheduler in ESP-IDF is configured with preemption enabled. This is also the case in Arduino-esp32. There are a few quirks to task switching behavior related to the multicore support (https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/freertos-smp.html#overview) but based on your code, you don't seem to be running into any of those.

The code sample you are running (from https://github.com/espressif/arduino-esp32/issues/2814#issuecomment-495917869), has a small bug which prevents you from getting the result you want.

Specifically, this part:
```c++
  esp_pthread_cfg_t cfg;
  esp_pthread_get_cfg(&cfg);
  cfg.prio=1;
  esp_pthread_set_cfg(&cfg);
```

Here, `esp_pthread_get_cfg` returns `ESP_ERR_NOT_FOUND`, because no esp_pthread configuration has been set yet. `esp_pthread_get_cfg` zero-fills the `cfg` structure before returning. Then you pass `cfg` to `esp_pthread_set_cfg`, which returns `ESP_ERR_INVALID_ARG` because `cfg.stack_size` is smaller than `CONFIG_ESP32_PTHREAD_STACK_MIN` (768 bytes in the default configuration).

Hence, `esp_pthread_set_cfg` has no effect, and the tasks get launched with the default configuration, set in menuconfig. In particular, default priority (`CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT`) is 5, so both your tasks have strictly higher priority than the Arduino task. Since the fibbonacci_loop doesn't yield, the lower priority Arduino task never runs.

To fix the issue, please check the return values of the functions you call, like this:
```c++
  esp_pthread_cfg_t cfg;
  if (esp_pthread_get_cfg(&cfg) != ESP_OK) {
      cfg = esp_pthread_get_default_config();
  }
  cfg.prio=1;
  if (esp_pthread_set_cfg(&cfg) != ESP_OK) {
      printf("esp_pthread_set_cfg failed\n");
      abort();
  };
```

With this change to your example code, I do see in the log that the loop function runs:
```
Fibbonacci of 30=832040
Fibbonacci of 31=1346269
blinker_loop (LOW) counter: 0
Fibbonacci of 32=2178309
blinker_loop (HIGH) counter: 1
Fibbonacci of 33=3524578
blinker_loop (LOW) counter: 1
blinker_loop (HIGH) counter: 2
Fibbonacci of 34=5702887
blinker_loop (LOW) counter: 2
main loop: 2
blinker_loop (HIGH) counter: 3
```

dsyleixa, 2019-05-27:
hi igrr,
thank you very much, that sounds very promising!

I tried to update my code, setup() now looks like this:

```
void setup() {
  Serial.begin(115200);
  //debug
  delay(1000);
  
  esp_pthread_cfg_t cfg;
  if (esp_pthread_get_cfg(&cfg) != ESP_OK) {
      cfg = esp_pthread_get_default_config();
  }
  cfg.prio=1;
  if (esp_pthread_set_cfg(&cfg) != ESP_OK) {
      printf("esp_pthread_set_cfg failed\n");
      abort();
  };
  
  thread_1 = new std::thread(blinker_loop);
  thread_2 = new std::thread(fibonacci_loop);
}
```

Unfortunately it cannot be compiled of some reason:

```
Sketch wird kompiliert...
"D:\\arduino\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/config" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_trace" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_update" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/asio" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bootloader_support" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bt" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/coap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/console" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/driver" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp-tls" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp32" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_adc_cal" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_event" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_http_client" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_http_server" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_https_ota" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_ringbuf" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/ethernet" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/expat" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/fatfs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/freemodbus" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/freertos" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/heap" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/idf_test" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/jsmn" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/json" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/libsodium" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/log" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/lwip" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mbedtls" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mdns" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/micro-ecc" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mqtt" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/newlib" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/nghttp" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/nvs_flash" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/openssl" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/protobuf-c" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/protocomm" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/pthread" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/sdmmc" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/smartconfig_ack" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/soc" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/spi_flash" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/spiffs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/tcp_transport" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/tcpip_adapter" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/ulp" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/vfs" "-ID:\\arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/wear_levelling [... cut ...] sers\\hw\\AppData\\Local\\Temp\\arduino_build_644293\\sketch\\std_thread_005.ino.cpp.o"
C:\Users\hw\AppData\Local\Temp\arduino_modified_sketch_894747\std_thread_005.ino: In function 'void setup()':

std_thread_005:74:44: error: 'esp_pthread_get_default_config' was not declared in this scope

       cfg = esp_pthread_get_default_config();

                                            ^

exit status 1
'esp_pthread_get_default_config' was not declared in this scope
```

what am I missing eventually?

dsyleixa, 2019-05-28:
is there an extra iclude needed for
esp_pthread_get_default_config
:?:

dsyleixa, 2019-05-28:
is there an extra #include needed for
esp_pthread_get_default_config
:?:

or how can the compile error 
```
exit status 1
'esp_pthread_get_default_config' was not declared in this scope
```
be fixed?

dsyleixa, 2019-05-29:
@igrr:
can you please explain what you did exactly to get the code compilied? I tried several things, but I always get the error
```
exit status 1
'esp_pthread_get_default_config' was not declared in this scope
```

igrr, 2019-05-29:
@dsyleixa I think Arduino-esp32 currently uses ESP-IDF v3.2, but this function was introduced in a later version (which is not released yet, i.e. v3.3 or v4.0). For me the code compiled because I was using Arduino-esp32 as a ESP-IDF component.

You can probably initialize fields of the esp_pthread_cfg_t structure and set them to the desired values, avoiding the call to esp_pthread_get_default_config and simplifying the code somewhat.

dsyleixa, 2019-05-29:
@igrr,
thank you for that hint!
Admittedly I am far too noob to know how to to that correctly because it does not exist from the start at all, IIUC... :-/

Alvin1Zhang, 2021-03-02:
@dsyleixa Thanks for reporting, has your issue resolved? Thanks.

dsyleixa, 2021-03-03:
hello,
unfortunately the ESP-IDF preemptive multthreading syntax is too different from std::thread and the Posix pthread API syntax and comparatively far too complicated to implement for my own programs that I was not able to use it with satisfactory success :(
IMO either C or C++ MT lib for Arduino boards should strictly mimic the standard C/C++ API lib functions.

closed 2021-07-20 by igrr