espressif/esp-idf #3545: "idf.py menuconfig" fails while "$ idf.py build" and "$ idf.py -p COM8 flash" work

jcabad100, 2019-05-25:
## Environment

- Development Kit:      ESP32-Wrover-Kit 
- Kit version: V4.1
- Module or chip used: ESP32-WROVER (?)
- IDF version (run ``git describe --tags`` to find it): 
    v4.0-dev-562-g2b301f53e
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
  xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0

- Operating System:     Windows 10
- Power Supply:         USB

## Problem Description

Tring to build any of the toolchain examples provided with "idf.py menuconfig" FAILS while "$ idf.py build" and "$ idf.py -p COM8 flash" work

### Expected Behavior
The menuconfig dialog should be build
### Actual Behavior
The build process fail with message "ninja: build stopped: subcommand failed.
ninja failed with exit code 1"

### Steps to repropduce

1. L:aunch MINGW64 Gitbash
2. cd till you reach the given project main directory e.g. ~/esp/esp-idf/examples/get-started/hello_world 
3. type "idf.py menuconfig"
// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

The code is the original toolchain example from ...esp\esp-idf\examples\get-started\hello_world\main\hello_world_main.c

## Debug Logs

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf/examples/get-started/hello_world (master)
$ idf.py menuconfig
Checking Python dependencies...
Python requirements from C:\Users\DELL\esp\esp-idf\requirements.txt are satisfie
d.
Running ninja in directory c:\users\dell\esp\esp-idf\examples\get-started\hello_
world\build
Executing "ninja menuconfig"...
[0/1] cmd.exe /C "cd /D C:\Users\DELL\...les/get-started/hello_world/sdkconfig"
/usr/bin/winpty: error: cannot start 'C:/Users/DELL/esp/esp-idf/Kconfig': %1 is
not a valid Win32 application. (error 0xc1)
FAILED: CMakeFiles/menuconfig
cmd.exe /C "cd /D C:\Users\DELL\esp\esp-idf\examples\get-started\hello_world\bui
ld && D:\Phyton\Python27\python.exe C:/Users/DELL/esp/esp-idf/tools/kconfig_new/
confgen.py --kconfig C:/Users/DELL/esp/esp-idf/Kconfig --config C:/Users/DELL/es
p/esp-idf/examples/get-started/hello_world/sdkconfig --env "COMPONENT_KCONFIGS=C
:/Users/DELL/esp/esp-idf/components/app_trace/Kconfig C:/Users/DELL/esp/esp-idf/
components/bt/Kconfig C:/Users/DELL/esp/esp-idf/components/driver/Kconfig C:/Use
rs/DELL/esp/esp-idf/components/efuse/Kconfig C:/Users/DELL/esp/esp-idf/component
s/esp32/Kconfig C:/Users/DELL/esp/esp-idf/components/esp_adc_cal/Kconfig C:/User
s/DELL/esp/esp-idf/components/esp_event/Kconfig C:/Users/DELL/esp/esp-idf/compon
ents/esp_http_client/Kconfig C:/Users/DELL/esp/esp-idf/components/esp_http_serve
r/Kconfig C:/Users/DELL/esp/esp-idf/components/esp_https_ota/Kconfig C:/Users/DE
LL/esp/esp-idf/components/esp_wifi/Kconfig C:/Users/DELL/esp/esp-idf/components/
espcoredump/Kconfig C:/Users/DELL/esp/esp-idf/components/ethernet/Kconfig C:/Use
rs/DELL/esp/esp-idf/components/fatfs/Kconfig C:/Users/DELL/esp/esp-idf/component
s/freemodbus/Kconfig C:/Users/DELL/esp/esp-idf/components/freertos/Kconfig C:/Us
ers/DELL/esp/esp-idf/components/heap/Kconfig C:/Users/DELL/esp/esp-idf/component
s/libsodium/Kconfig C:/Users/DELL/esp/esp-idf/components/log/Kconfig C:/Users/DE
LL/esp/esp-idf/components/lwip/Kconfig C:/Users/DELL/esp/esp-idf/components/mbed
tls/Kconfig C:/Users/DELL/esp/esp-idf/components/mdns/Kconfig C:/Users/DELL/esp/
esp-idf/components/mqtt/Kconfig C:/Users/DELL/esp/esp-idf/components/nvs_flash/K
config C:/Users/DELL/esp/esp-idf/components/openssl/Kconfig C:/Users/DELL/esp/es
[... 55 lines cut ...]
esp/esp-idf/components/mbedtls/Kconfig C:/Users/DELL/esp/esp-idf/components/mdns
/Kconfig C:/Users/DELL/esp/esp-idf/components/mqtt/Kconfig C:/Users/DELL/esp/esp
-idf/components/nvs_flash/Kconfig C:/Users/DELL/esp/esp-idf/components/openssl/K
config C:/Users/DELL/esp/esp-idf/components/pthread/Kconfig C:/Users/DELL/esp/es
p-idf/components/spi_flash/Kconfig C:/Users/DELL/esp/esp-idf/components/spiffs/K
config C:/Users/DELL/esp/esp-idf/components/tcpip_adapter/Kconfig C:/Users/DELL/
esp/esp-idf/components/unity/Kconfig C:/Users/DELL/esp/esp-idf/components/vfs/Kc
onfig C:/Users/DELL/esp/esp-idf/components/wear_levelling/Kconfig" --env "COMPON
ENT_KCONFIGS_PROJBUILD=C:/Users/DELL/esp/esp-idf/components/app_update/Kconfig.p
rojbuild C:/Users/DELL/esp/esp-idf/components/bootloader/Kconfig.projbuild C:/Us
ers/DELL/esp/esp-idf/components/esptool_py/Kconfig.projbuild C:/Users/DELL/esp/e
sp-idf/components/partition_table/Kconfig.projbuild" --env IDF_CMAKE=y --env IDF
_TARGET=esp32 --env IDF_TARGET=esp32 --output config C:/Users/DELL/esp/esp-idf/e
xamples/get-started/hello_world/sdkconfig"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder) -> ORIGINAL from toolchain 
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3219666/sdkconfig.zip)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
[hello-world.zip](https://github.com/espressif/esp-idf/files/3219668/hello-world.zip)

- [ ] coredump (This provides stacks of tasks.)  ->does not apply as application is not built.

jcabad100, 2019-05-27:
@projectgus this is the issue you requested me to create....i actually have multiple issues that prevent me to build most of the examples that come with the tool chain....i hope you can help me fix them.....i will create multiple issues as i encounter them....thank you!

dobairoland, 2019-05-27:
Hello @jcabad100.

Thank you for reporting the issue. Can you please check that the following patch solves your issue?

[0001-Tools-Fix-CMake-menuconfig.txt](https://github.com/espressif/esp-idf/files/3224415/0001-Tools-Fix-CMake-menuconfig.txt)

jcabad100, 2019-05-27:
@dobairoland ..thank you for our reply...i am very new to this dev. environment, please be very specific of where do i have to copy that file or what do i do with it? thanks!

dobairoland, 2019-05-27:
Hi @jcabad100!

Please dowload it into your IDF directory and use the following steps:

```
cd $IDF_PATH
git checkout master
git am < 0001-Tools-Fix-CMake-menuconfig.txt
```

You can restore your `master` after you've tested the patch.
```
git reset --hard origin/master
rm 0001-Tools-Fix-CMake-menuconfig.txt
```

jcabad100, 2019-05-27:
The fix attempt was not sucessful....and the "idf.py build" also didn't succeed this time on the hello_world application, see attached log, thanks.  
[Log of Failed Fix Attempt using 0001-Tools-Fix-CMake-menuconfig.txt](https://github.com/espressif/esp-idf/files/3224512/Log.of.Failed.Fix.Attempt.using.0001-Tools-Fix-CMake-menuconfig.txt)

jcabad100, 2019-05-27:
The attempt seems to have damaged more thinks, the restore attempt looks like this:

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf (master)
$ git reset --hard origin/master
HEAD is now at 2b301f53e Merge branch 'bugfix/fix_build_make_macos' into 'master'

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf (master)
$ rm 0001-Tools-Fix-CMake-menuconfig.txt

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf (master)
$ idf.py build
Checking Python dependencies...
Python requirements from C:\Users\DELL\esp\esp-idf\requirements.txt are satisfied.
Running cmake in directory c:\users\dell\esp\esp-idf\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized C:\Users\DELL\esp\esp-idf"...
Warn about uninitialized values.
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Didn't find assembler
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_ASM_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "C:/Users/DELL/esp/esp-idf/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/DELL/esp/esp-idf/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

jcabad100, 2019-05-27:
The idf.py  bulid fails now too:

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf/examples/get-started/hello_world (master)
$ idf.py build
Checking Python dependencies...
Python requirements from C:\Users\DELL\esp\esp-idf\requirements.txt are satisfied.
Running ninja in directory c:\users\dell\esp\esp-idf\examples\get-started\hello_world\build
Executing "ninja all"...
[0/1] Re-running CMake...
CMake Warning at C:/Users/DELL/esp/esp-idf/tools/cmake/component.cmake:122 (message):
  Component directory C:/Users/DELL/esp/esp-idf/components/esp-idf-lib-master
  does not contain a CMakeLists.txt file.  No component will be added
Call Stack (most recent call first):
  C:/Users/DELL/esp/esp-idf/tools/cmake/build.cmake:150 (__component_add)
  C:/Users/DELL/esp/esp-idf/tools/cmake/idf.cmake:42 (__build_init)
  C:/Users/DELL/esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- Project version: v4.0-dev-562-g2b301f53e
-- Building ESP-IDF components for target esp32
-- Adding linker script C:/Users/DELL/esp/esp-idf/examples/get-started/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-locale.ld
-- Components: ads111x app_trace app_update asio bh1750 bmp180 bmp280 bootloader bootloader_support bt coap console cxx dht driver ds1307 ds18x20 ds3231 efuse esp-tls esp32 esp_adc_cal esp_common esp_
event esp_http_client esp_http_server esp_https_ota esp_https_server esp_ringbuf esp_rom esp_wifi espcoredump esptool_py ethernet expat fatfs freemodbus freertos hd44780 heap hmc5883l i2cdev idf_test
jsmn json libsodium log lwip main max7219 mbedtls mcp23017 mdns mqtt ms5611 newlib nghttp nvs_flash onewire openssl partition_table pca9685 pcf8574 pcf8591 protobuf-c protocomm pthread sdmmc smartconf
ig_ack soc spi_flash spiffs tcp_transport tcpip_adapter tda74xx tsl2561 ulp ultrasonic unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/DELL/esp/esp-idf/components/ads111x C:/Users/DELL/esp/esp-idf/components/app_trace C:/Users/DELL/esp/esp-idf/components/app_update C:/Users/DELL/esp/esp-idf/components/asi
o C:/Users/DELL/esp/esp-idf/components/bh1750 C:/Users/DELL/esp/esp-idf/components/bmp180 C:/Users/DELL/esp/esp-idf/components/bmp280 C:/Users/DELL/esp/esp-idf/components/bootloader C:/Users/DELL/esp/
esp-idf/components/bootloader_support C:/Users/DELL/esp/esp-idf/components/bt C:/Users/DELL/esp/esp-idf/components/coap C:/Users/DELL/esp/esp-idf/components/console C:/Users/DELL/esp/esp-idf/component
s/cxx C:/Users/DELL/esp/esp-idf/components/dht C:/Users/DELL/esp/esp-idf/components/driver C:/Users/DELL/esp/esp-idf/components/ds1307 C:/Users/DELL/esp/esp-idf/components/ds18x20 C:/Users/DELL/esp/es
p-idf/components/ds3231 C:/Users/DELL/esp/esp-idf/components/efuse C:/Users/DELL/esp/esp-idf/components/esp-tls C:/Users/DELL/esp/esp-idf/components/esp32 C:/Users/DELL/esp/esp-idf/components/esp_adc_
cal C:/Users/DELL/esp/esp-idf/components/esp_common C:/Users/DELL/esp/esp-idf/components/esp_event C:/Users/DELL/esp/esp-idf/components/esp_http_client C:/Users/DELL/esp/esp-idf/components/esp_http_se
rver C:/Users/DELL/esp/esp-idf/components/esp_https_ota C:/Users/DELL/esp/esp-idf/components/esp_https_server C:/Users/DELL/esp/esp-idf/components/esp_ringbuf C:/Users/DELL/esp/esp-idf/components/esp_
rom C:/Users/DELL/esp/esp-idf/components/esp_wifi C:/Users/DELL/esp/esp-idf/components/espcoredump C:/Users/DELL/esp/esp-idf/components/esptool_py C:/Users/DELL/esp/esp-idf/components/ethernet C:/User
s/DELL/esp/esp-idf/components/expat C:/Users/DELL/esp/esp-idf/components/fatfs C:/Users/DELL/esp/esp-idf/components/freemodbus C:/Users/DELL/esp/esp-idf/components/freertos C:/Users/DELL/esp/esp-idf/c
omponents/hd44780 C:/Users/DELL/esp/esp-idf/components/heap C:/Users/DELL/esp/esp-idf/components/hmc5883l C:/Users/DELL/esp/esp-idf/components/i2cdev C:/Users/DELL/esp/esp-idf/components/idf_test C:/U
sers/DELL/esp/esp-idf/components/jsmn C:/Users/DELL/esp/esp-idf/components/json C:/Users/DELL/esp/esp-idf/components/libsodium C:/Users/DELL/esp/esp-idf/components/log C:/Users/DELL/esp/esp-idf/compon
ents/lwip C:/Users/DELL/esp/esp-idf/examples/get-started/hello_world/main C:/Users/DELL/esp/esp-idf/components/max7219 C:/Users/DELL/esp/esp-idf/components/mbedtls C:/Users/DELL/esp/esp-idf/components
/mcp23017 C:/Users/DELL/esp/esp-idf/components/mdns C:/Users/DELL/esp/esp-idf/components/mqtt C:/Users/DELL/esp/esp-idf/components/ms5611 C:/Users/DELL/esp/esp-idf/components/newlib C:/Users/DELL/esp/
esp-idf/components/nghttp C:/Users/DELL/esp/esp-idf/components/nvs_flash C:/Users/DELL/esp/esp-idf/components/onewire C:/Users/DELL/esp/esp-idf/components/openssl C:/Users/DELL/esp/esp-idf/components/
[... cut ...]
[45/52] Building C object esp-idf/micro-ecc/CMakeFiles/__idf_micro-ecc.dir/micro-ecc/uECC.c.obj
[46/52] Linking C static library esp-idf\micro-ecc\libmicro-ecc.a
[47/52] Linking C static library esp-idf\bootloader_support\libbootloader_support.a
[48/52] Linking C static library esp-idf\efuse\libefuse.a
[49/52] Linking C static library esp-idf\spi_flash\libspi_flash.a
[50/52] Linking C static library esp-idf\main\libmain.a
[51/52] Linking C executable bootloader.elf
[52/52] Generating .app_hash
esptool.py v2.7-dev
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

jcabad100, 2019-05-27:
So, sorry for my ignorance ...i partly may have understood what have caused this misbehavior....as you see in above logs there are libraries like esp-idf-lib-master ,  hd44780, dht, bmp180 bmp280 etc etc that i added two days ago (trying to carry out examples from a book on IoT & ESP32) from another Github repository by simply copying .under .\esp\esp-idf\components ..so it seems that that "Executing "ninja all"..." try to build all components under the components folder despite those that i added has nothing to do with the hello_world application! .....in fact they are reflected under the folder ..L\esp\esp-idf\examples\get-started\hello_world\build\esp-idf ...that is crazy....i will remove those components from the components folder and try again...

jcabad100, 2019-05-27:
In fact, i deleted all those external libraries and"$ idf.py build" and "$ idf.py -p COM8 flash" work again...but  'idf.py menuconfig" still fails ...i will attempt to apply your patch again...

jcabad100, 2019-05-27:
i reaplied the patch but it still fails..

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf/examples/protocols/mqtt/publish_test (master)
$ cd $IDF_PATH

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf (master)
$ git checkout master
Already on 'master'
M       components/bt/lib
Your branch is up to date with 'origin/master'.

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf (master)
$ git am < 0001-Tools-Fix-CMake-menuconfig.txt
Applying: Tools: Fix CMake menuconfig used with combination of MSYS and CMD

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf (master)
$ cd e*

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf/examples (master)
$ cd g*

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf/examples/get-started (master)
$ cd h*

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf/examples/get-started/hello_world (master)
$ dir
build  CMakeLists.txt  main  Makefile  README.md  sdkconfig  sdkconfig.zip

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf/examples/get-started/hello_world (master)
$ idf.py menuconfig
Checking Python dependencies...
Python requirements from C:\Users\DELL\esp\esp-idf\requirements.txt are satisfied.
Running ninja in directory c:\users\dell\esp\esp-idf\examples\get-started\hello_world\build
Executing "ninja menuconfig"...
[0/1] Re-running CMake...
-- Project version: v4.0-dev-563-gc40b8dd2b
-- Building ESP-IDF components for target esp32
-- Adding linker script C:/Users/DELL/esp/esp-idf/examples/get-started/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/DELL/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-locale.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_event esp_http_client esp_http_server esp_https_ota esp
_https_server esp_ringbuf esp_rom esp_wifi espcoredump esptool_py ethernet expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash ope
nssl partition_table protobuf-c protocomm pthread sdmmc smartconfig_ack soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/DELL/esp/esp-idf/components/app_trace C:/Users/DELL/esp/esp-idf/components/app_update C:/Users/DELL/esp/esp-idf/components/asio C:/Users/DELL/esp/esp-idf/components/bootlo
ader C:/Users/DELL/esp/esp-idf/components/bootloader_support C:/Users/DELL/esp/esp-idf/components/bt C:/Users/DELL/esp/esp-idf/components/coap C:/Users/DELL/esp/esp-idf/components/console C:/Users/DEL
L/esp/esp-idf/components/cxx C:/Users/DELL/esp/esp-idf/components/driver C:/Users/DELL/esp/esp-idf/components/efuse C:/Users/DELL/esp/esp-idf/components/esp-tls C:/Users/DELL/esp/esp-idf/components/es
p32 C:/Users/DELL/esp/esp-idf/components/esp_adc_cal C:/Users/DELL/esp/esp-idf/components/esp_common C:/Users/DELL/esp/esp-idf/components/esp_event C:/Users/DELL/esp/esp-idf/components/esp_http_client
 C:/Users/DELL/esp/esp-idf/components/esp_http_server C:/Users/DELL/esp/esp-idf/components/esp_https_ota C:/Users/DELL/esp/esp-idf/components/esp_https_server C:/Users/DELL/esp/esp-idf/components/esp_
ringbuf C:/Users/DELL/esp/esp-idf/components/esp_rom C:/Users/DELL/esp/esp-idf/components/esp_wifi C:/Users/DELL/esp/esp-idf/components/espcoredump C:/Users/DELL/esp/esp-idf/components/esptool_py C:/U
sers/DELL/esp/esp-idf/components/ethernet C:/Users/DELL/esp/esp-idf/components/expat C:/Users/DELL/esp/esp-idf/components/fatfs C:/Users/DELL/esp/esp-idf/components/freemodbus C:/Users/DELL/esp/esp-id
f/components/freertos C:/Users/DELL/esp/esp-idf/components/heap C:/Users/DELL/esp/esp-idf/components/idf_test C:/Users/DELL/esp/esp-idf/components/jsmn C:/Users/DELL/esp/esp-idf/components/json C:/Use
rs/DELL/esp/esp-idf/components/libsodium C:/Users/DELL/esp/esp-idf/components/log C:/Users/DELL/esp/esp-idf/components/lwip C:/Users/DELL/esp/esp-idf/examples/get-started/hello_world/main C:/Users/DEL
L/esp/esp-idf/components/mbedtls C:/Users/DELL/esp/esp-idf/components/mdns C:/Users/DELL/esp/esp-idf/components/mqtt C:/Users/DELL/esp/esp-idf/components/newlib C:/Users/DELL/esp/esp-idf/components/ng
http C:/Users/DELL/esp/esp-idf/components/nvs_flash C:/Users/DELL/esp/esp-idf/components/openssl C:/Users/DELL/esp/esp-idf/components/partition_table C:/Users/DELL/esp/esp-idf/components/protobuf-c C:
/Users/DELL/esp/esp-idf/components/protocomm C:/Users/DELL/esp/esp-idf/components/pthread C:/Users/DELL/esp/esp-idf/components/sdmmc C:/Users/DELL/esp/esp-idf/components/smartconfig_ack C:/Users/DELL/
esp/esp-idf/components/soc C:/Users/DELL/esp/esp-idf/components/spi_flash C:/Users/DELL/esp/esp-idf/components/spiffs C:/Users/DELL/esp/esp-idf/components/tcp_transport C:/Users/DELL/esp/esp-idf/compo
[... cut ...]
Kconfig C:/Users/DELL/esp/esp-idf/components/unity/Kconfig C:/Users/DELL/esp/esp-idf/components/vfs/Kconfig C:/Users/DELL/esp/esp-idf/components/wear_levelling/Kconfig" --env "COMPONENT_KCONFIGS_PROJB
UILD=C:/Users/DELL/esp/esp-idf/components/app_update/Kconfig.projbuild C:/Users/DELL/esp/esp-idf/components/bootloader/Kconfig.projbuild C:/Users/DELL/esp/esp-idf/components/esptool_py/Kconfig.projbui
ld C:/Users/DELL/esp/esp-idf/components/partition_table/Kconfig.projbuild" --env IDF_CMAKE=y --env IDF_TARGET=esp32 --env IDF_TARGET=esp32 --output config C:/Users/DELL/esp/esp-idf/examples/get-starte
d/hello_world/sdkconfig"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

dobairoland, 2019-05-28:
Hi @jcabad100.

Thank you for your feedback. It seems to me that the original issue was solved and you've encountered another one. I'm not sure how to solve this, therefore, I would like to ask you to open another issue which can be handled by my colleague. I will notify him.

Please provide the following information when opening a new issue. Please remove your `build` directory. After that please run `idf.py -v menuconfig` and provide the output log. Please ZIP the content of the `build` directory and upload it. Thank you very much!

dobairoland, 2019-05-29:
Hi @jcabad100!

Maybe it won't be necessary to open another issue because with some other feedback I was able to proceed. Here is another patch which should solve the issue:

[0001cmakemenuconfigfix.txt](https://github.com/espressif/esp-idf/files/3232370/0001cmakemenuconfigfix.txt)

Can you please try it? Note that this must be applied to the clean master without the previous patch.

jcabad100, 2019-05-29:
..i just see your message...ok...i will try it asap...thx..

jcabad100, 2019-05-29:
The process seems to succeed and the configuration menu is launched but it appearance and behavior is totally messy....visually and functionally....check attachment...and as i try to navigate the different menu buttons it gets even more messy and nasty...nevrtheless i manage to exit .... 
[image]

dobairoland, 2019-05-29:
I'm happy that the issue got solved.

This is something unrelated. Have you tried make the window bigger and/or changing the settings (fonts) of the terminal?

jcabad100, 2019-05-29:
...i dont know if that is unrelated...at the very beginning (or with previous version?) that configuration menu was ok ...i just tried to enlarge the window but it was worst, showing the previous logs on top...totally messy and therefore useless.....anyhow i still can build and flash with the other two commands. Thanks.  
QUESTION: my next problem is how to properly import external components library https://github.com/UncleRus/esp-idf-lib ...i dont manage to use that ...i have similar problems like the one reported from someone else: https://github.com/UncleRus/esp-idf-lib/issues/13 ....can you please point to a document that would indicate how to do it properly or what is missing?

xanthome, 2019-07-02:
I have a very similar issue to this (on windows 10 using git bash)
[esp-idf-build-success.txt](https://github.com/espressif/esp-idf/files/3347815/esp-idf-build-success.txt)
:
$ git describe --tags
v4.0-dev-1093-gbca4afb46

$ echo $IDF_PATH
C:\dev\esp\esp-idf

$ idf.py -v menuconfig
Checking Python dependencies...
Python requirements from C:\dev\esp\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running ninja in directory c:\dev\esp\hello_world\build
Executing "ninja -v menuconfig"...
[0/1] cmd.exe /C "cd /D C:\dev\esp\hello_world\build && C:\Python27\python.exe C:/dev/esp/esp-idf/tools/kconfig_new/confgen.py --kconfig C:/dev/esp/esp-idf/Kconfig --config C:/dev
/esp/hello_world/sdkconfig --env-file C:/dev/esp/hello_world/build/config.env --env IDF_TARGET=esp32 --output config C:/dev/esp/hello_world/sdkconfig && "C:\Program Files\CMake\bi
n\cmake.exe" -E env "COMPONENT_KCONFIGS=C:/dev/esp/esp-idf/components/app_trace/Kconfig C:/dev/esp/esp-idf/components/bt/Kconfig C:/dev/esp/esp-idf/components/driver/Kconfig C:/de
v/esp/esp-idf/components/efuse/Kconfig C:/dev/esp/esp-idf/components/esp-tls/Kconfig C:/dev/esp/esp-idf/components/esp32/Kconfig C:/dev/esp/esp-idf/components/esp_adc_cal/Kconfig
C:/dev/esp/esp-idf/components/esp_common/Kconfig C:/dev/esp/esp-idf/components/esp_eth/Kconfig C:/dev/esp/esp-idf/components/esp_event/Kconfig C:/dev/esp/esp-idf/components/esp_ht
tp_client/Kconfig C:/dev/esp/esp-idf/components/esp_http_server/Kconfig C:/dev/esp/esp-idf/components/esp_https_ota/Kconfig C:/dev/esp/esp-idf/components/esp_https_server/Kconfig
C:/dev/esp/esp-idf/components/esp_wifi/Kconfig C:/dev/esp/esp-idf/components/espcoredump/Kconfig C:/dev/esp/esp-idf/components/fatfs/Kconfig C:/dev/esp/esp-idf/components/freemodb
us/Kconfig C:/dev/esp/esp-idf/components/freertos/Kconfig C:/dev/esp/esp-idf/components/heap/Kconfig C:/dev/esp/esp-idf/components/libsodium/Kconfig C:/dev/esp/esp-idf/components/
log/Kconfig C:/dev/esp/esp-idf/components/lwip/Kconfig C:/dev/esp/esp-idf/components/mbedtls/Kconfig C:/dev/esp/esp-idf/components/mdns/Kconfig C:/dev/esp/esp-idf/components/mqtt/
Kconfig C:/dev/esp/esp-idf/components/newlib/Kconfig C:/dev/esp/esp-idf/components/nvs_flash/Kconfig C:/dev/esp/esp-idf/components/openssl/Kconfig C:/dev/esp/esp-idf/components/pt
hread/Kconfig C:/dev/esp/esp-idf/components/spi_flash/Kconfig C:/dev/esp/esp-idf/components/spiffs/Kconfig C:/dev/esp/esp-idf/components/tcpip_adapter/Kconfig C:/dev/esp/esp-idf/c
omponents/unity/Kconfig C:/dev/esp/esp-idf/components/vfs/Kconfig C:/dev/esp/esp-idf/components/wear_levelling/Kconfig C:/dev/esp/esp-idf/components/wifi_provisioning/Kconfig C:/d
ev/esp/esp-idf/components/wpa_supplicant/Kconfig" "COMPONENT_KCONFIGS_PROJBUILD=C:/dev/esp/esp-idf/components/app_update/Kconfig.projbuild C:/dev/esp/esp-idf/components/bootloader
/Kconfig.projbuild C:/dev/esp/esp-idf/components/esptool_py/Kconfig.projbuild C:/dev/esp/esp-idf/components/partition_table/Kconfig.projbuild" IDF_CMAKE=y KCONFIG_CONFIG=C:/dev/es
p/hello_world/sdkconfig IDF_TARGET=esp32 "C:/Program Files/Git/usr/bin/winpty.exe" "C:/Program Files/Espressif/ESP-IDF Tools/mconf-idf/mconf-idf.exe" C:/dev/esp/esp-idf/Kconfig &&
 C:\Python27\python.exe C:/dev/esp/esp-idf/tools/kconfig_new/confgen.py --kconfig C:/dev/esp/esp-idf/Kconfig --config C:/dev/esp/hello_world/sdkconfig --env-file C:/dev/esp/hello_
world/build/config.env --env IDF_TARGET=esp32 --output config C:/dev/esp/hello_world/sdkconfig"
C:/dev/esp/esp-idf/components/bt/Kconfig:386: undefined variable "$IDF_PATH/components/bt/host/bluedroid/Kconfig.in"
FAILED: CMakeFiles/menuconfig

dobairoland, 2019-07-02:
The issue is solved by https://github.com/espressif/esp-idf/commit/5034a57605ba22241abad989d38120a241607eb2.

@xanthome, the issue reported by you is a different one but it was already fixed maybe by this https://github.com/espressif/esp-idf/commit/bd699985169ecf44cd957e3832077c79b180760e commit (I'm not sure).

closed 2019-07-02 by dobairoland

xanthome, 2019-07-02:
Thanks, I verified I had those changes already.

I was using esp-idf-tools-setup-1.2.exe but upgrading to esp-idf-tools-setup-2.0.exe resolved the menuconfig issue for me.

avanbremen, 2019-07-03:
I just grabbed the latest esp-idf (v4.0-dev-1095-g16014079f) and can confirm the same issue as reported by @xanthome when running idf.py menuconfig.

`C:/Users/Arjan/esp/esp-idf/components/bt/Kconfig:386: undefined variable "$IDF_PATH/components/bt/host/bluedroid/Kconfig.in"`

When manually replacing `$IDF_PATH` on lines 386, 391 and 404 in Kconfig with the actual IDF_PATH, idf.py menuconfig runs just fine.

dobairoland, 2019-07-03:
`menuconfig` fails because `mconf-idf` cannot handle the `source "$IDF_PATH...` line in the Kconfig file. `mconf-idf` for Windows is in ESP IDF Tools installer (https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/windows-setup.html#esp-idf-tools-installer).  So please update it. 

Sorry for the inconvenience.

avanbremen, 2019-07-03:
@dobairoland Thank you very much for your reply. I just removed ESP IDF Tools v1.2 and tried installing v2.0. Unfortunately the installation fails with a PermissionError (see log below). Also, I noticed that when changing the install directory during the setup process, the installer will still copy data to the proposed location (`C:\Users\<user>\.espressif`). Do you know of anything I can try to resolve this issue?

```
                          Extracting C:\Users\Arjan\.espressif\dist\cmake-3.13.4-win64-x64.zip to C:\Users\Arjan\.espressif\tools\cmake\3.13.4
                          Traceback (most recent call last):
                            File "C:\Users\Arjan\esp\esp-idf\tools\idf_tools.py", line 1249, in <module>
                              main(sys.argv[1:])
                            File "C:\Users\Arjan\esp\esp-idf\tools\idf_tools.py", line 1245, in main
                              action_func(args)
                            File "C:\Users\Arjan\esp\esp-idf\tools\idf_tools.py", line 1048, in action_install
                              tool_obj.install(tool_version)
                            File "C:\Users\Arjan\esp\esp-idf\tools\idf_tools.py", line 554, in install
                              strip_container_dirs(dest_dir, self._current_options.strip_container_dirs)
                            File "C:\Users\Arjan\esp\esp-idf\tools\idf_tools.py", line 261, in strip_container_dirs
                              os.rename(path, tmp_path)
                          PermissionError: [WinError 5] Toegang geweigerd: 'C:\\Users\\Arjan\\.espressif\\tools\\cmake\\3.13.4' -> 'C:\\Users\\Arjan\\.espressif\\tools\\cmake\\3.13.4.tmp'
```

igrr, 2019-07-03:
When the installer fails, does it offer to view the installation log? If yes, could you please attach the complete log here?

avanbremen, 2019-07-03:
@dobairoland I tried running the installer with different Python versions, 3.6 (32-bit), 3.7 (64-bit) and 2.7. All failed. I copied the traceback of the most recent attempt below. "Toegang geweigerd" means "Access denied". I run the installer in administrator mode.

```
2019-07-03 14:01:27.523   Running command: C:\Python27\\python.exe C:\Users\Arjan\esp\esp-idf\tools\idf_tools.py install
                          Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
                          Installing xtensa-esp32-elf@esp32-2019r1-8.2.0
                          file xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-win32.zip is already downloaded
                          Extracting C:\Users\Arjan\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-win32.zip to C:\Users\Arjan\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0
                          Installing esp32ulp-elf@2.28.51.20170517
                          file binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip is already downloaded
                          Extracting C:\Users\Arjan\.espressif\dist\binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip to C:\Users\Arjan\.espressif\tools\esp32ulp-elf\2.28.51.20170517
                          Installing cmake@3.13.4
                          file cmake-3.13.4-win64-x64.zip is already downloaded
                          Extracting C:\Users\Arjan\.espressif\dist\cmake-3.13.4-win64-x64.zip to C:\Users\Arjan\.espressif\tools\cmake\3.13.4
                          Traceback (most recent call last):
                            File "C:\Users\Arjan\esp\esp-idf\tools\idf_tools.py", line 1249, in <module>
                              main(sys.argv[1:])
                            File "C:\Users\Arjan\esp\esp-idf\tools\idf_tools.py", line 1245, in main
                              action_func(args)
                            File "C:\Users\Arjan\esp\esp-idf\tools\idf_tools.py", line 1048, in action_install
                              tool_obj.install(tool_version)
                            File "C:\Users\Arjan\esp\esp-idf\tools\idf_tools.py", line 554, in install
                              strip_container_dirs(dest_dir, self._current_options.strip_container_dirs)
                            File "C:\Users\Arjan\esp\esp-idf\tools\idf_tools.py", line 261, in strip_container_dirs
                              os.rename(path, tmp_path)
                          WindowsError: [Error 5] Toegang geweigerd
```

avanbremen, 2019-07-03:
@dobairoland I just ran CCleaner to cleanup the registry. The installation now succeeded, but when I try to run idf.py menuconfig I get the following error. Do you know what should be added to PATH? Thanks.

```
C:\Users\Arjan\esp\esp-test>idf.py menuconfig
Checking Python dependencies...
Python requirements from C:\Users\Arjan\esp\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
To use C:\Users\Arjan\esp\esp-idf\tools\idf.py, either the 'ninja' or 'GNU make' build tool must be available in the PATH
```

igrr, 2019-07-03:
@avanbremen Interesting, thanks. If the issue related to os.rename shows up again, please let us know and will try to find the cause.

Can you please run `echo %PATH%` in the command line where you receive the most recent error message? Also, is this the cmd.exe window opened from the Start menu shortcut which the installer has created, or some other cmd.exe window? In the latter case, did you run `export.bat` in IDF_PATH before trying to build the project?

avanbremen, 2019-07-03:
@igrr Thanks for your reply. I did not run export.bat in IDF_PATH before trying to build the project, thanks for the tip (I guess the installer normally runs export.bat, correct?). Instead I tried rebuilding the System Path variable based on the errors I received when building, which resulted in the following additions:

`;C:\Users\Arjan\.espressif\tools\ninja\1.9.0;C:\Users\Arjan\.espressif\tools\cmake\3.13.4\bin;C:\Users\Arjan\.espressif\tools\mconf\v4.6.0.0-idf-20190628;C:\Users\Arjan\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin`

At the moment I'm able to build my project, but I get the following error:

`xtensa-esp32-elf-gcc.exe: error: DEBUG=1: No such file or directory`

This definition comes from the main CMakeLists.txt file, which contains (among other things):

```
# Add preprocessor definitions.
add_definitions(-DDEBUG=1)
```

This never caused issues with the previous esp-idf and tools. Do you have any clue what might be causing this?

igrr, 2019-07-03:
I would not recommend adding the tools to system PATH variable, as different IDF versions will require different versions of tools. `export.bat` is supposed to solve this, giving you the correct paths depending on the IDF version. You can also use the "IDF Command Prompt (cmd.exe)" shortcut which the installer creates in the Start Menu. This will open cmd.exe and run the equivalent of export.bat there, preparing the environment variables inside this command prompt session.

Regarding the add_definitions issue, this could be related to a different CMake version than the one you have used before. I would suggest trying to use target_compile_definitions function instead, maybe that will work? @renzbagaporo might also be able to help with this one.

avanbremen, 2019-07-03:
@igrr Thanks for your reply. After you suggested using export.bat I removed the manual additions from System Path.

Chaning `add_definitions` to `add_compile_options` did the trick. Apparently `add_definitions` has been superseded by alternatives ([link](https://cmake.org/cmake/help/v3.15/command/add_definitions.html)).

Thanks for your help.

yuanyanhui, 2019-08-30:
@jcabad100 Have you solved the issue of messy menuconfig GUI. I am having the same problem.

yuanyanhui, 2019-08-30:
I changed my PC's language and region settings to English. The initial popup window of menuconfig looks normal, however when you max the window everything appears out of place. The GUI does not go back to its clean state if you change the window back to its initial size.

FJ-VEL, 2021-01-21:
Hi to everyone,

maybe is a silly contribution but I found this thread  searching the solution for the next problem. 
[image]

FJ-VEL, 2021-01-21:
which more or less is similar to above comments, the solution in my case was really easy.
I sent the order to programming flash but I did not press the boot button on the PCB. So after a while the system shown me the above fail.