espressif/esp-idf #3546: Multiple tool-chain examples do not build

jcabad100, 2019-05-25:
## Environment

- Development Kit:      ESP32-Wrover-Kit 
- Kit version: V4.1
- Module or chip used: ESP32-WROVER (?)
- IDF version (run ``git describe --tags`` to find it): 
    v4.0-dev-562-g2b301f53e
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
  xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0

- Operating System:     Windows 10
- Power Supply:         USB

## Problem Description

As reported in separated issue the "idf.py menuconfig" FAILS and therefore i found a workaround to successfully build and flash the toolchain most simple examples i.e. "hello_world" and "blink" ( using "$ idf.py build" and then "$ idf.py -p COM8 flash" work); nevertheless more complex examples using other libraries fail e..g I2C and I2S 

### Expected Behavior
The toolchain provided and unmodified examples are expected to work / be built without problem.
### Actual Behavior
The toolchain provided and unmodified examples fail to build. 

### Steps to repropduce

1. L:aunch MINGW64 Gitbash
2. cd till you reach the given project main directory e.g.  ~/esp/esp-idf/examples/peripherals/i2c/i2c_self_test
3. type "$ idf.py build" 

### Code to reproduce this issue

The code is the original toolchain example from ...C:\Users\DELL\esp\esp-idf\examples\peripherals\i2c\i2c_self_test\main\i2c_example_main.c

## Debug Logs

DELL@DESKTOP-MVQO1NB MINGW64 ~/esp/esp-idf/examples/peripherals/i2c/i2c_self_test (master)
$ idf.py build
Checking Python dependencies...
Python requirements from C:\Users\DELL\esp\esp-idf\requirements.txt are satisfied.
Running ninja in directory c:\users\dell\esp\esp-idf\examples\peripherals\i2c\i2c_self_test\build
Executing "ninja all"...
[1/161] Performing build step for 'bootloader'
ninja: no work to do.
[2/159] Building C object esp-idf/hd44780/CMakeFiles/__idf_hd44780.dir/hd44780.c.obj
FAILED: esp-idf/hd44780/CMakeFiles/__idf_hd44780.dir/hd44780.c.obj
D:\Programs\Espressif\ESP-IDF-Tools\toolchain\bin\xtensa-esp32-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Icon
fig -I../../../../../components/hd44780/. -I../../../../../components/newlib/platform_include -I../../../../../components/newli
b/include -I../../../../../components/freertos/include -I../../../../../components/heap/include -I../../../../../components/log
/include -I../../../../../components/soc/esp32/include -I../../../../../components/soc/include -I../../../../../components/esp_
rom/include -I../../../../../components/esp_common/include -I../../../../../components/xtensa/include -I../../../../../componen
ts/xtensa/esp32/include -I../../../../../components/esp32/include -I../../../../../components/app_update/include -I../../../../
../components/spi_flash/include -I../../../../../components/bootloader_support/include -I../../../../../components/mbedtls/port
/include -I../../../../../components/mbedtls/mbedtls/include -I../../../../../components/lwip/include/apps -I../../../../../com
ponents/lwip/include/apps/sntp -I../../../../../components/lwip/lwip/src/include -I../../../../../components/lwip/port/esp32/in
clude -I../../../../../components/lwip/port/esp32/include/arch -I../../../../../components/vfs/include -I../../../../../compone
nts/esp_wifi/include -I../../../../../components/driver/include -I../../../../../components/esp_ringbuf/include -I../../../../.
./components/esp_event/include -I../../../../../components/tcpip_adapter/include -I../../../../../components/ethernet/include -
I../../../../../components/efuse/include -I../../../../../components/efuse/esp32/include -I../../../../../components/pthread/in
clude -I../../../../../components/app_trace/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstri
ct-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unus
ed-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-st
yle-declaration -DHAVE_CONFIG_H -DIDF_VER=\"v4.0-dev-562-g2b301f53e\" -DGCC_NOT_5_2_0=0 -DESP_PLATFORM -MD -MT esp-idf/hd44780/
CMakeFiles/__idf_hd44780.dir/hd44780.c.obj -MF esp-idf\hd44780\CMakeFiles\__idf_hd44780.dir\hd44780.c.obj.d -o esp-idf/hd44780/
CMakeFiles/__idf_hd44780.dir/hd44780.c.obj   -c C:/Users/DELL/esp/esp-idf/components/hd44780/hd44780.c
C:/Users/DELL/esp/esp-idf/components/hd44780/hd44780.c: In function 'write_nibble':
C:/Users/DELL/esp/esp-idf/components/hd44780/hd44780.c:48:29: error: implicit declaration of function 'ets_delay_us' [-Werror=i
mplicit-function-declaration]
 #define toggle_delay() do { ets_delay_us(DELAY_TOGGLE); } while (0)
                             ^
C:/Users/DELL/esp/esp-idf/components/hd44780/hd44780.c:66:9: note: in expansion of macro 'toggle_delay'
         toggle_delay();
         ^
C:/Users/DELL/esp/esp-idf/components/hd44780/hd44780.c: At top level:
cc1.exe: warning: unrecognized command line option '-Wno-frame-address'
cc1.exe: some warnings being treated as errors
[7/159] Building C object esp-idf/json/CMakeFiles/__idf_json.dir/cJSON/cJSON.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder) -> ORIGINAL from toolchain 
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3219678/sdkconfig.zip)

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**) -> not available...apparently process is interrupted before...

- [ ] coredump (This provides stacks of tasks.)  ->does not apply as application is not built.

igrr, 2019-05-27:
The errors seem to refer to `esp-idf/components/hd44780`. However such component directory doesn't exist in ESP-IDF. Could you please try getting a clean copy of ESP-IDF as explained in the getting started guides? https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-get-esp-idf

jcabad100, 2019-05-27:
@igrr ....i am rather new to this community and IDE..i did just installed the tool chain last week , twice, two different versions....i followed as much as possible the very cofusing multiplatform getting started guide and in one form or the other it doesnt work properly, some examples build other dont...and if the hd444780 library doesnt belong to the ESP-IDF why the examples that come with the tool dont at least have a documentation pointing out which external libraries must be installed and how?

jcabad100, 2019-05-27:
@igrr ...so, sorry for my ignorance ...i understood what have caused this misbehavior....as you already noticed, there are libraries like esp-idf-lib-master , hd44780, dht, bmp180 bmp280 etc etc that i added two days ago (trying to carry out examples from a book on IoT & ESP32) from another Github repository by simply copying .under .\esp\esp-idf\components ..so it seems that that "Executing "ninja all"..." try to build all components under the components folder despite those that i added has nothing to do with the hello_world application! .....in fact they are reflected under the folder ..L\esp\esp-idf\examples\get-started\hello_world\build\esp-idf ...that is crazy, i would have expected that the build is smart and just try to build the objects used by the application but that is not the case, all the libraries under components get build for all applications! ....i dir remove those components from the components folder and in fact the example applications  (i tried hello_world and protocols/mqtt ) can be build again (using "idy.py build" followed by "idf.py -p COM8 flash" because 'idf.py menuconfig" does not work as reported separately. Therefore this issue could be CLOSED. I would just appreciate if you @igrr or someone else could explain me how to properly import external libraries into the toolchain? (maybe just those libraries used make instead of Cmake?)...thank you very much in advance!

jhubbardsf, 2019-11-18:
> @igrr ....i am rather new to this community and IDE..i did just installed the tool chain last week , twice, two different versions....i followed as much as possible the very cofusing multiplatform [...]

I wanted to add to this that I think it'd be useful to create separate guides for MacOS/Linux/Windows. Obviously that's a little more work but the espressif documentation is already very thick, anything to make work easier would be good. Possible idea for a feature request.

igrr, 2022-01-04:
I'm sorry for the late follow-up. Some time ago we have added an example of adding an external library into an IDF project: https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/import_lib. The specific steps will vary depending on the library being imported, but example this shows what the integration would like for a generic CMake library.

Regarding splitting the getting started guides into OS-specific versions, this is something we have started working on!

closed 2022-01-04 by igrr