espressif/esp-idf #3547: about nvs_partition_gen

nopnop2002, 2019-05-25:
I want to read vale from 2nd NVS partition, but i can't read.
I followed the following page.
What is wrong??

My Environment:

	freeRTOS version:V8.2.0
	NEWLIB version:2.2.0
	lwIP version:2-0-3-255
	ESP-IDF version:v4.0-dev-512-g6fd535c98

My partition:

	$ cat partitions.csv
	# Name,   Type, SubType, Offset,  Size, Flags
	# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
	nvs,      data, nvs,     0x9000,  0x6000,
	phy_init, data, phy,     0xf000,  0x1000,
	fctry, data, nvs, 0x10000, 0x6000,
	factory,  app,  factory, 0x20000, 1M,

	I (47) boot: Partition Table:
	I (51) boot: ## Label            Usage          Type ST Offset   Length
	I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
	I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
	I (73) boot:  2 fctry            WiFi data        01 02 00010000 00006000
	I (81) boot:  3 factory          factory app      00 00 00020000 00100000
	I (88) boot: End of partition table

Generate My factory data:

	$ cat mfg_config.csv
	key,type,encoding,value
	device_data,namespace,,
	serial_no,data,string,164589345735
	mac_addr,data,string,0A:0B:0C:0D:0E:0F

	$ python $IDF_PATH/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py --input mfg_config.csv --output my_mfg.bin --size 0x6000
	Version:  v2 - Multipage Blob Support Enabled
	NVS binary created: /home/nop/rtos/nvs_partition_gen/sample.bin

	$ ls -l sample.bin
	-rw-rw-r-- 1 nop nop 24576  5月 14 23:21 sample.bin

	$ python $IDF_PATH/components/esptool_py/esptool/esptool.py  write_flash 0x10000  my_mfg.bin
	esptool.py v2.7-dev
	Found 3 serial ports
	Serial port /dev/ttyUSB0
	Connecting........___
	Detecting chip type... ESP32
	Chip is ESP32D0WDQ6 (revision 1)
	Features: WiFi, BT, Dual Core, Coding Scheme None
	MAC: 30:ae:a4:24:bc:5c
	Uploading stub...
	Running stub...
	Stub running...
	Configuring flash size...
	Auto-detected Flash size: 4MB
	Compressed 24576 bytes to 258...
	Wrote 24576 bytes (258 compressed) at 0x00010000 in 0.0 seconds (effective 7288.8 kbit/s)...
	Hash of data verified.

	Leaving...
	Hard resetting via RTS pin...

My source Code:

	#include <stdio.h>
	#include "freertos/FreeRTOS.h"
	#include "freertos/task.h"
	#include "esp_system.h"
	#include "nvs_flash.h"
	#include "nvs.h"

	#define MFG_PARTITION_NAME "fctry"

	void app_main()
	{
	    // Initialize NVS
	        esp_err_t err = nvs_flash_init_partition(MFG_PARTITION_NAME);
	    if (err != ESP_OK) {
	        printf("Error (%s) nvs_flash_init_partition!\n", esp_err_to_name(err));
	        }

	    // Open
	    printf("Opening Non-Volatile Storage (NVS) handle... ");
	        nvs_handle fctry_handle;
	    err = nvs_open_from_partition(MFG_PARTITION_NAME, "fctryNamespace",
	               NVS_READWRITE, &fctry_handle);
	    if (err != ESP_OK) {
	        printf("Error (%s) nvs_open_from_partition!\n", esp_err_to_name(err));
	    } else {
	        printf("Done\n");

	        // Read
	        printf("Reading serial_number from NVS ... \n");
	                char buf[64];
	                size_t buflen;
	                err = nvs_get_str(fctry_handle, "serial_number", buf, &buflen);
	        switch (err) {
	            case ESP_OK:
	                printf("Done\n");
	                printf("serial_number = %s\n", buf);
	                break;
	            case ESP_ERR_NVS_NOT_FOUND:
	                printf("The value is not initialized yet!\n");
	                break;
	            default :
	                printf("Error (%s) reading!\n", esp_err_to_name(err));
	        }

	        // Close
	        nvs_close(fctry_handle);
	    }

	    printf("\n");
	}

Result:

	Opening Non-Volatile Storage (NVS) handle... Done
	Reading serial_number from NVS ...
	The value is not initialized yet!

filzek, 2019-05-26:
Follow this steps?

/ Example (without error checking) of using nvs_get_str to get a string into dynamic array:
size_t required_size;
nvs_get_str(my_handle, "server_name", NULL, &required_size);
char* server_name = malloc(required_size);
nvs_get_str(my_handle, "server_name", server_name, &required_size);

// Example (without error checking) of using nvs_get_blob to get a binary data
into a static array:
uint8_t mac_addr[6];
size_t size = sizeof(mac_addr);
nvs_get_blob(my_handle, "dst_mac_addr", mac_addr, &size);

nopnop2002, 2019-05-26:
Yes.
I follow this steps.

	char buf[64];
	size_t buflen;
	err = nvs_get_str(fctry_handle, "serial_number", buf, &buflen);

But err is ESP_ERR_NVS_NOT_FOUND.

nopnop2002, 2019-05-26:
Hello.
Yes.I follow this steps.

	size_t buflen;
  err = nvs_get_str(fctry_handle, "serial_number", NULL, &buflen);

But err is ESP_ERR_NVS_NOT_FOUND.

avsheth, 2019-05-27:
`$ cat mfg_config.csv
key,type,encoding,value
device_data,namespace,,
serial_no,data,string,164589345735
mac_addr,data,string,0A:0B:0C:0D:0E:0F`

As per your config data here, your namespace is `device_data` and key is `serial_no`, whereas in your API calls
`nvs_open_from_partition(MFG_PARTITION_NAME, "fctryNamespace", NVS_READWRITE, &fctry_handle);
`nvs_get_str(fctry_handle, "serial_number", NULL, &buflen);`

You are looking for `serial_number` in `fctryNamespace`. Can you try

`nvs_open_from_partition(MFG_PARTITION_NAME, "device_data", NVS_READWRITE, &fctry_handle);
`nvs_get_str(fctry_handle, "serial_no", NULL, &buflen);`

nopnop2002, 2019-05-27:
Thank you very much.
It become to work right.
But, I encounter a strange phenomenon.

The 1st time, [mac_addr] can be read, the 2nd time, [mac_addr] can't be read.
Why Reading 2nd mac_addr from NVS can't work?

My source:

	#include <stdio.h>
	#include "freertos/FreeRTOS.h"
	#include "freertos/task.h"
	#include "esp_system.h"
	#include "nvs_flash.h"
	#include "nvs.h"

	#define MFG_PARTITION_NAME "fctry"

	void app_main()
	{
	    esp_err_t err;

	    // Initialize NVS
	        err = nvs_flash_init_partition(MFG_PARTITION_NAME);
	    if (err != ESP_OK) {
	        printf("Error (%s) nvs_flash_init_partition!\n", esp_err_to_name(err));
	        }

	    // Open
	    printf("Opening Non-Volatile Storage (NVS) handle... ");
	        nvs_handle fctry_handle;
	    err = nvs_open_from_partition(MFG_PARTITION_NAME, "device_data",
	               NVS_READWRITE, &fctry_handle);
	    if (err != ESP_OK) {
	        printf("Error (%s) nvs_open_from_partition!\n", esp_err_to_name(err));
	    } else {
	        printf("Done\n");

	#if 1
	        // Read
	        printf("Reading 1st mac_addr from NVS ... \n");
	                char buf[64];
	                size_t buflen;

	                err = nvs_get_str(fctry_handle, "mac_addr", buf, &buflen);
	        switch (err) {
	            case ESP_OK:
	                printf("Done\n");
	                printf("bulen=%d mac_addr=[%s]\n", buflen, buf);
	                break;
	            case ESP_ERR_NVS_NOT_FOUND:
	                printf("The value is not initialized yet!\n");
	                break;
	            default :
	                printf("Error (%s) reading!\n", esp_err_to_name(err));
	        }
	#endif

	        // Read
	        printf("Reading serial_no from NVS ... \n");
	                err = nvs_get_str(fctry_handle, "serial_no", buf, &buflen);
	        switch (err) {
	            case ESP_OK:
	                printf("Done\n");
	                printf("buflen=%d serial_no=[%s]\n", buflen, buf);
	                break;
	            case ESP_ERR_NVS_NOT_FOUND:
	                printf("The value is not initialized yet!\n");
	                break;
	            default :
	                printf("Error (%s) reading!\n", esp_err_to_name(err));
	        }

	#if 1
	        // Read
	        printf("Reading 2nd mac_addr from NVS ... \n");
	                err = nvs_get_str(fctry_handle, "mac_addr", buf, &buflen);
	        switch (err) {
	            case ESP_OK:
	                printf("Done\n");
	                printf("bulen=%d mac_addr=[%s]\n", buflen, buf);
	                break;
	            case ESP_ERR_NVS_NOT_FOUND:
	                printf("The value is not initialized yet!\n");
	                break;
	            default :
	                printf("Error (%s) reading!\n", esp_err_to_name(err));
	        }
	#endif

	        // Close
	        nvs_close(fctry_handle);
	    }

	    printf("\n");
	}

My result:

	Opening Non-Volatile Storage (NVS) handle... Done
	Reading 1st mac_addr from NVS ...
	Done
	bulen=18 mac_addr=[0A:0B:0C:0D:0E:0F]
	Reading serial_no from NVS ...
	Done
	buflen=13 serial_no=[164589345735]
	Reading 2nd mac_addr from NVS ...
	Error (ESP_ERR_NVS_INVALID_LENGTH) reading!

avsheth, 2019-05-27:
nvs_get_str actually modifies the last parameter, i.e. buflen (it populates buflen with actual data read.
since you read serial number before second mac address, your buflen is now set to 13, and mac address is of length 18, hence it throws an error which says, buffer is inadequate to hold entire mac address.
Before calling nvs_get_str please set `buflen = sizeof(buf)`. (even in your code when buflen is first defined, you should initialise it to sizeof(buf), since it's on a stack and could have any random value, which could lead to corruption or undefined behavior)

nopnop2002, 2019-05-27:
Thank you.
It work fine.
Please close this issues.

closed 2019-05-27 by Alvin1Zhang