espressif/esp-idf #3548: flash read err, 1000

rhr407, 2019-05-26:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2019-05-27:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

renzbagaporo, 2019-05-27:
Also, the problem description seem to be just the issue template copied verbatim and do not actually contain details about the issue. We encourage filling this template out so that we can help you as best as we could.

negativekelvin, 2019-05-29:
You can't erase the bootloader after you have previously enabled secure boot.  You also can't enable flash encryption after you have previously enabled secure boot, you have to enable them both together. 

I do agree that there should be more explicit warnings like "Production Only! Not for development use!" and a development mode that doesn't read protect the efuse keys.

rhr407, 2019-05-29:
So it means if I have to use the Secure boot with flash encryption I have to enable both simultaneously first? 
and then follow the key burning step
then make bootloader step and
then make flash monitor
All the above steps after I have generated the .pem key file

Please correct me if I am wrong.

ghemal, 2019-05-29:
Sorry to hear about your problem and thanks for the detailed information

The continuous boot loop indicates the ROM code can not understand the flash contents when it tries to load the second stage boot loader from flash. This can happen due to wrong configuration (such as flash encryption is enabled but flash contents are not encrypted or vice versa). Also once enabled it is not possible to disable secure boot in device (even if you disable it through menuconfig)

In your case it seems you have enabled secure boot and flash encryption individually one at a time and then tried to disable secure boot and then flash the device (perhaps bootloader got programmed multiple times with different configuration). If you want to use both the features it is recommended to enable them together to avoid any misconfiguration issue. However, good thing I see here is you have selected secure boot mode as “re-flashable" and flash encryption key is generated "externally". So you have access to both the keys (secure boot & flash encryption) to create new bootable image.

I would suggest you to try following steps and let us know if it helps

1. In menuconfig ensure secure boot mode as “reflashable” as well as flash encryption is enabled
2. make bootloader, this will generate a bootloader binary with secure digest
3. Encrypt the bootloader binary using flash encryption key with encrypt_flash_data command. For example please refer [.](https://docs.espressif.com/projects/esp-idf/en/latest/security/flash-encryption.html#reflashing-with-pregenerated-key) The bootloader is written at address 0x1000
4. Next you can program the newly encrypted bootloader binary using the same command as was printed by make bootloader (but instead use encrypted bootloader bin)
5. Similarly encrypt the partition table binary and application binary with same flash encryption key using appropriate addresses in your case (usually the partition table resides at 0x8000 and default app image at 0x10000). If you have OTA partitions do the same for them
6. Program all the encrypted partitions in above step using write_flash command and corresponding address for each partition. Sample command

> esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash 0x10000 build/my-app.bin

7. Reset the device

rhr407, 2019-06-05:
@ghemal : Thank you so much for your detailed reply. @gh@gh

I have followed the above steps in the following manner:

## BEFORE ENCRYPTING THE PARTITIONS:

1. 
```   
make menuconfig
```  

2. Set the following:
a. Secure boot and flash encryption steps in menuconfig:
[image]
b. Set two OTA partitions:
[image]

Above are the outputs in my case.

3. Do the Key Generation steps for secureboot and flash encryption putting the secure_boot_signing_key.pem and my_flash_encryption_key.bin in the project folder.
```
openssl ecparam -name prime256v1 -genkey -noout -out my_secure_boot_signing_key.pem
```
```
espsecure.py digest_private_key --keyfile secure_boot_signing_key.pem --keylen 256 my_flash_encryption_key.bin
```

4. 
``` 
make clean 
``` 

5. To make it sure that we do not have any other data in flash:
``` 
make erase_flash
``` 

5. This following command will generate the `app.bin` and `partitions_two_ota.bin` files 
``` 
make -j4
``` 

6. Generate bootloader.bin file
``` 
make bootloader
``` 

will generate the `bootlaoder.bin` file inside build/bootloader/bootloader.bin folder

7. Check addresses of partitions you have in generated partition table:

``` 
make partition_table
``` 
[image]
The above is the output in my case.

The above command will show the addresses for each partition in generated partition table before using in the following section:

## ENCRYPTING ALL THE PARTITIONS  (generated above) AND BURNING IN THE FLASH:

1. First encrypt the `bootloader.bin` file using the my_flash_encryption_key.bin file at addres 0x1000:

```  
espsecure.py encrypt_flash_data --keyfile my_flash_encryption_key.bin --address 0x1000 -o build/bootloader/bootloader-encrypted.bin build/bootloader/bootloader.bin 
```  

2. Burn the encrypted `bootloader-encrypted.bin` in the flash using the following command:
``` 
python esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /build/bootloader/bootloader-encrypted.bin

``` 

3. Now encrypt the `partitions_two_ota.bin` using the my_flash_encryption_key.bin file at addres 0x9000:
``` 
espsecure.py encrypt_flash_data --keyfile my_flash_encryption_key.bin --address 0x9000 -o build/partitions_two_ota-encrypted.bin build/partitions_two_ota.bin
``` 
4. Burn the encrypted `partitions_two_ota-encrypted.bin` in the flash using the following command:
``` 
esptool.py --port /dev/ttyUSB0 --baud 921000 write_flash 0x9000 build/partitions_two_ota-encrypted.bin
``` 

5. Next encrypt the `app.bin` file at address 0x10000:
```
espsecure.py encrypt_flash_data --keyfile my_flash_encryption_key.bin --address 0x10000 -o build/app-encrypted.bin build/app.bin
```
6. Burn the `app-encrypted.bin` into the flash:
``` 
esptool.py --port /dev/ttyUSB0 --baud 921000 write_flash 0x10000 build/app-encrypted.bin
```

After doing ``make monitor `` , the screen is continuously printing as follows:
[image]

NOTE: I have not yet encrypted the OTA partitions as I just have to first test the app

rhr407, 2019-06-05:
Got new ESP32 DOIT devkit v1 
Did all the above steps. I have noticed that we have to burn the 'espefuse.py burn_efuse FLASH_CRYPT_CNT' and 'ABS_DONE_0' fuses manually. Just enabling the option for secure boot and flash encryption do nothing. I have tested this.
[image]

[image]

ghemal, 2019-06-08:
Hi rhr407,

Thanks for the details

The steps I recommended were for the device which already has flash encryption and secure boot enabled and when you want to recover the device with improper image. They wouldn't work on a _fresh_ device where flash encryption and secure boot is not yet enabled (for which the original documentation of flash encryption and secure boot needs to be followed)

I take it from your log that in your case you put the encrypted bootloader on a device which hasn't enabled flash encryption (as you can see from your efuse summary) so obviously it can't boot

You have manually programmed the FLASH_CRYPT_CNT and ABS_DONE_0. In addition for flash encryption to work you also need to program FLASH_CRYPT_CONFIG to 0xF as per [this](https://docs.espressif.com/projects/esp-idf/en/latest/security/flash-encryption.html#setting-flash-crypt-config). Ideally these efuses gets programmed when the bootloader runs for the first time (after enabling these features) but since in your case bootloader couldn't execute the efuses never got programmed

Two additional things to check are

1. I see that your default factory app is at offset 0x20000 as per partition table but you are programming the app.bin at 0x10000 which has phy init data so this needs to be corrected

2. Please check the size of your bootloader.bin. There is one known problem where ESP32 fails to boot when secure boot and flash encryption are enabled together as mentioned [here](https://docs.espressif.com/projects/esp-idf/en/latest/security/secure-boot.html#bootloader-size)

rhr407, 2019-06-10:
Thank you so much for your message. I really appreciate your help and assistance. Also thanks for highlighting my mistake of not burning flash at 0x20000 (that is the address of the factory app in my partition table)

I repeated all the steps you mentioned in your previous message. I did the following steps:
1. Showing the status of efuses:
```  
espefuse.py summary
```  
[image]

2. Compiling the app and generating bootloader:
```  
make clean && make erase_flash && make -j4 && make bootloader
```  
[image]

Here two binaries have been generated. bootloader.bin and bootloader-reflash-digest.bin files.

3. Check for the generated partition tables addresses:
```  
make partition_table
```  
[image]

4. Signing and writing the .bin files in to the flash:
   a. Signing bootloader.bin as follows with my_flash_encryption_key.bin file [NOTE**:I need to know if I have to do this step for bootloader-reflash-digest.bin as well??]:
      ```  
      espsecure.py encrypt_flash_data --keyfile my_flash_encryption_key.bin --address 0x1000 -o build/bootloader/bootloader-encrypted.bin build/bootloader/bootloader.bin
      ```  
      [image]

   b. Writing the encrypted bootloader to flash at 0x1000:
   ```  
   esptool.py --port /dev/ttyUSB0 --baud 921000 write_flash 0x1000 build/bootloader/bootloader-encrypted.bin
   ```  
   [image]

   c. Now encrypting partitions_singleapp.bin:
   ```  
   espsecure.py encrypt_flash_data --keyfile my_flash_encryption_key.bin --address 0x12000 -o build/partitions_singleapp-encrypted.bin build/partitions_singleapp.bin
   ```
   [image]

   d. Writing the app-encrypted.bin to flash at 0x12000
   ```    
   esptool.py --port /dev/ttyUSB0 --baud 921000 write_flash 0x12000 build/partitions_singleapp-encrypted.bin
   ``` 

   [image]

   e. Now encrypting app.bin: [It will take a while as the app size is bigger]
   ```  
   espsecure.py encrypt_flash_data --keyfile my_flash_encryption_key.bin --address 0x20000 -o build/app-encrypted.bin build/app.bin
   ```  

   [image]

   f. Writing the app-encrypted.bin to flash at 0x20000
   ```  
   esptool.py --port /dev/ttyUSB0 --baud 921000 write_flash 0x20000 build/app-encrypted.bin
   ``` 

   [image]

5. Reset the esp32 and watch monitor:
```  
make monitor
```  

[image]

NOW I am facing this issue. The above image shows the terminal output continuously printing the following text:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:10444
load:0x40078000,len:18888
load:0x40080400,len:6580
secure boot check fail
ets_main.c 371 
ets Jun  8 2016 00:22:57

negativekelvin, 2019-06-10:
Yes you have to write the encrypted bootloader digest to 0x0 or the combined encrypted bootloader-digest.bin to 0x0.

rhr407, 2019-06-10:
Awesome!! 
It's working now Alhamdulillah! 

[image]

Thank you so much for your kind assistance and support.

closed 2019-06-10 by rhr407

chhotu, 2019-07-29:
Hi,
I have enabled Secure boot(Reflashable mode) + flash encryption(Development mode).
I have built 'Simple_ota_example' and able to run properly.

Now, I want to change Secure boot mode to 'one-time-flash' from 'Reflashable'
Can anyone help me, which steps I need to follow for the same?

Note: I want to switch this configuration in the same board.

1, Reflashable Mode configuration : 
[image]

2. one-time flash mode configuration : 
[image]

Apart from this what kind of steps do I need to perform?

Replay would be appreciated !!!!

negativekelvin, 2019-07-30:
@chhotu i think the only difference is whether or not you have the bootloader signing key

diegosantos2, 2019-10-20:
Thk a lot rhr407
finally i find the way for the activated the secure boot with the encryption too.}
thk a again.