espressif/esp-idf #3549: stl::unordered_map insert() causes abort() when out-of-memory

zoliweiss, 2019-05-27:
## Environment

- Board: ESP32 DoIt module - Chip is ESP32D0WDQ6 (revision 1)
- Core Installation version: latest master in ARDUINO/ESP32
- IDE name: Arduino IDE 1.8.5
- Flash Frequency: 80Mhz
- PSRAM enabled: IDK
- Upload Speed: 921600
- Computer OS: Win 7
- Power Supply: USB

## Problem Description
https://github.com/espressif/arduino-esp32/issues/2827

When I insert too much entries into unordered_map container, the ESP32 suddenly reboots. Should be throwing bad_alloc exception. I tried catching (...) also. Exception decoder can't decode anything. I am not sure if this is an ESP32/GCC/... problem.

### Expected Behavior

Should be throwing bad_alloc exception.

### Actual Behavior

Module restarts.

### Steps to repropduce

Run the code.

### Code to reproduce this issue

```cpp
  std::unordered_map<int, std::string> l_map;

  for(int i = 0; i < 100000; i++)
  {
    printf("%d\n", i);

    try
    {
      auto l_result = l_map.insert({i, std::to_string(i)});

      std::cout << l_result.second << std::endl;
    }
    catch(const std::bad_alloc&)
    {
      printf("OOOOOPS\n");
    }
  }

```
## Debug Logs

```
1
5378
1
5379
1
5380
1
5381
abort() was called at PC 0x400f594f on core 1

Backtrace: 0x4008b384:0x3ffb1e40 0x4008b5b5:0x3ffb1e60 0x400f594f:0x3ffb1e80 0x400f5996:0x3ffb1ea0 0x400f5b71:0x3ffb1ec0 0x400e6140:0x3ffb1ee0 0x400d18aa:0x3ffb1f00 0x400d1986:0x3ffb1f30 0x40100eab:0x3ffb1fb0 0x4008765d:0x3ffb1fd0

Rebooting...
```

igrr, 2019-05-27:
Have verified that your example works in IDF:

```
1
7247
1
7248
1
7249
OOOOOPS
7250
OOOOOPS
7251
OOOOOPS
7252
OOOOOPS
7253
OOOOOPS
```

<details><summary>with code:</summary>

```c++
#define _GLIBCXX_USE_C99
#include <iostream>
#include <string>
#include <unordered_map>
#include <stdio.h>

extern "C" void app_main()
{
  std::unordered_map<int, std::string> l_map;

  for(int i = 0; i < 100000; i++)
  {
    printf("%d\n", i);

    try
    {
      auto l_result = l_map.insert({i, std::to_string(i)});

      std::cout << l_result.second << std::endl;
    }
    catch(const std::bad_alloc&)
    {
      printf("OOOOOPS\n");
    }
  }
}

```
</details>

and sdkconfig:
```
CONFIG_COMPILER_CXX_EXCEPTIONS=y
CONFIG_COMPILER_CXX_EXCEPTIONS_EMG_POOL_SIZE=1024
```

As you have observed, this does not work in arduino-esp32 because it is compiled with `CONFIG_COMPILER_CXX_EXCEPTIONS_EMG_POOL_SIZE=0`. Therefore, when you run out of memory, an exception object can not be created, leading to an abort.

I think the resolution for this should be in arduino-esp32, setting `CONFIG_COMPILER_CXX_EXCEPTIONS_EMG_POOL_SIZE` to some reasonable value to allow for creation of at least one exception object.

zoliweiss, 2019-05-27:
Thank you for the quick response. Should I close this issue?

igrr, 2019-06-14:
Yes, based on the above, I think the issue is that arduino-esp32 is configured with zero exception emergency pool size.

closed 2019-06-14 by igrr