espressif/esp-idf #3550: The audio connection of HFP profile works only in one direction during a call

AndrijKostio, 2019-05-27:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-D2WD|ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    v4.0-dev-621-g81ca1c013
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    1.22.0-80-g6c4433a
- Operating System:     Windows|Linux
- Power Supply:         USB|Battery

## Problem Description

The audio connection of the HFP profile works only in one direction, it is possible only to send or receive data. SCO is set over HCI, but the situation is the same when using the PCM interface as data path for SCO packets.
The "(SCO/eSCO) Max Connections" is set to different values in menuconfig up to max value.
When the outgoing data sending works (from the ESP32 to the phone), the incoming audio goes to the speaker of the phone and not being sent over the existing BT connection.

yunling-cn, 2019-06-23:
Hello, how do you implement the connection to the Bluetooth headset? I use ESP32 to connect to the Bluetooth headset and always stuck in "esp_a2d_source_connect". I just modified the official example a2dp_source. But connecting a computer can be, that is, connecting a Bluetooth headset does not work. Sometimes it suddenly crashes and then restarts and suddenly connects.

AbnerFederer, 2019-07-23:
What's your definition of the real time 2-way voice? According to the BT frame, the SCO packets will not be the real "real time 2-way" connection...if the consequence is the voice from phone to the ESP32 cannot be sent and instead play on the phone itself, then please give some more evidence please~

AbnerFederer, 2019-07-29:
Hello, there has some modification in the decode mode, we just added a new decodeing mode like this "param->audio_stat.state == ESP_HF_CLIENT_AUDIO_STATE_CONNECTED_MSBC",so please check your code to figure out whether you have add this. If the question still occurs, we can do osme further work on it, thank you.

AndrijKostio, 2019-11-30:
Hello, thank you for your efforts to make the HFP working full functionally.
Now I can send and receive SCO packets (over HCI), but the problem is that a number of sent packets being lost/dropped:
_W (39364) BT_BTM: SCO xmit Q overflow, pkt dropped
W (39364) BT_BTM: stat 10_
Can you advise something to avoid data loss during transaction?

AbnerFederer, 2019-12-02:
Hi, @AndrijKostio 

Maybe the frequency of `esp_hf_client_outgoing_data_ready()` is the root cause if this log. So, maybe you can constrain times calling this API like below:

```c
int ready_times=5; // Or another integer.
...
function()
{
int count=0;
...

if (count < ready_times)
   count ++;
else if (count == ready_times) {
   count=0;
  esp_hf_client_outgoing_data_ready();
}
...
}

```

Hasaken, 2019-12-10:
Hi, @AbnerFederer 

Changing the frequency of function calling doesn't help, unfortunately. Can you advise something else, please?

AndrijKostio, 2019-12-11:
Hi, @AbnerFederer 
Even when calling the esp_hf_client_outgoing_data_ready() less frequently there are still dropped packets and because of less frequent function call there are data loss of outgoing data.
Can we try something else?

AbnerFederer, 2019-12-19:
Hi, @AndrijKostio 

Change 
#define BTM_SCO_DATA_SIZE_MAX       120 
to
#define BTM_SCO_DATA_SIZE_MAX       240

Thanks
Abner

Alvin1Zhang, 2019-12-20:
@AndrijKostio Thanks for reporting. @AbnerFederer Thanks for the fix. @AndrijKostio Feel free to reopen if the issue still happens. Thanks.

closed 2019-12-20 by Alvin1Zhang

Tarun-Narain, 2020-02-12:
Hi guys, 
i've tried the suggested changes but error still persists
please HELP!

Thank-you
Tarun Narain