espressif/esp-idf #3551: BLE connection disconnect reason = 0x8 after call the esp_ota_begin() function

zhanlej, 2019-05-28:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
    ``v4.0-dev-621-g81ca1c013`` and ``v4.0-dev-447-g689032650`` and ``v3.2``
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

Sometimes executing the esp_ota_begin() function when the BLE connection is successfully established between the device and the APP will cause the BLE connection to be disconnected and prompt "ESP_GATTS_DISCONNECT_EVT, reason = 0x8"

### Expected Behavior

BLE connection will not be disconnected

### Actual Behavior

Sometimes the BLE connection disconnected

### Steps to repropduce

1. Device and APP establish BLE connection
2. APP writes 0x3344 to the characteristic UUID=FF02
3. Sometimes it causes BLE to disconnect

### Code to reproduce this issue

```cpp
int ota_begin()
{
    const esp_partition_t *configured = esp_ota_get_boot_partition();
    const esp_partition_t *running = esp_ota_get_running_partition();

    printf("%s,%d\r\n", __FUNCTION__, __LINE__);
    if (configured != running) {
        ESP_LOGW(TAG, "Configured OTA boot partition at offset 0x%08x, but running from offset 0x%08x",
                configured->address, running->address);
        ESP_LOGW(TAG, "(This can happen if either the OTA boot data or preferred boot image become corrupted somehow.)");
    }
    ESP_LOGI(TAG, "Running partition type %d subtype %d (offset 0x%08x)",
            running->type, running->subtype, running->address);

    printf("%s,%d\r\n", __FUNCTION__, __LINE__);
    update_partition = esp_ota_get_next_update_partition(NULL);
    ESP_LOGI(TAG, "Writing to partition subtype %d at offset 0x%x",
            update_partition->subtype, update_partition->address);
    if(update_partition == NULL)
    {
        ESP_LOGE(TAG, "update_partition == NULL");
        return -1;
    }
    
    printf("%s,%d\r\n", __FUNCTION__, __LINE__);
    update_handle = 0;
    iot_ble_esp_err = esp_ota_begin(update_partition, OTA_SIZE_UNKNOWN, &update_handle);
    if (iot_ble_esp_err != ESP_OK) {
        ESP_LOGE(TAG, "esp_ota_begin failed (%s)", esp_err_to_name(iot_ble_esp_err));
        return -1;
    }

    return 0;
}

static void gatts_profile_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param)
[... 129 lines cut ...]
I (2462) QL_c1246e: ESP_GATTS_MTU_EVT, MTU 185
I (2462) QL_c1246e: update connection params status = 15, min_int = 32, max_int = 64,conn_int = 24,latency = 0, timeout = 72
I (3002) QL_c1246e: ESP_GATTS_READ_EVT
I (3002) QL_c1246e: read.handle:42, IDX_CHAR_VAL_A.handle:42

I (3122) QL_c1246e: ESP_GATTS_READ_EVT
I (3122) QL_c1246e: read.handle:45, IDX_CHAR_VAL_A.handle:42

I (3242) QL_c1246e: ESP_GATTS_READ_EVT
I (3242) QL_c1246e: read.handle:43, IDX_CHAR_VAL_A.handle:42

I (8432) QL_c1246e: GATT_WRITE_EVT, handle = 45, value len = 2, value :
I (8432) QL_c1246e: 33 44 
ota_begin,276
I (8432) native_ota_example: Running partition type 0 subtype 0 (offset 0x00010000)
ota_begin,285
I (8442) native_ota_example: Writing to partition subtype 16 at offset 0x110000
ota_begin,295
I (10802) QL_c1246e: ESP_GATTS_DISCONNECT_EVT, reason = 0x8
I (10812) QL_c1246e: advertising start successfully
```

## Other items if possible

- [gatt_server_service_table_ota.zip](https://github.com/espressif/esp-idf/files/3226525/gatt_server_service_table_ota.zip)

chegewara, 2019-05-28:
error `0x08 Connection Timeout`

zhanlej, 2019-05-28:
Why trigger timeout

chegewara, 2019-05-28:
Maybe your peer device is timed out, maybe you need to increase timeout value in connection settings.
Your timeout value is very very low:
`update connection params status = 15, min_int = 32, max_int = 64,conn_int = 24,latency = 0, timeout = 72`

zhanlej, 2019-05-28:
The peer device is iphone6s and use ``nRF Connect`` APP. I don't know why it will time out.

How can I increase the timeout value? I modified the following code, but the result is the same:
```cpp
...
        case ESP_GATTS_CONNECT_EVT:
            ESP_LOGI(IOT_BLE_TAG, "ESP_GATTS_CONNECT_EVT, conn_id = %d", param->connect.conn_id);
            esp_log_buffer_hex(IOT_BLE_TAG, param->connect.remote_bda, 6);
            esp_ble_conn_update_params_t conn_params = {0};
            memcpy(conn_params.bda, param->connect.remote_bda, sizeof(esp_bd_addr_t));
            /* For the IOS system, please reference the apple official documents about the ble connection parameters restrictions. */
            conn_params.latency = 0;
            conn_params.max_int = 0x40;    // max_int = 0x20*1.25ms = 40ms
            conn_params.min_int = 0x20;    // min_int = 0x10*1.25ms = 20ms
            // conn_params.timeout = 1000;    // timeout = 400*10ms = 4000ms
            conn_params.timeout = 2000;    // timeout = 400*10ms = 4000ms
            //start sent the update connection parameters to the peer device.
            esp_ble_gap_update_conn_params(&conn_params);
...
```

chegewara, 2019-05-28:
You have setup it correctly in your code, which means its setup in nRF connect:
`conn_params.timeout = 2000;` <--- 20 sec

Check in nRF connect logs, also try to change it in `Request connection priority` after connecting to esp32.

Weijian-Espressif, 2019-05-30:
@zhanlej,  Your updated connection parameters are wrong. ` update connection params status = 15`,   `ESP_BT_STATUS_PARAM_OUT_OF_RANGE` indicates a connection update error.

chegewara, 2019-05-30:
Nice catch.

zhanlej, 2019-05-31:
@Weijian-Espressif, Thank you very much, solved my problem.

closed 2019-05-31 by zhanlej