espressif/esp-idf #3552: "error: ping.h: No such file or directory" with release v3.2

rahulbpatel8, 2019-05-28:
### Problem
Not able to compile release v3.2 while using ping.h. it is giving error "error: ping.h: No such file or directory", not able to find its reference.

### Possible solution
Looked into component.mk file of lwip, **include/apps/ping** is missing from COMPONENT_ADD_INCLUDEDIRS. Adding it resolved the error.

`COMPONENT_ADD_INCLUDEDIRS := \
	include/apps \
	include/apps/ping \
	lwip/src/include \
	port/esp32/include \
	port/esp32/include/arch \
	include_compat`

Also checked with latest master branch, this change is not there.

Can anyone check the issue.? Let me know if more information required or I am missing something.

Thanks.

rahulbpatel8, 2019-05-30:
Hello,
Has anyone looked into this issue.?

renzbagaporo, 2019-05-30:
Hi, can you work around this problem by including "ping/ping.h" ? It seems that even [apps/ping/ping.c](https://github.com/espressif/esp-idf/blob/da13efc17a73e773d8da6cd95c076fadabefc2cc/components/lwip/apps/ping/ping.c#L45) use it.

rahulbpatel8, 2019-05-30:
Hi, thanks for the response. Your suggestion is working.

closed 2019-05-30 by rahulbpatel8