espressif/esp-idf #3553: manufacturing tool

chegewara, 2019-05-28:
Hi, i am trying to learn how to work with manufacturing tool and im having issue on beginning. 
I know nothing about python, sorry. 
My environment is windows 10 and msys2 and here is my issue:
```
$ python ~/esp/esp-idf/tools/mass_mfg/mfg_gen.py --keygen true
Traceback (most recent call last):
  File "F:/esp32_master/home/darek/esp/esp-idf/tools/mass_mfg/mfg_gen.py", line 549, in <module>
    main()
  File "F:/esp32_master/home/darek/esp/esp-idf/tools/mass_mfg/mfg_gen.py", line 366, in main
    print_encrypt_arg_str, output_dir_path)
TypeError: check_input_args() takes at most 9 arguments (10 given)
```
Could you take a look and help me @projectgus, thanks.

dobairoland, 2019-05-29:
Hi @chegewara. The above command works well for me. Please make sure you are on latest master. We have fixed several related issues very recently. Not all the fixes were backported yet.

chegewara, 2019-05-29:
hi @dobairoland 
i believe i am on right branch (updated now, to make sure), but still no luck:
```
$ git show
commit 80fb837b29f8518892b44de1b111c8a0dbb912f4 (HEAD -> master, origin/master, origin/HEAD)
Merge: 9d0ca6039 d97d3c2bc
Author: Angus Gratton <angus@espressif.com>
Date:   Wed May 29 08:10:31 2019 +0800

    Merge branch 'bugfix/kconfig_checker_message' into 'master'

    Tools: Produce a clearer message for Kconfig prefix errors

    See merge request idf/esp-idf!5054

darek@DESKTOP-6NC7CO8 MINGW32 ~/esp/esp-idf
$ python ~/esp/esp-idf/tools/mass_mfg/mfg_gen.py
Traceback (most recent call last):
  File "F:/esp32_master/home/darek/esp/esp-idf/tools/mass_mfg/mfg_gen.py", line 549, in <module>
    main()
  File "F:/esp32_master/home/darek/esp/esp-idf/tools/mass_mfg/mfg_gen.py", line 366, in main
    print_encrypt_arg_str, output_dir_path)
TypeError: check_input_args() takes at most 9 arguments (10 given)
```
Thanks

dobairoland, 2019-05-29:
Thank you for clarifying this. I checked out the same commit you have and it still works for me on both Linux and Win 10 + MSYS2. We will look into this more deeply...

shivani-tipnis, 2019-05-29:
Hi, your nvs_partition_utility seems to be not in sync with the manufacturing utility, it seems to be few commits behind. Could you please check the latest commit for the nvs utility.

chegewara, 2019-05-29:
Hi @shivani-tipnis 
any tip how to do it, i am using `git submodule update --init --recursive` in esp-idf, unless i have to update toolchain?

shivani-tipnis, 2019-05-30:
Hi @chegewara 
That should be fine. 
Could you let me know the commit id currently for the utility.

chegewara, 2019-05-30:
@shivani-tipnis nvs tools, manufacturing tools and other tools are part of esp-idf, not an included submodule from separate repo:
https://github.com/espressif/esp-idf/tree/master/tools

@angu help please.

shivani-tipnis, 2019-05-30:
@chegewara Yes, that is why if you could share the latest log commit of the util, that would be good.
The error you have shared will occur if the nvs util is behind by a few commits.

shivani-tipnis, 2019-05-30:
@chegewara Could you please confirm your IDF_PATH.

chegewara, 2019-05-30:
> @chegewara Could you please confirm your IDF_PATH.

That was it, what i did is download 1 toolchain and then i made few copies to have separate toolchain for different esp-idf versions. With git pull i updated esp-idf in one folder (good one) but IDF_PATH is pointing to different folder (old one).

Many thanks.

closed 2019-05-30 by chegewara