espressif/esp-idf #3554: esp_vfs_fat_register and httpd not working together without USE_ONLY_LWIP_SELECT

felixcollins, 2019-05-28:
## Environment

- Development Kit:      Custom Board - see main.c line 20 for SD gpio assignment (this should be changed for your dev board)

- Module or chip used:  [ESP32-WROOM-32|
- IDF version v3.2-103-g4aa1058e8
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0

- Operating System:     Windows 10
- Power Supply:         custom board - powered from USB port with  linear reg on board (TC1301B)

## Problem Description
esp_vfs_fat_register and httpd not working together without USE_ONLY_LWIP_SELECT

I have been having a problem where the httpd will not work after calling esp_vfs_fat_sdmmc_mount(). It seems the problem can be fixed by running menuconfig and choosing the "lwip select only" option. Component config / LWIP /USE_ONLY_LWIP_SELECT [=y].

It seems the problem is that (without the lwip select option) the select() call in the httpd returns data on the wrong file descriptor after esp_vfs_fat_sdmmc_mount() is called.

I've attached a zip file with a minimal repro. See comment at line 71 in main.c to break the example.

### Expected Behavior
httpd and esp_vfs_fat_sdmmc_mount() do not interfere

### Actual Behavior
httpd stops working after esp_vfs_fat_sdmmc_mount() is called

### Steps to repropduce
1. build and run example 
2. connect to ESPTest-xxxxx access point.
3. web browser - http://192.168.4.1/hello
4. Observe page loads
5. Uncomment line 72 of main.c
6. build and run example 
7. connect to ESPTest-xxxxx access point.
8. web browser - http://192.168.4.1/hello
9. Observe page fails to load

### Code to reproduce this issue
[brokenwebserverexample.zip](https://github.com/espressif/esp-idf/files/3230120/brokenwebserverexample.zip)

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3230152/sdkconfig.zip)

igrr, 2019-05-29:
Linking https://github.com/espressif/esp-idf/issues/3149 which is possibly related (was fixed in master only, hasn't been backported to v3.2).

mvonflotow, 2019-09-17:
I have experienced the same issue, and the same USE_ONLY_LWIP_SELECT seems to work for me.

dobairoland, 2019-10-14:
Thank you @felixcollins for reporting the issue and giving a very useful use case to reproduce it. I found the root cause of the bug and we will push the fix ASAP to master and will be backported to older IDF versions as well.

I'm sorry that it took us too long to fix it.

closed 2019-10-25 by igrr