espressif/esp-idf #3555: NimBLE - how to initialize stack to run in main application task

bfriedkin, 2019-05-29:
Hello -

We are wrapping the NimBLE server/client implementations with our own platform-independent API. To initialize the NimBLE stack, we are following the examples provided:

```
void nimble_host_task(void *param)
{
	nimble_port_run();
}

void ble_host_task(void *param)
{
	nimble_host_task(param);
}
...
nimble_port_freertos_init(ble_host_task);
```

That all works well, but our callbacks execute in a separate task. In nearly every case, this hasn't been an issue. Unfortunately on the server side, we bump into a problem handling the `BLE_GATT_ACCESS_OP_READ_CHR` request from the access callback. Specifically, the server characteristic read function is running in a different task and it seems that the read needs to be satisfied in the access callback. The `om` data structure needs to be filled in from the callback. Hence we cannot post a message to a different task to read the characteristic and have the read result be asynchronously returned.

I suppose my first question is whether or not my understanding is correct. Is there a way to satisfy a characteristic read request outside the `ble_gatt_access_fn`?

If the read request must be satisfied from the callback, I would like to explore running the NimBLE stack such that the callbacks execute from the main task, i.e. the task that calls `nimble_port_freertos_init()`. It looks like the generic NimBLE examples might do this. For example, the generic `bleprph` example loops forever in the `main()` function:

```
    /*
     * As the last thing, process events from default event queue.
     */
    while (1) {
        os_eventq_run(os_eventq_dflt_get());
    }
```

Is it possible to service the BLE events from the main task on ESP32. If so, how does the initialization sequence change to make that work?

Of course any other suggestions to satisfy a read request from the main task would be more than welcome.

Regards,
Brian

dhrishi, 2019-05-29:
@bfriedkin The read response has to be returned in context of the same task (synchronously).

If you just want to process the read request in a different task, you can post a message to that task and wait in the original task till you get a response and then send that response. But again, this will make the flow synchronous. 

If you want to use main task to run the NimBLE host then:
1. Do not call `nimble_port_freertos_init()`
2. At the end of your `app_main()` you can call `nimble_port_run()`

Note that, with this, your main task will always be blocked on BLE events and will only service those without doing anything else. Also, you will have to manage the stack size of this task as per the requirements of BLE functionality.

bfriedkin, 2019-05-29:
@dhrishi -

Thank you for the follow-up.

I decided to use the cross-task approach where I post a message and wait for the response. That works well.

That said, I was also able to confirm that servicing the BLE event queue from the main task works as well. Our SDK supports timer-based callbacks that get executed from the main event loop. So instead of calling `nimble_port_run` I run the following code from a timer callback that executes every 20 ms:

```
struct ble_npl_eventq *eventq = nimble_port_get_dflt_eventq();
struct ble_npl_event *ev;
ev = ble_npl_eventq_get(eventq, 0);
if (NULL != ev)
   ble_npl_event_run(ev);
```

Works great.

Regards,
Brian

closed 2019-05-29 by bfriedkin