espressif/esp-idf #3557: asio link error: undefined reference to if_indextoname(unsigned int, char*) using esp32-2019r1-rc2

tgill880, 2019-05-29:
## Environment

- Development Kit:     Adafruit HUZZAH32
- Module or chip used:  ESP32-WROOM-32
- IDF version:   esp32-2019r1-rc2
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1-rc2) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description
C++17 network library demo that uses `asio` fails with link error

I am porting the demo programs from a C++17 header only network library, [chops-net-ip](https://github.com/connectivecpp/chops-net-ip), to the `ESP32`. `chops-net-ip` builds on top of `asio`. A demo program that sets up an echo server will not build due to a link error.

` undefined reference to if_indextoname(unsigned int, char*)`

This function is defined in `net/if.h`. Including that library does not fix the error.
`#include <net/if.h>`

The demo program compiles and works correctly on Windows, macOS and Linux.

### Expected Behavior
Program compiles, builds, uploads to the ESP32.

### Actual Behavior
Link error.

### Steps to reproduce
`make flash`

### Code to reproduce this issue 
[GIST](https://gist.github.com/tgill880/177236582fb1c80f7e3cdc039f5bd10c
The link error is triggered by line 135.
`iof.start_io(HDR_SIZE, msg_hndlr, msg_frame);`

## Debug Logs

```
Toolchain path: /home/thurman/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: esp32-2019r1-rc2
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from /home/thurman/esp/esp-idf/requirements.txt are satisfied.
WARNING: Toolchain version is not supported: esp32-2019r1-rc2
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.

Project is not inside a git repository, will not use 'git describe' to determine PROJECT_VER.
App "chops_echo_server" version: 1
LD build/chops_echo_server.elf
/home/thurman/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/thurman/Projects/ESP32/chops_echo_server/build/asio/libasio.a(asio.o):(.literal._ZN4asio6detail10socket_ops9inet_ntopEiPKvPcjmRSt10error_code+0x4): undefined reference to `if_indextoname(unsigned int, char*)'
/home/thurman/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/thurman/Projects/ESP32/chops_echo_server/build/asio/libasio.a(asio.o): in function `asio::detail::socket_ops::inet_ntop(int, void const*, char*, unsigned int, unsigned long, std::error_code&)':
/home/thurman/esp/esp-idf/components/asio/asio/asio/include/asio/detail/impl/socket_ops.ipp:2123: undefined reference to `if_indextoname(unsigned int, char*)'
collect2: error: ld returned 1 exit status
/home/thurman/esp/esp-idf/make/project.mk:520: recipe for target '/home/thurman/Projects/ESP32/chops_echo_server/build/chops_echo_server.elf' failed
make: *** [/home/thurman/Projects/ESP32/chops_echo_server/build/chops_echo_server.elf] Error 1

```

## Other items if possible

Alvin1Zhang, 2019-05-29:
@tgill880 Thanks for reporting the issue, we will look into it. Thanks.

david-cermak, 2019-06-09:
Hi @tgill880 

Thanks for reporting this issue. The problem is in ASIO port which does not correctly address ESP32 platform for all scenarios.

As a workaround (before fixed in ESP-IDF) you could supply a dummy definition of `if_indextoname` returning nullptr, such as:
```
char* if_indextoname(unsigned int , char* ) { return 0; }
```

tgill880, 2019-06-11:
Thanks, our program now links and loads onto the ESP32.

We are trying to port [chops-net-ip](https://connectivecpp.github.io).  Our library runs on top of `asio`. However, we now have 2 TCP server demo programs that do not work correctly. In both, messages are correctly received by the server (running on the ESP32), but replies sent back to the client are garbled or lost. Both programs run correctly on Linux, macOS and Windows.

We suspect the `asio` port is either broken or incomplete.

I will post additional issues when we have more information.

closed 2019-06-11 by tgill880