espressif/esp-idf #3558: CoAP server example logging level

Noxet, 2019-05-29:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.2.0-46-g552fc37
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

When running the example code for the coap server, using logging level 9, and sending data, i.e., a GET, PUT or DELETE, the system crashes and reboots.
Data is being sent from the libcoap test client, running on Arch.

### Expected Behavior
System should not crash/reboot when receiving data

### Actual Behavior
System crashes/reboots upon receiving data

### Steps to repropduce

1. Set the `COAP_LOGGING_LEVEL` to 9.
2. Run `make flash monitor`
3. Send some data from the client, `coap-client -m get coap://<IP>/Espressif`

### Code to reproduce this issue

```cpp
#include "coap.h"

/* Set this to 9 to get verbose logging from within libcoap */
#define COAP_LOGGING_LEVEL 9

static char espressif_data[100];
static int espressif_data_len = 0;

...

```

## Debug Logs

```
Jan 01 00:08:48 DEBG created UDP  endpoint 0.0.0.0:5683
Jan 01 00:08:48 DEBG created TCP  endpoint 0.0.0.0:5683
Jan 01 00:08:50 DEBG ***0.0.0.0:5683 <-> 192.168.12.1:49402 UDP : new incoming session
/home/noxet/esp32/esp-idf/components/freertos/queue.c:1449 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x4008c155 on core 0
0x4008c155: xQueueGenericReceive at /home/noxet/esp32/esp-idf/components/freertos/queue.c:2044

ELF file SHA256: dcc5282abf4d51b791b63d736a0e380b80b7e3551c917b023e00b5707f097f94

Backtrace: 0x40087c20:0x3ffc5f70 0x40087e6d:0x3ffc5f90 0x4008c155:0x3ffc5fb0 0x400827ef:0x3ffc5ff0 0x400828bd:0x3ffc6020 0x400e0e35:0x3ffc6040 0x40082694:0x3ffc6060 0x400e0f51:0x3ffc6080 0x400e8f5f:0x3ffc60b0 0x400e933f:0x3ffc60e0 0x400eb222:0x3ffc6940 0x400eb7fd:0x3ffc6f70 0x400ef8fb:0x3ffc6f90 0x400d31bb:0x3ffc70f0 0x4008c3d1:0x3ffc7140
0x40087c20: invoke_abort at /home/noxet/esp32/esp-idf/components/esp32/panic.c:716

0x40087e6d: abort at /home/noxet/esp32/esp-idf/components/esp32/panic.c:716

0x4008c155: xQueueGenericReceive at /home/noxet/esp32/esp-idf/components/freertos/queue.c:2044

0x400827ef: lock_acquire_generic at /home/noxet/esp32/esp-idf/components/newlib/locks.c:159

0x400828bd: _lock_acquire at /home/noxet/esp32/esp-idf/components/newlib/locks.c:167

0x400e0e35: get_adjusted_boot_time at /home/noxet/esp32/esp-idf/components/newlib/time.c:408

0x40082694: _gettimeofday_r at /home/noxet/esp32/esp-idf/components/newlib/time.c:408

0x400e0f51: clock_gettime at /home/noxet/esp32/esp-idf/components/newlib/time.c:418

0x400e8f5f: coap_ticks at /home/noxet/esp32/esp-idf/components/coap/libcoap/src/coap_time.c:85

0x400e933f: coap_log_impl at /home/noxet/esp32/esp-idf/components/coap/libcoap/src/coap_debug.c:831

0x400eb222: coap_read_endpoint at /home/noxet/esp32/esp-idf/components/coap/libcoap/src/net.c:185

0x400eb7fd: coap_read at /home/noxet/esp32/esp-idf/components/coap/libcoap/src/net.c:185

0x400ef8fb: coap_run_once at /home/noxet/esp32/esp-idf/components/coap/port/coap_io.c:1371

0x400d31bb: coap_example_thread at /home/noxet/squidward/esp32-examples/fota_coap/main/main.c:141

0x4008c3d1: vPortTaskWrapper at /home/noxet/esp32/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57
rts Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6340
load:0x40078000,len:11304
load:0x40080400,len:6700
entry 0x40080764
I (29) boot: ESP-IDF v4.0-dev-621-g81ca1c013 2nd stage bootloader
I (29) boot: compile time 10:23:00
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1ba64 (113252) map
I (133) esp_image: segment 1: paddr=0x0002ba8c vaddr=0x3ffb0000 size=0x030d4 ( 12500) load
I (139) esp_image: segment 2: paddr=0x0002eb68 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/noxet/esp32/esp-idf/components/freertos/xtensa_vectors.S:1779

I (141) esp_image: segment 3: paddr=0x0002ef70 vaddr=0x40080400 size=0x010a0 (  4256) load
I (151) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x72c2c (470060) map
0x400d0018: _stext at ??:?

I (322) esp_image: segment 5: paddr=0x000a2c4c vaddr=0x400814a0 size=0x0e824 ( 59428) load
0x400814a0: tick_hook at /home/noxet/esp32/esp-idf/components/esp32/int_wdt.c:46

I (357) boot: Loaded app from partition at offset 0x10000
I (357) boot: Disabling RNG early entropy source...
I (357) cpu_start: Pro cpu up.
I (361) cpu_start: Application information:
I (366) cpu_start: Project name:     coap_server
I (371) cpu_start: App version:      a1c1091-dirty
I (377) cpu_start: Compile time:     May 29 2019 10:22:59
I (383) cpu_start: ELF file SHA256:  dcc5282abf4d51b7...
I (389) cpu_start: ESP-IDF:          v4.0-dev-621-g81ca1c013
I (395) cpu_start: Starting app cpu, entry point is 0x40081058
0x40081058: call_start_cpu1 at /home/noxet/esp32/esp-idf/components/esp32/cpu_start.c:267

I (387) cpu_start: App cpu up.
I (406) heap_init: Initializing. RAM available for dynamic allocation:
I (413) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (419) heap_init: At 3FFB9208 len 00026DF8 (155 KiB): DRAM
I (425) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (431) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (438) heap_init: At 4008FCC4 len 0001033C (64 KiB): IRAM
I (444) cpu_start: Pro cpu start user code
I (126) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (196) wifi: wifi driver task: 3ffc0af0, prio:23, stack:3584, core=0
I (196) wifi: wifi firmware version: a13dcc4
I (196) wifi: config NVS flash: enabled
I (206) wifi: config nano formating: disabled
I (206) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (216) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (246) wifi: Init dynamic tx buffer num: 32
I (246) wifi: Init data frame dynamic rx buffer num: 32
I (246) wifi: Init management frame dynamic rx buffer num: 32
I (246) wifi: Init management short buffer num: 32
I (256) wifi: Init static rx buffer size: 1600
I (256) wifi: Init static rx buffer num: 10
I (256) wifi: Init dynamic rx buffer num: 32
I (266) example_connect: Connecting to test...
I (356) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (356) wifi: mode : sta (30:ae:a4:d3:46:10)
I (486) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (486) wifi: state: init -> auth (b0)
I (1486) wifi: state: auth -> init (200)
I (1486) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (1486) example_connect: Wi-Fi disconnected, trying to reconnect...
I (3546) example_connect: Wi-Fi disconnected, trying to reconnect...
I (3786) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (3786) wifi: state: init -> auth (b0)
I (3796) wifi: state: auth -> assoc (0)
I (3806) wifi: state: assoc -> run (10)
I (3806) wifi: connected with test, channel 1, bssid = a4:34:d9:aa:21:35
I (3806) wifi: pm start, type: 1

I (4696) tcpip_adapter: sta ip: 192.168.12.164, mask: 255.255.255.0, gw: 192.168.12.1
I (5196) example_connect: Connected to test
I (5196) example_connect: IPv4 address: 192.168.12.164
I (5196) example_connect: IPv6 address: fe80:0000:0000:0000:32ae:a4ff:fed3:4610
Jan 01 00:11:25 DEBG created UDP  endpoint 0.0.0.0:5683
Jan 01 00:11:25 DEBG created TCP  endpoint 0.0.0.0:5683

```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3231601/sdkconfig.txt)
- [x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
[coap_server.elf.txt](https://github.com/espressif/esp-idf/files/3231608/coap_server.elf.txt)
- [ ] coredump (This provides stacks of tasks.)

mahavirj, 2019-08-21:
@Noxet Sorry for delayed reply. We have refactored CoAP server/client examples recently in IDF `master` branch, can you please try once? (I have verified that enabling logging does not cause any issues, but would be great if you can re-confirm)

mahavirj, 2019-08-28:
Closing due to lack of response.

closed 2019-08-28 by mahavirj