espressif/esp-idf #3559: Can't build C++14/17 code with GCC 8.2 and toolchain 2019r1-rc2

marcojez, 2019-05-29:
## Environment

- IDF version: v4.0-dev-653-g80fb837b2
- Build System: Make
- Compiler version: 8.2.0 (crosstool-NG esp32-2019r1-rc2)
- Operating System: Windows

## Problem Description

Building code in C++14 or C++17 mode produces linker errors (undefined reference to `nanosleep') when std::thread's sleep_*() functions are used. 

### Expected Behavior

The build process completes succesfully.

### Actual Behavior

The build process fails:

```
LD build/toolchain_bug.elf
c:/msys32-esp32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/msys32-esp32/home/marco/src/toolchain_bug/build/main\libmain.a(main.o):(.literal.app_main+0xc): undefined reference to `nanosleep'
c:/msys32-esp32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/msys32-esp32/home/marco/src/toolchain_bug/build/main\libmain.a(main.o): in function `app_main':
c:\msys32-esp32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.2.0/thread:369: undefined reference to `nanosleep'
collect2.exe: error: ld returned 1 exit status
make: *** [C:\esp-idf/make/project.mk:521: /home/marco/src/toolchain_bug/build/toolchain_bug.elf] Error 1

```
### Steps to repropduce

1. create a minimal project with a single source file:

* Makefile
* main
  * main.cpp
  * component.mk

2. run `make defconfig`
3. run `make`

### Code to reproduce this issue

Makefile:

```
PROJECT_NAME := toolchain_bug
include $(IDF_PATH)/make/project.mk

```

main/main.cpp:

```cpp
#include <thread>
#include <chrono>
#include <iostream>

extern "C" void app_main()
{
	std::cout << "Please wait...\n";
	std::this_thread::sleep_for(std::chrono::seconds{3});
	std::cout << "Done.\n";
}
```

main/component.mk:

`CXXFLAGS += -std=c++17`

Alvin1Zhang, 2019-05-30:
@marcojez Thanks for reporting the issue, would you help check whether [issue 3557](https://github.com/espressif/esp-idf/issues/3557) and [issue 3467](https://github.com/espressif/esp-idf/issues/3467) are similar to yours? Thanks.

igrr, 2019-05-30:
@marcojez We are fixing this in the final release of esp32-2019r1 toolchain. The bug exists on Windows, macOS and linux-arm. Toolchains for x86/x64 Linux don't have this bug.

marcojez, 2019-05-30:
@Alvin1Zhang I don't think they are related.
@igrr Thank you. Is there any workaround I can apply in the meanwhile?

igrr, 2019-05-30:
@marcojez I think you can work around this by replacing `libstdc++.a`.

Download a copy of the Linux x86-64 toolchain, extract it somewhere, find `libstdc++.a`, there will be 2 copies:
xtensa-esp32-elf/xtensa-esp32-elf/lib/esp32-psram/libstdc++.a
xtensa-esp32-elf/xtensa-esp32-elf/lib/libstdc++.a

Then copy them to the corresponding locations of your Windows toolchain, replacing the original libstdc++.a files.

marcojez, 2019-05-30:
> @marcojez I think you can work around this by replacing `libstdc++.a`.
>
> Download a copy of the Linux x86-64 toolchain, extract it somewhere, find `libstdc++.a`, there will be 2 copies: [...]

Unfortunately, that doesn't seem to do the trick. Still getting the same linker error...

igrr, 2019-05-30:
Ah, sorry about that, the difference is also in the header files. Can you please give the following version of the toolchain a try?

https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-win32.zip

(It is still being tested, not necessarily the final release, but should work in place of the -rc2 version.)

marcojez, 2019-05-30:
> Ah, sorry about that, the difference is also in the header files. Can you please give the following version of the toolchain a try? [...]

My code builds correctly with the updated toolchain, thanks @igrr!.

mirronelli, 2019-05-31:
I had a similar issue with undefined reference in sntp.h
I fixed it by adding a cpluplus guard as is present in other files.
that fixed it for me and. Both c++ and c projects now build and work.

The file i fixed was:
INSTALLDIR\esp\esp-idf\components\lwip\include\apps\sntp\sntp.h

this is what i added:
after initial defines
```
#ifdef __cplusplus
extern "C"
{
#endif
```

and the closing bracket at the bottom of the file

```
#ifdef __cplusplus
}
#endif
```

igrr, 2019-06-03:
@marcojez Thanks for the feedback. We will keep this issue open until the new version of the toolchain  (the one you have tried, or newer) is released.

igrr, 2019-06-14:
Has been updated in https://github.com/espressif/esp-idf/commit/ca323fbaad540cf66cb267c0f749ad24f3a9b211.

closed 2019-06-14 by igrr