espressif/esp-idf #3560: crash when esp_bt_controller_init(&bt_cfg) is called

kvbulusu, 2019-05-29:
----------------------------- Delete below -----------------------------

## Environment

Development Kit: ESP32-LyraT
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.0-dev-621-g81ca1c013
Build System: Make
Compiler version (run xtensa-esp32-elf-gcc --version to find it): 5.20
Operating System: Linux
Power Supply: USB

## Problem Description

we are facing a couple of issues while integrating ibeacon example with amazon-alexa example code from (https://github.com/espressif/esp-va-sdk/tree/master/examples/amazon_alexa). Any suggestions would be helpful. 

Issue details;
As recommended in this issue (espressif/esp-idf#3179) , we turn off/disable/deinit wifi before we turn on BT, but alexa asst code crashes immediately. Below is the code snippet and crash snippet. It crashes with esp_bt_controller_init(&bt_cfg)

The example standalone works and when integrated with wifi based prov process (earlier ver esp-va-sdk ). Their latest code uses BLE based provisioning process. 

### Expected Behavior

### Actual Behavior

### Steps to repropduce

conn_mgr_prov_mem_release();
 app_prov_done_cb();
 wifi_init_sta();
 va_led_set(VA_CAN_START);
 va_board_init();
 alexa_init(va_cfg);
ESP_ERROR_CHECK(esp_wifi_stop() );
ESP_ERROR_CHECK(esp_wifi_deinit() );
// Not sure if BT is disabled/deinit after BLE based prov is done...
esp_bluedroid_disable();
esp_bluedroid_deinit();
esp_bt_mem_release(ESP_BT_MODE_CLASSIC_BT);
esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
esp_bt_controller_init(&bt_cfg);
esp_bt_controller_enable(ESP_BT_MODE_BTDM);
 /* set scan parameters */
 #if (IBEACON_MODE == IBEACON_RECEIVER)
            esp_ble_gap_set_scan_params(&ble_scan_params);
          // Restart wifi .
         wifi_init_sta();
          //vTaskDelay(1000 / portTICK_PERIOD_MS);
 #elif (IBEACON_MODE == IBEACON_SENDER)
            esp_ble_ibeacon_t ibeacon_adv_data;
            esp_err_t status = esp_ble_config_ibeacon_data (&vendor_config, &ibeacon_adv_data);
            if (status == ESP_OK){
                   esp_ble_gap_config_adv_data_raw((uint8_t*)&ibeacon_adv_data, 
                 sizeof(ibeacon_adv_data));
            }
           else {
                ESP_LOGE(BEACON_TAG, "Config iBeacon data failed: %s\n", esp_err_to_name(status));
            }
        #endif
    
### Code to reproduce this issue
## Debug Logs
assertion "res == coreID || res == portMUX_FREE_VAL" failed: file "/home/Firmware/19May09/esp-idf/coGuru Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC : 0x4009a3c9 PS : 0x00060234 A0 : 0x8009b5cf A1 : 0x3facd5e0
0x4009a3c9: uxPortCompareSet at /home/Firmware/19May09/esp-idf/components/freertos/tasks.c:5064
(inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /home/Firmware/19May09/esp-idf/components/freertos/portmux_impl.inc.h:86
(inlined by) vPortCPUAcquireMutexIntsDisabled at /home/Firmware/19May09/esp-idf/components/freertos/portmux_impl.h:99
(inlined by) vTaskEnterCritical at /home/Firmware/19May09/esp-idf/components/freertos/tasks.c:4202

[... 66 lines cut ...]

0x401387b9: avs__avsconfig_payload__free_unpacked at ??:?

Entering gdb stub now.
$T06#baGNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later http://gnu.org/licenses/gpl.html
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law. Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
http://www.gnu.org/software/gdb/bugs/.
Find the GDB manual and other documentation resources online at:
http://www.gnu.org/software/gdb/documentation/.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/Firmware/19May09/esp-va-sdk/examples/demo1/build/alexa.elf...done.
Remote debugging using /dev/ttyUSB0
uxPortCompareSet (set=, compare=3007315967, addr=0x3ffbeb04 )
at /home/Firmware/19May09/esp-idf/components/freertos/include/freertos/portmacro.h:352
352	asm volatile (
(gdb) q
A debugging session is active.

Weijian-Espressif, 2019-05-30:
@kvbulusu , From the log analysis you provided, the crash happens in `esp_tls_conn_delete`, Do you disconnect WIFI before calling `esp_wifi_stop()` ?

kvbulusu, 2019-05-30:
No I don't disconnect wifi.. May be it lost connection..

Weijian-Espressif, 2019-05-30:
@kvbulusu , From the log analysis you provided, this issue is not caused by Bluetooth. Please check if the `sh2lib_free()` function is called ,  check if the parameters passed in are wrong.

closed 2019-06-01 by kvbulusu