espressif/esp-idf #3564: CMake: Cannot specify EXTRA_COMPONENT_DIRS for my components when building unit-test-app

bpietsch, 2019-05-30:
When using the old make based system, I can include my own components in the unit-test-app project by adding the EXTRA_COMPONENT_DIRS environment variable when invoking make. The same does not seem to work when using CMake. How do I add my own components to the unit-test app when using CMake (short of modifying the unit-test-app CMakeLists.txt file)?

renzbagaporo, 2019-05-30:
Hi @bpietsch, you can pass `EXTRA_COMPONENT_DIRS` when invoking `idf.py`:

```
idf.py build -DEXTRA_COMPONENT_DIRS=...
```

bpietsch, 2019-05-30:
Yup... the help for idf.py doesn't particularly make it obvious that the -D option serves that purpose. Thanks for the quick reply!

closed 2019-05-30 by bpietsch

renzbagaporo, 2019-05-30:
The -D option can pass any cache entry in general, not just EXTRA_COMPONENT_DIRS, similar to passing -D arguments for bare CMake (in fact this is just what idf.py does behind the scenes)