espressif/esp-idf #3565: Use the stat function to get the file time attribute, the value obtained is very different from the time obtained by the gettimeofday function.

GitOfLance, 2019-05-30:
### Environment
- Module or chip used: ESP32-WROOM-32
- IDF version : v4.0-dev-207-g76da10a66
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

### Problem Description
After connecting to the `SNTP` server and synchronizing the time, the time attribute of the file is obtained by `stat`, and the obtained value is incorrect.

### Log Information
```
This file [/spiffs/mqtt_pubrecordlog.txt] last modifate time is 0 -1 0 1073457348 -2146593045 1073577800, size = 122
W (1619951) device_info: [ ilock_use_gettimeofday ]Time is real time
E (1619961) device_info: The current time is Thu May 30 15:06:24 2019
Current time is 1559199984, 1559199985
```
```
This file [/spiffs/mqtt_pubrecordlog.txt] last modifate time is 0 -1 0 1073457212 1 1, size = 37
W (5842) device_info: [ ilock_use_gettimeofday ]Time is real time
E (5842) device_info: The current time is Thu May 30 15:06:52 2019
Current time is 1559200012, 1559200013
```

### Corresponding code
```
    printf("This file [%s] last modifate time is %ld %ld %ld %ld %ld %ld, size = %ld\n", SPIFFS_MQTT_PUBRECORDLOG, st.st_atime, st.st_mtime, st.st_ctime, st.st_spare1, st.st_spare2, st.st_spare3, st.st_size);
    ilock_use_gettimeofday(&currenttime);
    printf("Current time is %ld, %ld\n", currenttime.tv_sec, currenttime.tv_sec - st.st_mtime);
```

```
int ilock_use_gettimeofday(struct timeval *time)
{
	gettimeofday(time, NULL);
	if (time->tv_sec > 100000000UL)
	{
		ESP_LOGW(TAG, "[ %s ]Time is real time", __FUNCTION__);
		ilock_show_real_time_now();	
		return 0;
	}
	else
	{
		ESP_LOGE(TAG, "[ %s ]Not real time, again", __FUNCTION__);
		vTaskDelay(3000 / portTICK_PERIOD_MS);
		gettimeofday(time, NULL);
		if (time->tv_sec > 100000000UL)
		{
			ESP_LOGE(TAG, "[ %s ]get real time", __FUNCTION__);
			ilock_show_real_time_now();
			return 0;
		}
		else
		{
			return -1;
		}
	}
	
	return 0;
}
```

KonstantinKondrashov, 2019-05-30:
Hi @GitOfLance! What do you mean `the obtained value is incorrect`? Which have to be the expected result? How do you get these vars (st.st_atime, st.st_mtime, st.st_ctime)? Which a function do you use?

GitOfLance, 2019-05-30:
@KonstantinKondrashov 
Sorry, I didn't express it clearly.
Because the time difference between the time I am modifying these files and getting the time with the `gettimeofday` function is not long. The time interval is less than 1 minute. The current time is `1559200012s(Thu May 30 15:06:52 2019)`, so the modification time of this file should be between `(1559200012-60)s` and `1559200012s`, but the result is obviously not this value.
What I want to get is the last time I modified the file, which is `st.st_mtime`. The other values are used for testing.

KonstantinKondrashov, 2019-05-30:
Could you check your sdkconfig file, is there the [`CONFIG_SPIFFS_USE_MTIME`](https://github.com/espressif/esp-idf/blob/master/components/spiffs/Kconfig#L109-L117) option is enabled?

Please, provide your sdkconfig.

GitOfLance, 2019-05-30:
I looked at my configuration file. This option is configured and uses the default configuration.Like this:
[image]

My `sdkconfig` , I converted this file to `txt format` for easy uploading.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3236475/sdkconfig.txt)

`CONFIG_SPIFFS_USE_MTIME=y`
[image]

KonstantinKondrashov, 2019-05-30:
I think [this unit test](https://github.com/espressif/esp-idf/blob/master/components/spiffs/test/test_spiffs.c#L613-L626) looks similar to your use case. After files were updated were they closed (fclose(f) == 0)?

GitOfLance, 2019-05-30:
At the end of every function that involves reading and writing files, I close the file through like this:
```
        ...
        //fopen/fread/fwrite failure
        ret = -1;
        goto ERR_mqtt_sync_information;
        ...
	ret = 0;
	
ERR_mqtt_sync_information:
	if (fp)
	{
		fclose(fp);
		fp = NULL;
	}
	if (string)
	{
		free(string);
		string= NULL;
	}
       ...
        return ret;
```
And these files will be placed in the `while(1)` loop to read and write, so if the file is not closed after reading and writing the file, the next time the `fopen` function is called, an error will be reported, but in fact, it is normal for me to read and write files multiple times in succession.
The file will be read and written after calling the stat function, so stat is before fopen.

KonstantinKondrashov, 2019-05-31:
Did you use `spiffsgen.py` to create these files or create them through the code?

GitOfLance, 2019-05-31:
I create these files through `fopen(FILENAME, "w");`

KonstantinKondrashov, 2019-05-31:
So, try to do it with "wb".

GitOfLance, 2019-05-31:
I change `a+` to `ab+`, `w` to `wb`, but no use:
[image]

Other related code has not been changed.

KonstantinKondrashov, 2019-05-31:
How many files are opened at the same time?     

You have to have an initialisation like this:
```
esp_vfs_spiffs_conf_t conf = {
      .base_path = "/spiffs",
      .partition_label = NULL,
      .max_files = 5, <---------
      .format_if_mount_failed = true
    };
    esp_err_t ret = esp_vfs_spiffs_register(&conf);
```
Is it exceeded (max_files )?

GitOfLance, 2019-05-31:
I initialize the spiffs system like this:
```
	esp_vfs_spiffs_conf_t spiconfig = {
		.base_path = "/spiffs",
		.partition_label = PARTITION_LABEL_SPIFFS,
		.max_files = 20,
		.format_if_mount_failed = true,
		//.format_if_mount_failed = true,
	};
```
And the maximum number of files is no more than ten, so there is no upper limit.

KonstantinKondrashov, 2019-06-03:
Hi @GitOfLance. I checked to our implementation and can say that using `fwrite()` does not lead to the `st_mtime` will be updated. It will be updated only after you close the file. 
open() -> write() -> close() -> stat(). In this case the st_mtime = the opening time.
You can use `utime()` after each `fwrite()` to update the time immediately. Or close the file after each write operation.
```c
#include <sys/stat.h>
#include <sys/utime.h>

static void print_st_mtime(const char *filename) {
    struct stat st;
    stat(filename, &st);
    printf("st_mtime=%ld\n", st.st_mtime);
}

static void my_write(const char *filename) {
    ESP_LOGI(TAG,"__my_write__");
    FILE* f_w = fopen(filename, "w");
    if (f_w == NULL) {
        ESP_LOGE(TAG, "Failed to open file for writing");
        return;
    }
    fwrite("write_1 ", 1, 8, f_w);
    utime(filename, NULL);
    print_st_mtime(filename);
    vTaskDelay(1000 / portTICK_PERIOD_MS);

    fwrite("write_2 ", 1, 8, f_w);
    utime(filename, NULL);
    print_st_mtime(filename);
    vTaskDelay(1000 / portTICK_PERIOD_MS);

    fclose(f_w);
    print_st_mtime(filename);
}
```

KonstantinKondrashov, 2019-08-12:
Hi @GitOfLance. Could you close this issue if it solved? Thanks.

closed 2019-08-13 by GitOfLance