espressif/esp-idf #3566: BLE mesh node fatal-error

devendratank, 2019-05-30:
Hi,
I am simply compiling the example code of BLE mesh node and I am getting the following errors. fatal error: esp_ble_mesh_defs.h: No such file or directory.  Error 1
/project.mk:468: recipe for target 'component-main-build' failed
Please help me on this.

Campou, 2019-05-30:
Hi,
1. Please make sure the whole project has been cloned successfully, use `git clone` and `git submodule update --init --recursive`;
2. Which branch did you use? If the branch is correct, please run `make defconfig` before compiling.

Thanks.

devendratank, 2019-05-30:
From the given link I can see there is BLE_mesh and I have cloned it successfully  
https://github.com/espressif/esp-idf/tree/feature/esp-ble-mesh-v0.5/components/bt/ble_mesh/mesh_docs
But after cloning and reading the readme file I cannot see any example for BLE_mesh in (../../../../examples/bluetooth/ble_mesh/ble_mesh_node)

chegewara, 2019-05-30:
Time to switch to latest BLE MESH branch?
https://github.com/espressif/esp-idf/tree/ble_mesh_release/esp-ble-mesh-v0.6.1

dhrishi, 2019-05-30:
@devendratank After you clone the repository, make sure that you checkout the `feature/esp-ble-mesh-v0.5` branch. The BLE mesh code is present on a separate branch. You should see the examples and the mesh component there.

If you want, you can also use the latest BLE Mesh branch instead `ble_mesh_release/esp-ble-mesh-v0.6.1`

devendratank, 2019-05-31:
Thank you...
It is working after changing the branch to-  ble_mesh_release/esp-ble-mesh-v0.6.1

closed 2019-05-31 by devendratank