espressif/esp-idf #3567: MCPWM Use of Internal Sync

fungo16, 2019-05-30:
### Environment

-     Development Kit: ESP32-DevKitC
-     Kit version : v4
-     Module or chip used: ESP32-WROOM-32
-     IDF version (run git describe --tags to find it): v3.2-dev-1148-g96cd3b75c
-     Build System: Make
-     Compiler version (run xtensa-esp32-elf-gcc --version to find it): 1.22.0-80-g6c4433a
-     Operating System: Windows
-     Power Supply: USB

### Problem Description

I am using the example code from "MCPWM basic config" and issue 3361 to generate 2 PWM signals from Timer 0 and Timer 1 MCPWM Unit 0 with 25% Phase shift of Timer 1 on Timer 0 through the internal sync. If inverted (see code for inverse section), no sync is happening and signals are completely out of phase.

Additionally, do I have to sync again after each change of frequency? 

## Expected Behavior

2 synchronized pwm signals in both configuration with inverted shift
## Actual Behavior

2 pwm signals for inverted case not synchronized
## Steps to repropduce
please see code
###  Code to reproduce this issue

```
    //1. mcpwm gpio initialization
    mcpwm_example_gpio_initialize();

    //2. initialize mcpwm configuration
    printf("Configuring Initial Parameters of mcpwm...\n");

    bool inverse = false;

    mcpwm_config_t pwm_config;
    pwm_config.frequency = 100;    //frequency = 1000Hz
    pwm_config.cmpr_a = 0.0;    //duty cycle of PWMxA = 0%
    pwm_config.cmpr_b = 0.0;    //duty cycle of PWMxb = 0%
    pwm_config.counter_mode = MCPWM_UP_DOWN_COUNTER;
    pwm_config.duty_mode = MCPWM_DUTY_MODE_0;
    mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_0, &pwm_config);
    mcpwm_init(MCPWM_UNIT_0, MCPWM_TIMER_1, &pwm_config);
    printf("Configuration Done \n");
   
    if(inverse)
   {
    mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_0, 1, 500);
    mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_1, 1, 0);
    MCPWM0.timer[1].sync.out_sel = 1;
    vTaskDelay(1000/portTICK_PERIOD_MS);
    MCPWM0.timer[1].sync.out_sel = 0; 
    mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_0,  MCPWM_OPR_A, 50);
    mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_1,  MCPWM_OPR_A, 50);
   } else {
    mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_0, 1, 0);
    mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_1, 1, 500);
    MCPWM0.timer[0].sync.out_sel = 1;
    vTaskDelay(1000/portTICK_PERIOD_MS);
    MCPWM0.timer[0].sync.out_sel = 0;
    mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_0,  MCPWM_OPR_A, 50);
    mcpwm_set_duty(MCPWM_UNIT_0, MCPWM_TIMER_1,  MCPWM_OPR_A, 50);
    }

    vTaskDelete(NULL);
```

negativekelvin, 2019-05-30:
Change this back to timer[1]

```
    MCPWM0.timer[0].sync.out_sel = 1;
    vTaskDelay(1000/portTICK_PERIOD_MS);
    MCPWM0.timer[0].sync.out_sel = 0;
```

fungo16, 2019-06-01:
Thanks for your answer!
But then I sync on the output of timer 0? What does the line exactly represent?

negativekelvin, 2019-06-01:
`mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_0, **1**, 500);`
The 1 means use timer[0] as sync input so you have to match it to
`MCPWM0.timer[0].sync.out_sel = 1;`
Otherwise you have to change it to
`mcpwm_sync_enable(MCPWM_UNIT_0, MCPWM_TIMER_0, **2**, 500);`
And 
`MCPWM0.timer[1].sync.out_sel = 1;`

Alvin1Zhang, 2019-06-21:
@negativekelvin Thanks for the notes. @fungo16 Thanks for reporting the issue, would you help share if any updates for the issue? Has it been resolved? Thanks.

Alvin1Zhang, 2019-07-15:
@fungo16 Thanks for reporting the issue, will close this issue due to few updates. Feel free to reopen if you still have the issue, thanks.

closed 2019-07-15 by Alvin1Zhang