espressif/esp-idf #3568: CMake: confgen.py can't handle hex ranges when generating JSON metadata

devanlai, 2019-05-30:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version: `v4.0-dev-667-gda13efc17`
- Build System:         CMake
- Compiler version: `crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5 5.2.0`
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

When using the CMake build system and a KConfig file with a symbol of type hex and the range of valid values is defined using hex literals, `idf.py menuconfig` fails due to `confgen.py` throwing an exception when it tries to convert the hex literals to integers to output the JSON "range" field. 

I believe that this could be fixed by tweaking the code that outputs the JSON range field to use `int(min_range.str_value, 0)` instead of `int(min_range.str_value)` so that it will accept any well-formed numeric literal that Python accepts.

### Steps to repropduce

1. Create a project with a Kconfig file and add a hex symbol with a range defined using hex literals (see below)
2. Run `idf.py menuconfig`.

### Code to reproduce this issue
Sample `Kconfig.projbuild` to reproduce the issue:

    menu "Test hex"
        config TEST_HEX_RANGE
        hex "A hex value"
        range 0x44 0x45
        default 0x44
    endmenu

## Debug Logs

`confgen.py` traceback:

    Traceback (most recent call last):
      File "C:/Users/devanl/esp/esp-idf/tools/kconfig_new/confgen.py", line 544, in <module>
        main()
      File "C:/Users/devanl/esp/esp-idf/tools/kconfig_new/confgen.py", line 255, in main
        output_function(deprecated_options, config, temp_file)
      File "C:/Users/devanl/esp/esp-idf/tools/kconfig_new/confgen.py", line 501, in write_json_menus
        config.walk_menu(write_node)
      File "C:\Users\devanl\esp\esp-idf\tools\kconfig_new\kconfiglib.py", line 971, in walk_menu
        callback(node)
      File "C:/Users/devanl/esp/esp-idf/tools/kconfig_new/confgen.py", line 469, in write_node
        greatest_range = [int(min_range.str_value), int(max_range.str_value)]
    ValueError: invalid literal for int() with base 10: '0x44'

Alvin1Zhang, 2019-05-31:
@devanlai Thanks for reporting the issue. We will look into it.

ulfalizer, 2019-05-31:
Might want to take the symbol type into account. Something like this will work:

```python
for min_range, max_range, cond_expr in sym.ranges:
    if kconfiglib.expr_value(cond_expr):
        base = 16 if sym.type == kconfiglib.HEX else 10
        greatest_range = [int(min_range.str_value, base),
                          int(max_range.str_value, base)]
        break
```

`hex` values are not required to have a `0x` prefix. This is some silly compatibility with the C tools.

Related note from the `str_value` docstring:

    Gotcha: For int/hex symbols, the exact format of the value must often be
    preserved (e.g., when writing a .config file), hence why you can't get it
    directly as an int. Do int(int_sym.str_value) or
    int(hex_sym.str_value, 16) to get the integer value.

dobairoland, 2019-07-03:
Thank you @devanlai for letting us know about this problem and thank you @ulfalizer for the suggestion. The fix will be added soon.

ulfalizer, 2019-07-04:
You guys should update Kconfiglib someday by the way. Tons of improvements and optimizations, and probably no API changes that'd affect you (there's [a list](https://github.com/ulfalizer/kconfiglib#installation-with-pip) in the README).

Would help avoid user issues [like this one](https://github.com/aws/amazon-freertos/issues/802#issuecomment-504336802) too.

Sorry for nagging.~

closed 2019-07-05 by igrr