espressif/esp-idf #3572: esp_bluedroid_disable() crashes after esp_bt_mem_release(ESP_BT_MODE_BLE)

kvbulusu, 2019-06-01:
## Environment

Development Kit: ESP32-LyraT
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.0-dev-621-g81ca1c013
Build System: Make
Compiler version (run xtensa-esp32-elf-gcc --version to find it): 5.20
Operating System: Linux
Power Supply: USB

## Problem Description
We use BLE based wifi provisioning process (https://github.com/espressif/esp-va-sdk/blob/master/examples/amazon_alexa/main/app_main.c)  and after it's done, we call the below code  but after mem release, esp_bluedroid_disable crashes.

### Steps to repropduce
esp_bt_mem_release(ESP_BT_MODE_BLE);
 esp_bluedroid_disable();
 esp_bluedroid_deinit();
 esp_bt_controller_disable();
 esp_bt_controller_deinit();
 esp_bt_controller_mem_release(ESP_BT_MODE_BLE);

## Debug Logs

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400984e9  PS      : 0x00060d30  A0      : 0x80139729  A1      : 0x3ffbbec0  
0x400984e9: xQueueGenericSend at /home/kvbulusu/SHS/mydev/palmnest/Firmware/19May09/esp-idf/components/freertos/queue.c:2526

A2      : 0x00000008  A3      : 0x3ffbbf00  A4      : 0xffffffff  A5      : 0x00000000  
A6      : 0x03000000  A7      : 0x00000000  A8      : 0x800984dc  A9      : 0x3ffbbea0  
A10     : 0x00000002  A11     : 0x00000000  A12     : 0x000002cf  A13     : 0x3f40ae2f  
A14     : 0x0000000a  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x400970e0  LEND    : 0x4009710e  LCOUNT  : 0xffffffff  
0x400970e0: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:168

0x4009710e: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: 0b41a0de6ba884feeee63ef30d611b271df0b6f9c66889de6c8daf70b059d6b2

Backtrace: 0x400984e9:0x3ffbbec0 0x40139726:0x3ffbbf00 0x4013917d:0x3ffbbf30 0x40108adf:0x3ffbbf60 0x400d0d83:0x3ffbc090
0x400984e9: xQueueGenericSend at /home/kvbulusu/SHS/mydev/palmnest/Firmware/19May09/esp-idf/components/freertos/queue.c:2526

0x40139726: btc_task_post at /home/kvbulusu/SHS/mydev/palmnest/Firmware/19May09/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:129
 (inlined by) btc_transfer_context at /home/kvbulusu/SHS/mydev/palmnest/Firmware/19May09/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:162

0x4013917d: esp_bluedroid_disable at /home/kvbulusu/SHS/mydev/palmnest/Firmware/19May09/esp-idf/components/bt/bluedroid/api/esp_bt_main.c:101

0x40108adf: app_main at /home/kvbulusu/SHS/mydev/palmnest/Firmware/19May09/esp-va-sdk/examples/demo1/main/app_main.c:338

0x400d0d83: main_task at /home/kvbulusu/SHS/mydev/palmnest/Firmware/19May09/esp-idf/components/esp32/cpu_start.c:529

Entering gdb stub now.
$T0b#e6GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/kvbulusu/SHS/mydev/palmnest/Firmware/19May09/esp-va-sdk/examples/demo1/build/alexa.elf...done.
Remote debugging using /dev/ttyUSB0
0x7ffb52b0 in ?? ()
(gdb)

chegewara, 2019-06-01:
Hi, 
you dont have to call both functions to release memory. One is internal function.
```
esp_bt_mem_release(ESP_BT_MODE_BLE);
esp_bt_controller_mem_release(ESP_BT_MODE_BLE);
```
Please remove `esp_bt_mem_release` from your code.

Weijian-Espressif, 2019-06-04:
@kvbulusu, 
1. call ``esp_bt_controller_mem_release(ESP_BT_MODE_BLE)``  instead of ``esp_bt_mem_release(ESP_BT_MODE_BLE)``

2.  This is my test code, it works well.
``
void app_main()
{
    // Initialize NVS.
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        ESP_LOGE(GATTC_TAG, "%s initialize controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
    if (ret) {
        ESP_LOGE(GATTC_TAG, "%s enable controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGE(GATTC_TAG, "%s init bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(GATTC_TAG, "%s enable bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

esp_bluedroid_disable();
esp_bluedroid_deinit();
esp_bt_controller_disable();
esp_bt_controller_deinit();
esp_bt_controller_mem_release(ESP_BT_MODE_BLE);

}
``

kvbulusu, 2019-06-04:
> "%s enable bluetooth failed: %s\n", __func__, esp_err_to_name(ret));

As I mentioned , the crash only happens if we enable BT with the alexa code ...below is the code snippet. I opened a ticket with that repo as well, but thought this was independent , but looks like some dependency. Below is the code snippet which crashes.

#include <freertos/FreeRTOS.h>
#include <freertos/event_groups.h>
#include <esp_wifi.h>
#include <esp_log.h>
#include <esp_event_loop.h>
#include <esp_pm.h>
#include <nvs_flash.h>

#include <conn_mgr_prov.h>
#include <conn_mgr_prov_mode_ble.h>

#include <voice_assistant.h>
#include <alexa.h>

#include <va_mem_utils.h>
#include <scli.h>
#include <va_diag_cli.h>
#include <wifi_cli.h>
#include "app_auth.h"
#include <media_hal.h>
#include <tone.h>
#include <avs_config.h>
#include "va_dsp.h"
#include "va_board.h"
#define SOFTAP_SSID_PREFIX  "ESP-Alexa-"

static const char *TAG = "[app_main]";

static EventGroupHandle_t cm_event_group;
const int CONNECTED_BIT = BIT0;
const int PROV_DONE_BIT = BIT1;

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    conn_mgr_prov_event_handler(ctx, event);

    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        printf("%s: Connected with IP Address: %s\n", TAG, ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
        xEventGroupSetBits(cm_event_group, CONNECTED_BIT);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        printf("%s: Disconnected. Connecting to the AP again\n", TAG);
        esp_wifi_connect();
        break;
    default:
        break;
    }
    return ESP_OK;
}

static void wifi_init_sta()
{
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK(esp_wifi_start() );
    ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));
}

#define MEDIA_HAL_DEFAULT()     \
    {   \
        .op_mode    = MEDIA_HAL_MODE_SLAVE,              \
[... 118 lines cut ...]
            if  (ret)   {
                ESP_LOGE(TAG,  "%s init bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
                return;
            }

            ret =   esp_bluedroid_enable();
            if  (ret)   {
                ESP_LOGE(TAG,  "%s enable bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
                return;
            }
        
            //register the scan callback function to the gap module
            if ((ret = esp_ble_gap_register_callback(esp_gap_cb)) != ESP_OK) {
                ESP_LOGE(TAG, "gap register error: %s", esp_err_to_name(ret));
            } else {
                ESP_LOGW(TAG, "%s gap register done: %d",__func__, ret );
            }

            esp_ble_gap_set_scan_params(&ble_scan_params);
            //esp_bt_controller_mem_release(ESP_BT_MODE_BLE);
    #endif
    va_mem_free(va_cfg);
    return;

}

Weijian-Espressif, 2019-06-05:
@kvbulusu , Sorry , we can not reproduce the issue. Please Please provide all your test code and Tell me the detailed method of reproduction

kvbulusu, 2019-06-27:
@Weijian-Espressif  . This has bit of dependency on the esp-va-sdk git repo and it uses BLE based provisioning. The crashes can only be reproduced after provisioning process is completed.

Steps to reproduce: 
1.git clone https://github.com/espressif/esp-va-sdk.git
2.cd esp-va-sdk/examples/amazaon_alexa
3. Replace the attached file with main/app_main.c 
4. Provision using the espressif android app (https://play.google.com/store/apps/details?id=com.espressif.provbleavs)

You will see the crash.
===
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40092bbb  PS      : 0x00060c30  A0      : 0x80092bf6  A1      : 0x3ffbbe30  
0x40092bbb: multi_heap_internal_lock at DBG/esp-idf/components/heap/multi_heap.c:699

A2      : 0x81004136  A3      : 0x3ffb831c  A4      : 0x8008fac8  A5      : 0x00000001  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x80090cb9  A9      : 0x3ffbbe20  
A10     : 0x00000003  A11     : 0x00060c23  A12     : 0x00060c20  A13     : 0x00000001  
A14     : 0x000000fe  A15     : 0x00060023  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x81004136  LBEG    : 0x4009c04c  LEND    : 0x4009c07a  LCOUNT  : 0x00000000  
0x4009c04c: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:168

0x4009c07a: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:201

Backtrace: 0x40092bbb:0x3ffbbe30 0x40092bf3:0x3ffbbe50 0x400870e5:0x3ffbbe80 0x4008711a:0x3ffbbeb0 0x40087725:0x3ffbbed0 0x4000beaf:0x3ffbbef0 0x401620d2:0x3ffbbf10 0x401261c9:0x3ffbbf50 0x40104579:0x3ffbbf70 0x400d0cd7:0x3ffbc090
0x40092bbb: multi_heap_internal_lock at DBG/esp-idf/components/heap/multi_heap.c:699

0x40092bf3: multi_heap_malloc_impl at DBG/esp-idf/components/heap/multi_heap.c:699

0x400870e5: heap_caps_malloc at DBG/esp-idf/components/heap/heap_caps.c:111

0x4008711a: heap_caps_malloc_default at DBG/esp-idf/components/heap/heap_caps.c:146

0x40087725: _malloc_r at DBG/esp-idf/components/newlib/syscalls.c:37

0x401620d2: btdm_controller_init at ??:?

0x401261c9: esp_bt_controller_init at DBG/esp-idf/components/bt/bt.c:1329

0x40104579: app_main at DBG/esp-avs-sdk/examples/amazon_alexa/main/app_main.c:166

0x400d0cd7: main_task at DBG/esp-idf/components/esp32/cpu_start.c:497
==========

[3572.zip](https://github.com/espressif/esp-idf/files/3333500/3572.zip)

dhruvchikhaliya12, 2019-12-13:
I had the same problem with esp32 alexa.
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6772
load:0x40078000,len:10016
load:0x40080400,len:7072
entry 0x40080750
I (61) boot: Chip Revision: 1
I (64) boot_comm: mismatch chip revision, expect 1, found 0
I (39) boot: ESP-IDF v3.2.3-33-g90af68bd3 2nd stage bootloader
I (39) boot: compile time 13:31:34
I (39) boot: Enabling RNG early entropy source...
I (45) qio_mode: Enabling default flash chip QIO
I (50) boot: SPI Speed      : 80MHz
I (54) boot: SPI Mode       : QIO
I (58) boot: SPI Flash Size : 4MB
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00390000
I (96) boot: End of partition table
I (100) boot_comm: mismatch chip revision, expect 1, found 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1f2884 (2041988) map
I (629) esp_image: segment 1: paddr=0x002028ac vaddr=0x3ffbdb60 size=0x048e8 ( 18664) load
I (635) esp_image: segment 2: paddr=0x0020719c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/dhruv/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (636) esp_image: segment 3: paddr=0x002075a4 vaddr=0x40080400 size=0x08a6c ( 35436) load
I (656) esp_image: segment 4: paddr=0x00210018 vaddr=0x400d0018 size=0x1318f0 (1251568) map
0x400d0018: _stext at ??:?

I (971) esp_image: segment 5: paddr=0x00341910 vaddr=0x40088e6c size=0x0eca0 ( 60576) load
0x40088e6c: ld_inq_frm_isr_wrapper at ??:?

I (1004) boot: Loaded app from partition at offset 0x10000
I (1005) boot: Disabling RNG early entropy source...
E (123) [app_main]: Failed to alloc voice assistant config
abort() was called at PC 0x40102f43 on core 0
0x40102f43: app_main at C:/msys32/home/dhruv/esp/esp-va-sdk/examples/amazon_alexa/main/app_main.c:154 (discriminator 2)

Backtrace: 0x4009155b:0x3ffbc040 0x40091855:0x3ffbc060 0x40102f43:0x3ffbc080 0x400d0ccb:0x3ffbc1a0
0x4009155b: invoke_abort at C:/msys32/home/dhruv/esp/esp-idf/components/esp32/panic.c:707

0x40091855: abort at C:/msys32/home/dhruv/esp/esp-idf/components/esp32/panic.c:707

0x40102f43: app_main at C:/msys32/home/dhruv/esp/esp-va-sdk/examples/amazon_alexa/main/app_main.c:154 (discriminator 2)

0x400d0ccb: main_task at C:/msys32/home/dhruv/esp/esp-idf/components/esp32/cpu_start.c:506

avsheth, 2019-12-13:
Hi @dhruvchikhaliya12
The channel you're referring to, uses a separate custom source. Which IMHO, has very limited functionality (WW detection, full-fledged media streaming support (Saavn, Amazon music etc.), BT streaming being a few). Also, esp-va-sdk has been certified from Amazon which includes passing some of stress tests such as response within a specified time, device running for > 24 hours when put under stress test (such as constant barge-in with media running in background, resuming audio immediately after query has ended, running smoothly on a slow speed network etc.). These functionalities require more buffers.
It could very well be used without PSRAM, but I suspect it may not be able to hold up to all the certification requirements.

dhruvchikhaliya12, 2019-12-15:
> Hi @dhruvchikhaliya12
> The channel you're referring to, uses a separate custom source. Which IMHO, has very limited functionality (WW detection, full-fledged media streaming support (Saavn, Amazon music etc.), BT streaming being a few missing). Also, esp-va-sdk has been certified from Amazon which includes passing some of stress tests such as response within a specified time, device running for > 24 hours when put under stress test (such as constant barge-in with media running in background, resuming audio immediately after query has ended, running smoothly on a slow speed network etc.). These functionalities require more buffers.
> It could very well be used without PSRAM, but I suspect it may not be able to hold up to all the certification requirements.
> 
> P.S. Please raise esp-va-sdk specific issues/concerns on esp-va-sdk repo. :)

I know that but none of that code run on my board and give me error and restart automatically so please help me.
Thanks in advance.

avsheth, 2019-12-16:
Hi @dhruvchikhaliya12
Could you please open a new issue on esp-va-sdk with the failure log, this would help us track the original subject issue and relevant person can look into both the issues.

Thanks

Alvin1Zhang, 2019-12-18:
@kvbulusu @dhruvchikhaliya12 Thanks for reporting. As suggested, will close this ticket, and please help open a new issue on esp-va-sdk at https://github.com/espressif/esp-va-sdk/issues. Thanks.

closed 2019-12-18 by Alvin1Zhang