espressif/esp-idf #3573: Updated CMake build system does not link pre-built binaries

PerMalmberg, 2019-06-02:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version v3
- Module or chip used:  [ESP32-WROOM-32
- IDF version v4.0-dev-667-gda13efc17
- Build System:        CMake
- Compiler version tensa-esp32-elf-g++ (crosstool-NG esp32-2019r1-rc2) 8.2.0
- Operating System:    Linux
- Power Supply:         USB

## Problem Description

This is a an issue related to the updated CMake build system as announced by ESP_renz in this [forum post](https://www.esp32.com/viewtopic.php?p=44234#p43590).

The problem, if I understand it correctly, is that the pre-built binaries for wifi etc that ships with IDF. are not linked to the final binary when IDF is used as a library.

To make it easier to troubleshoot this issue (i.e. without having to get to know my library linked in the discussions on the forum), I've created a minimal project based on the example build_system/cmake/idf_as_lib and wifi/getting_started/station which exhibit the same linker issues as I'm reporting in the forum thread. [You can find that repo here.](https://github.com/PerMalmberg/idf-as-lib-with-wifi-link-issues)

### Expected Behavior

Pre-built binaries should be linked to final binary.
### Actual Behavior

Linked encounters undefined references.

```
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp32/libesp32.a(esp_adapter.c.obj):(.literal.sc_ack_send_wrapper+0x0): undefined reference to `sc_ack_send'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp32/libesp32.a(esp_adapter.c.obj):(.literal.get_time_wrapper+0x0): undefined reference to `os_get_time'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp32/libesp32.a(esp_adapter.c.obj): in function `sc_ack_send_wrapper':
/home/permal/esp/esp-idf/components/esp32/esp_adapter.c:432: undefined reference to `sc_ack_send'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp32/libesp32.a(esp_adapter.c.obj): in function `get_time_wrapper':
/home/permal/esp/esp-idf/components/esp32/esp_adapter.c:403: undefined reference to `os_get_time'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp32/libesp32.a(esp_adapter.c.obj):(.data.g_wifi_osi_funcs+0x110): undefined reference to `os_get_random'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp32/libesp32.a(esp_adapter.c.obj):(.data.g_wifi_osi_funcs+0x118): undefined reference to `os_random'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp32/libesp32.a(esp_adapter.c.obj):(.data.g_wifi_osi_funcs+0x160): undefined reference to `sc_ack_send_stop'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a(ieee80211_api.o):(.text.wifi_deinit_in_caller_task+0x4): undefined reference to `misc_nvs_deinit'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a(ieee80211_api.o):(.text.wifi_deinit_in_caller_task+0x11): undefined reference to `misc_nvs_deinit'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a(ieee80211_api.o):(.text.wifi_init_in_caller_task+0xc): undefined reference to `misc_nvs_init'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a(ieee80211_api.o): in function `wifi_init_in_caller_task':
(.text.wifi_init_in_caller_task+0x80): undefined reference to `misc_nvs_init'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a(ieee80211_api.o):(.text.wifi_osi_funcs_register+0x10): undefined reference to `net80211_printf'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a(ieee80211_api.o): in function `wifi_osi_funcs_register':
(.text.wifi_osi_funcs_register+0x3a): undefined reference to `net80211_printf'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: (.text.wifi_osi_funcs_register+0x54): undefined reference to `net80211_printf'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libphy.a(phy_chip_v7_ana.o):(.text.ram_wait_rfpll_cal_end+0x4): undefined reference to `phy_printf'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libphy.a(phy_chip_v7_ana.o): in function `ram_wait_rfpll_cal_end':
(.text.ram_wait_rfpll_cal_end+0x36): undefined reference to `phy_printf'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libphy.a(phy_chip_v7_ana.o): in function `set_channel_rfpll_freq':
(.text.set_channel_rfpll_freq+0x43): undefined reference to `phy_printf'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libphy.a(phy_chip_v7.o): in function `ram_gen_rx_gain_table':
(.text.ram_gen_rx_gain_table+0x100): undefined reference to `phy_printf'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: (.text.ram_gen_rx_gain_table+0x11a): undefined reference to `phy_printf'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libphy.a(phy_chip_v7.o):(.text.set_rx_gain_cal_iq+0x204): more undefined references presumably due to missing librarto `phy_printf' follow
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libpp.a(pp.o):(.wifi0iram.9+0x4): undefined reference to `coex_params'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a(ieee80211_hostap.o):(.text.hostap_input+0x0): undefined reference to `g_log_mod'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_set_wps_type_process+0x4): undefined reference to `g_misc_nvs'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_restore_process+0x0): undefined reference to `misc_nvs_restore'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_restore_process+0x7): undefined reference to `misc_nvs_restore'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a(ieee80211_scan.o):(.text.scan_op_start+0x0): undefined reference to `coex_params'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp_wifi/libesp_wifi.a(fast_crypto_ops.c.obj):(.rodata.g_wifi_default_wpa_crypto_funcs+0x8): undefined reference to `fast_aes_wrap'
[... cut ...]
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libwpa.a(wpa_auth.o):(.text.wpa_group_config_group_keys$isra$10+0x0): undefined reference to `wep'
/home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libwpa.a(wpa_auth.o):(.text.wpa_group_config_group_keys$isra$10+0x4): undefined reference to `tkip'
```

### Steps to repropduce

[Build the  project linked above.](https://github.com/PerMalmberg/idf-as-lib-with-wifi-link-issues)

renzbagaporo, 2019-06-03:
Hi @PerMalmberg,

Regarding the issue, the commit that fixes this has already been merged in our internal repo. It should be synced to the GitHub repo in a couple of days.

Just to give some context, as you may have observed with the changes we've been making to our build system, we're sort of relinquishing control of how ESP-IDF libraries are linked to external libraries/executables (whereareas the previous implementation required `idf_link_components`). This allows users to plainly link ESP-IDF component libraries to their own targets; with build specifications and usage requirements of its dependencies, and the dependencies of its dependencies, still propagating properly. 

This also means moving away from linking ESP-IDF libraries inside a link group. This previously hid flaws in dependency relationship between components (via COMPONENT_REQUIRES and COMPONENT_PRIV_REQUIRES). With this transition, we're expecting some 'undefined reference to xxx' link errors, so please bear with us :)

PerMalmberg, 2019-06-03:
Hi @renzbagaporo 

Glad to hear that it has been fixed.

There's no way to say this without sounding snarky, but please take it as constructive feedback: May I kindly ask that next time you ask for feedback on something you are a bit more quick to reply and to keep us posted on changes related to the feature in questions? I spend quite a lot of time on this, only to find out that you've already fixed it. Would it be possible to do more frequent syncs from your internal repo to an "experimental" public branch for situations like this?

renzbagaporo, 2019-06-04:
I understand @PerMalmberg. Apologies for my initial late response after the initial announcement. We usually push internal commits to Github it passes internal CI runs.  We're a little reluctant to push stuff that didn't pass at least the tests that we have...

PerMalmberg, 2019-06-06:
@renzbagaporo Understandable. Still, it'd reduce the cycle times quite substantially when testing things like this.

Just tried the latest master (826ff7186) and we're nearly back to a functioning build system:

> /home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: test/http_server_test/libhttp_server_test.a(http_server_test.cpp.obj):(.literal._ZSt9use_facetIKSt5ctypeIcEERKT_RKSt6locale[std::ctype<char> const& std::use_facet<std::ctype<char> const>(std::locale const&)]+0x4): undefined reference to `typeinfo for std::ctype<char>'
> /home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: test/http_server_test/libhttp_server_test.a(http_server_test.cpp.obj):(.literal._ZSt9use_facetIKSt5ctypeIcEERKT_RKSt6locale[std::ctype<char> const& std::use_facet<std::ctype<char> const>(std::locale const&)]+0x8): undefined reference to `typeinfo for std::locale::facet'
> /home/permal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: lib/smooth/libsmooth.a(Task.cpp.obj):(.rodata+0x228): undefined reference to `typeinfo for std::thread::_State'

Here's the last command: 

> && /home/permal/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address  -nostdlib test/main/CMakeFiles/smooth_test.dir/generated_test.cpp.obj  -o smooth_test  lib/smooth/libsmooth.a test/http_server_test/libhttp_server_test.a lib/smooth/libsmooth.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp32/libesp32.a esp-idf/freertos/libfreertos.a esp-idf/fatfs/libfatfs.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/sdmmc/libsdmmc.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/wear_levelling/libwear_levelling.a esp-idf/libsodium/liblibsodium.a esp-idf/lwip/liblwip.a esp-idf/json/libjson.a esp-idf/mbedtls/libmbedtls.a esp-idf/xtensa/libxtensa.a esp-idf/cxx/libcxx.a esp-idf/newlib/libnewlib.a esp-idf/freertos/libfreertos.a esp-idf/heap/libheap.a esp-idf/log/liblog.a esp-idf/soc/libsoc.a esp-idf/esp_rom/libesp_rom.a esp-idf/esp_common/libesp_common.a esp-idf/xtensa/libxtensa.a esp-idf/esp32/libesp32.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libcoexist.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libcore.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libespnow.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libmesh.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libphy.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libpp.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/librtc.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libsmartconfig.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libwpa2.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libwpa.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libwps.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/smartconfig_ack/libsmartconfig_ack.a esp-idf/app_update/libapp_update.a esp-idf/driver/libdriver.a esp-idf/esp_event/libesp_event.a esp-idf/efuse/libefuse.a esp-idf/pthread/libpthread.a esp-idf/app_trace/libapp_trace.a esp-idf/spi_flash/libspi_flash.a esp-idf/mbedtls/libmbedtls.a esp-idf/vfs/libvfs.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a esp-idf/lwip/liblwip.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/ethernet/libethernet.a esp-idf/cxx/libcxx.a esp-idf/newlib/libnewlib.a esp-idf/freertos/libfreertos.a esp-idf/heap/libheap.a esp-idf/log/liblog.a esp-idf/soc/libsoc.a esp-idf/esp_rom/libesp_rom.a esp-idf/esp_common/libesp_common.a esp-idf/xtensa/libxtensa.a esp-idf/esp32/libesp32.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libcoexist.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libcore.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libespnow.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libmesh.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libphy.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libpp.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/librtc.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libsmartconfig.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libwpa2.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libwpa.a /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32/libwps.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/smartconfig_ack/libsmartconfig_ack.a esp-idf/app_update/libapp_update.a esp-idf/driver/libdriver.a esp-idf/esp_event/libesp_event.a esp-idf/efuse/libefuse.a esp-idf/pthread/libpthread.a esp-idf/app_trace/libapp_trace.a esp-idf/spi_flash/libspi_flash.a esp-idf/mbedtls/libmbedtls.a esp-idf/vfs/libvfs.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a esp-idf/lwip/liblwip.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/ethernet/libethernet.a -lstdc++ -u __cxa_guard_dummy esp-idf/ne [... cut ...] me/permal/electronics/IO-Card-G3/software/externals/smooth/build/esp-idf/esp32 -T esp32_out.ld -u app_main -L /home/permal/esp/esp-idf/components/esp32/ld -T esp32.extram.bss.ld -T esp32.project.ld.in.ld -T esp32.peripherals.ld -u call_user_start_cpu0 -u ld_include_panic_highint_hdl -mfix-esp32-psram-cache-issue -u esp_app_desc -lgcov -lc -lm -lgcc -u vfs_include_syscalls_impl -L /home/permal/esp/esp-idf/components/esp_wifi/lib_esp32 /home/permal/esp/esp-idf/components/xtensa/esp32/libhal.a && :

igrr, 2019-06-06:
"Undefined reference to typeinfo for <libstdc++ type name>" is probably because libstdc++ is not built with RTTI support enabled. Please see https://github.com/espressif/esp-idf/issues/1684.

PerMalmberg, 2019-06-06:
@igrr So, previous versions of libstdc++ was built with RTTI enabled? I'm just trying to understand what part is causing the break as this all builds with gcc-8 and older IDFs (6fd535c98).

igrr, 2019-06-06:
No, previous toolchain versions also didn't have RTTI enabled in libstdc++.

My speculation is that this is related to linking order. Typeinfo for most of the libstdc++ types is also generated when compiling your application's files, since most of libstdc++ types are defined in headers included from the application. In the older master branch, which was using link groups, an undefined reference to typeinfo would occur somewhere in libstdc++.a. On the next pass over the link group, it would be resolved because corresponding typeinfo would be found in your application's object files.
Now that link groups are not used, and given that libstdc++ does not "depend" (in CMake sense) on the application object files, the linker is not instructed to link application object files after linking libstdc++. This results in an unresolved reference.

PerMalmberg, 2019-06-06:
@igrr Thanks for the detailed explanation.

As I'm not using RTTI I've added `target_compile_options(${target} PUBLIC $<$<COMPILE_LANGUAGE:CXX>: -fno-rtti>)`  to my library so it now links.

Now I just have to figure out how to generate the `project_description.json` required by idf.py.

renzbagaporo, 2019-06-06:
Generating the `project_description.json` is now under the project (`project.cmake`) and not part of the core build system. Note though that even in the previous ESP-IDF as a library implementation, using idf.py for custom CMake projects was not a supported use case (which is why `idf_as_lib` example always used a custom shell script to build the example).

PerMalmberg, 2019-06-06:
@renzbagaporo Appreciate quick responses.

It never was a supported use case? That's unexpected since I've always been running `~/esp/esp-idf/tools/idf.py -C .. -p /dev/ttyUSB1 app-flash monitor` without issues.

So the correct way is now to run `esp_too.py`and `idf_monitor.py` separately?

renzbagaporo, 2019-06-06:
The stance was that technically a custom CMake project is not an ESP-IDF app project, like the ones cut from https://github.com/espressif/esp-idf-template. It is a  project who just so happens to have dependencies on IDF component libraries, along with the user's own non-component libraries. That user might create their own equivalent `project_description.json` and wrap common actions in their custom `idf.py`. Since we don't have full control of how the project works, there is no guarantee that will work. I guess now this change with where `project_description.json` makes `idf.py` fully unusable for custom projects.

PerMalmberg, 2019-06-06:
@renzbagaporo Ok, that's fair.

As the original problem has been solved, and the RTTI issue is already tracked in #1684, I'm closing this issue. Thank you for the help!

closed 2019-06-06 by PerMalmberg