espressif/esp-idf #3574: Add support for SMB2 network access

sahlberg, 2019-06-02:
Libsmb2 is a userspace library to access SMB2 shares across a network (https://github.com/sahlberg/libsmb2).
This allows reading/writing files to a windows file share.

A user recently contributed support to build and use libsmb2 for the ESP32 successfully.

I think SMB2 support would be useful for many ESP32 users and suggest.
Maybe you can include libsmb2 in your SDK, or link to it from your SDK, thus making it easy
for ESP32 developer to gain SMB2 network file sharing capability to their projects.

Alvin1Zhang, 2019-06-03:
@sahlberg Thanks for reporting this, we will look into this. Thanks.

tore-espressif, 2021-09-22:
Hello @sahlberg 

It's been a while, since you posted this, but if you are still interested I'd love to help you to get your libsmb2 to our new Espressif's idf-component-manager service, 

Idf-component-manager is a new function that allows ESP-IDF users to integrate software package into their project seamlessly. Libsmb2 could be one of the first external components hosted there.

More information about idf-component manager can be found in [Espressif API guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html) or [PyPi registry](https://pypi.org/project/idf-component-manager/). The component service itself is hosted [here](https://components.espressif.com/).

Let me know if you agree and I can push a PR to libsmb2 repository, so the sw package is deployed after you release new version of libsmb2 (or we can setup a different trigger too)

sahlberg, 2021-09-22:
> Hello @sahlberg <https://github.com/sahlberg>
>
> It's been a while, since you posted this, but if you are still interested
> I'd love to help you to get your libsmb2 to our new Espressif's [...]

Please do.  Note that I do/can not build ESP-IF myself but I have a few
users that do build libsmb2 for their ESP projects and
they make sure that the library keep building.

I appreciate your help. Please add it to your component management service
and send me a PR for the things wou want me to check into the repository.

Thanks!

Ronnei Sahlberg

closed 2021-09-27 by sahlberg