espressif/esp-idf #3575: Crosstool-ng development workflow

c0d3z3r0, 2019-06-02:
This is meant as a request for crosstool-ng (https://github.com/espressif/crosstool-NG) but one can't create issues on forks... :/

Is there any plan to switch to "master-branch-based" development just like for esp-idf and others? The current branch situation is not very clear

Alvin1Zhang, 2019-06-03:
@c0d3z3r0 Thanks for reporting this, we will look into this. Thanks.

igrr, 2019-06-03:
@c0d3z3r0 We don't plan to switch to the master branch based workflow in the near future. Currently our workflow is determined by the facts that:
1) we are currently behind the upstream versions of crosstool-ng and the projects it depends on (gcc, binutils-gdb, newlib), and we want to update to more recent versions in the future
2) we would also like to see our patches committed on top of the upstream branch

If you wish to submit some patches, please let me know, and we can suggest the branch/commit to use as a base.

c0d3z3r0, 2019-06-03:
Thanks for your fast response!

Well, it seems I wasn‘t clear. I meant keeping the master branch as the modt recent one, where further development happens. Currently it‘s not really clear what those branches and tags are and what is the current one.

igrr, 2019-06-03:
Unfortunately we can't guarantee that the tip of crosstool-NG development branch produces a toolchain which works without issues with latest ESP-IDF. Before updating the toolchain version in IDF, we do the necessary testing to make sure that the new toolchain produces working code and any new warnings generated in IDF code are fixed.

To find the branch of crosstool-NG which is currently supported in ESP-IDF, please refer to the documentation page: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup-scratch.html#compile-the-toolchain-from-source.

Occasionally, we do preview releases of new toolchain versions, for example as in this forum post: https://www.esp32.com/viewtopic.php?t=7400. In such case, we indicate the branch of crosstool-NG required to build this toolchain.

c0d3z3r0, 2019-06-03:
Hm, a master branch IMHO is always WIP/unstable and thus not guaranteed to work, isn‘t it? That‘s what the documentation says, too, about the IDF master branch.

That‘s what a stable/release branch is for.

igrr, 2019-06-03:
Okay, I understand your point. While we probably won't use the "master" branch for the reasons indicated in my earlier reply, we will look into updating the "default" branch shown on Github to whatever the current development branch is.

Please note however that while the combination of (the preferred crosstool-NG toolchain version) and (IDF master branch) is not guaranteed to be bug-free, it goes through some QA before being synchronized to Github: all examples should compile without warnings, all unit and automated integration tests should pass, all examples which have automated tests should also work. The combination of (tip of the crosstool-NG development branch) and (IDF master branch) does not receive such QA, so it is quite probably broken, most of the time. Which means that taking the latest crosstool-NG toolchain version is not necessarily as practical as taking the latest ESP-IDF master branch, but with a supported toolchain.

c0d3z3r0, 2019-06-03:
> Okay, I understand your point. While we probably won't use the "master" branch for the reasons indicated in my earlier reply, we will look into updating the "default" branch shown on Github to whatever the current development branch is.

Great, thank you :-) Maybe an alternative to this would be a branch called e.g. "develop"
One more question on this: what actually is the current dev branch?

> Please note however that while the combination of (the preferred crosstool-NG toolchain version) and (IDF master branch) is not guaranteed to be bug-free, it goes through some QA before being [...]

Thank you very much for sharing those valuable insight to your process! This helps a lot.

igrr, 2019-06-03:
> One more question on this: what actually is the current dev branch?

It's called `esp_develop`, and it seems that our scripts do not sync it to Github at the moment. Will update this issue when that is fixed.

c0d3z3r0, 2019-06-06:
@igrr any news here?

igrr, 2019-06-06:
No, sorry, haven't gotten to this yet. Are you blocked on this?

c0d3z3r0, 2019-06-09:
Can I assist you anyhow? Fixing the script maybe?

antmak, 2019-08-13:
Finally done :) @c0d3z3r0 Please check if you may  close this

Alvin1Zhang, 2019-10-11:
@c0d3z3r0 Thanks for reporting, would you please help check whether your issue is resolved? Could this ticket be closed? Thanks.

closed 2019-10-12 by c0d3z3r0