espressif/esp-idf #3579: When subscribing to mqtt broker's data through `esp_mqtt_client_subscribe`, if the data is large (65KB), it will accept failure.

GitOfLance, 2019-06-03:
### Environment
- Module or chip used: ESP32-WROOM-32
- IDF version : v4.0-dev-207-g76da10a66
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

### Problem Description
Usually, I can get mqtt broker data through `esp_mqtt_client_subscribe`, but once I add a lot of data to this topic, sometimes I can't trigger `MQTT_EVENT_DATA`.
- Does esp32 produce `MQTT_EVENT_DATA` with data size limitations? 
- Why can't larger data acceptance trigger this event?

david-cermak, 2019-06-05:
Hi @GitOfLance 

Can you please test with latest master? There's been an update to mqtt reception in https://github.com/espressif/esp-idf/commit/db121a56e13b4b93b6bcc9217f7890bc4783cabd so this might have fixed the issue.

No there should be no data size limitation (other than MQTT protocol itself), just that the data will be received in multiple data events if total data size > configured buffer size

GitOfLance, 2019-06-06:
Hi @david-cermak 
Ok, thank you very much, I have pulled the latest master, and I will give feedback here after the test.

Alvin1Zhang, 2019-10-11:
@GitOfLance Thanks for reporting. Any feedback that could be shared? Thanks.

Alvin1Zhang, 2020-01-16:
@GitOfLance Thanks for reporting. Sorry for the slow turnaround. Would you please help share if any updates for the issue? Has your issue been resolved? Thanks.

GitOfLance, 2020-01-16:
Hi， @david-cermak @Alvin1Zhang 
I updated to the latest master and this problem did not occur.

In order to ensure the security of big data transmission and reception. On the one hand, I have increased the send and receive buffers of mqtt, and on the other hand, I have subcontracted the big data.
Thank you for your suggestions and support.

closed 2020-01-16 by GitOfLance