espressif/esp-idf #3581: ESP MQTT Client always connects via STA IF even if STA is not default IF

no1seman, 2019-06-03:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-667-gda13efc17
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB|external 5V&Battery

## Problem Description
When 2 interfaces is up: ETH and STA MQTT Client always connects to server via STA interface even if default IF is ETH. 
### Expected Behavior
MQTT Client should use default IF for connection.
### Actual Behavior
MQTT Client always connects to server via STA interface if both ETH and STA is up and ETH is set to default IF.

### Steps to repropduce

1. Establish ETH and STA IFs
2. Set ETH as default IF
3. Establish MQTT connection
4. netstat -nt | grep 1883 on MQTT server server host to see that STA ip is used for connection

### Code to reproduce this issue

code to set default IF:
esp_err_t setDefaultIf(tcpip_adapter_if_t tcpip_if)
{
    struct netif *_netif;
    esp_err_t ret = tcpip_adapter_get_netif(tcpip_if, (void **)&_netif);
    if (ret != ESP_OK)
    {
        return ret;
    }
    netif_set_default(_netif);
    return ESP_OK;
}

.....
setDefaultIf(TCPIP_ADAPTER_IF_ETH);
....
esp_mqtt_client_start(client);
......

## Debug Logs

## Other items if possible

david-cermak, 2019-06-10:
Hi @no1seman 

The issue is probably that both your interfaces route to the server. Setting the default interface in LwIP means that the netif is "used to output all packets for which no specific route is found", 
so it applies only to scenarios where routing could not be determined. Please note the implementation of [`ip4_route` here](https://github.com/espressif/esp-lwip/blob/fe9a58cb7abdc74e7a2196b7c4e0a29ff9042f88/src/core/ipv4/ip4.c#L210), `default_netif` is used only if no match in IP 
could be found in the first section.

no1seman, 2019-06-10:
@david-cermak,

yes, you are right, both interfaces in the same subnet:
eth: IP: 192.168.3.222/24, GW: 192.168.3.1
sta: IP: 192.168.3.130/24, GW: 192.168.3.1
mqtt server: IP: 192.168.3.211

So, if this not an issue in that case, how can I specify interface (IP) mqtt client to connect to server? The probem is that sometimes eth connects first, and sometimes STA first, but mqtt always connects to server with STA interface.

david-cermak, 2019-06-10:
AFAIK it's not so easy to work with LwIP's `netif`s when on socket interface (which MQTT uses), especially when the interface is duplicated. Might not be a solution to split to two subnets perhaps?

Some ideas:
* LwIP provides route hooks, so one can redefine a macro (such as `LWIP_HOOK_IP4_ROUTE`) to return valid netif based on provided destination (or possibly source address). These probably might not work as (again) the routing is clearly valid for both duplicated netifs.
* using a raw API can bind directly to a specific netif, but that means a low level access to network (even below socket interface used by mqtt library)
* a dirty hack could be a modification of `ip4_route` to return default_netif (if up and not null)
```
ip4_route(const ip4_addr_t *dest)
{
  if ((netif_default != NULL) && netif_is_up(netif_default))  return netif_default;
... // continue...
...
```
* another not a clean solution could be to employ `MQTT_EVENT_BEFORE_CONNECT` event to bring the STA interface down with `tcpip_adapter_down` and bring it up again upon connection.

With that said, I do not think there really is a good solution (other then possibly using two separate subnets for these netifs)

no1seman, 2019-08-08:
@david-cermak,
thank you for detailed answer, first variant works fine, but for now need to modify esp-idf lwip component. If there any example of howto redefine LWIP_HOOK_IP4_ROUTE (and any other lwip settings: lwipopts.h/opts.h) directly from projects code and leave esp-idf framework code clean?

david-cermak, 2019-08-08:
Thanks for taking the investigation (wasn't sure if that would work) and sharing the results!

It is possible to add compile definitions to your project to redefine LWIP_HOOK_FILENAME, that is an extra header file included by lwip core sources. In that header declare a prototype of custom routing function and define `LWIP_HOOK_IP4_ROUTE` to the name of that function.

Project makefile:
```
CFLAGS += "-DLWIP_HOOK_FILENAME=\"my_ip4_route.h\""
```

my_ip4_route.h (somewhere in the project dirs (might need to add -I${PRJ_DIR} to compile definitions too):
```
#define LWIP_HOOK_IP4_ROUTE my_route
extern struct netif *my_route(const ip4_addr_t *dest);
```

define of `my_route()` anywhere in the project dir

no1seman, 2019-08-09:
@david-cermak,
thank you for answer, but i'm using cmake and still can't understand howto solve my problem without changing LwIP source code.

Let me explain the problem:
In my project I've got 2 different connections: WiFi STA & Ethernet. If Ethernet is connected it must be the main interface that esp services must use, if not then they have to use WiFi STA. So, in my case Ethernet (if up) is main interface and WiFi is only standby. When esp starts any of WiFi or Ethernet interfaces may get connection and IP first. In my tests Ethernet gets fully connected (IF up, link up & got IP from DHCP) first only in 5% of runs. Sousually WiFi got ready faster than Ethernet. When WiFi is connected application doesn't know if Ethernet will get up and when it will be (so delay is not a solution). Also during long run WiFi and Ethernet may disconnect for many-many different purposes.

So I need to realise some universal logic:
if (WiFi STA IF is up & Ethernet IF is up) all outcoming connections from esp services must use Ethernet IF;
if (WiFi is up & Ethernet is down) all outcoming connections from esp services must use WiFi IF;
if (WiFi is down & Ethernet is up) all outcoming connections from esp services must use Ethernet IF;
if (WiFi is down & Ethernet is down) nothing to do;  just smoke :)

To realise that logic i've made:
1. Function to set default IF:
```
esp_err_t setDefaultIf(tcpip_adapter_if_t tcpip_if)
{
    struct netif *_netif;
    esp_err_t ret = tcpip_adapter_get_netif(tcpip_if, (void **)&_netif);
    if (ret != ESP_OK)
    {
        ESP_LOGE(TAG, "Error setting default netif: %s\n", esp_err_to_name(ret));
        return ret;
    }
    char ip_str[IP_ADDR_STR_LEN];
    netif_set_default(_netif);
    return ESP_OK;
}
```
2. Ethernet got IP event hook:
```
static void on_ethernet_got_ip(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    esp_err_t ret;
    xEventGroupSetBits(app_state, ETHERNET_GOT_IP_BIT);
    ip_event_got_ip_t *event = (ip_event_got_ip_t *)event_data;
    if (event->ip_changed)
    {
        ESP_ERROR_CHECK(tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_ETH, _HOST_NAME));
        ret = setDefaultIf(TCPIP_ADAPTER_IF_ETH);
        if (ret != ESP_OK)
            ESP_LOGE(TAG, "Set Ethernet as default IF error: %s", esp_err_to_name(ret));
       //starting/restarting services if needed
      ...
    }
} 
```
3. Function to check if Ethernet got IP:
```
bool is_ethernet_got_ip(void)
{
    return xEventGroupWaitBits(app_state, ETHERNET_GOT_IP_BIT, false, true, 0) & ETHERNET_GOT_IP_BIT;
}
```
4. WiFi STA got IP event hook:
```
static void on_wifi_sta_got_ip(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
    esp_err_t ret;
    if (event->ip_changed)
    {
        xEventGroupSetBits(app_state, WIFI_STA_GOT_IP_BIT);
        if (!is_ethernet_got_ip())
        {
            ret = setDefaultIf(TCPIP_ADAPTER_IF_STA);
            if (ret != ESP_OK) 
                ESP_LOGE(TAG, "Set WiFi STA as default IF error: %s", esp_err_to_name(ret));
            //starting/restarting services if needed
           ...
        }
    }
}
5. Modified ip4_route_src_hook() in LwIP component:
```
struct netif * 
ip4_route_src_hook(const ip4_addr_t *src,const ip4_addr_t *dest)
{
  if (netif_default != NULL && !ip4_addr_isany((ip4_addr_t *)&netif_default->ip_addr)) {
     if (netif_is_up(netif_default) && netif_is_link_up(netif_default) && !ip4_addr_isany_val(*netif_ip4_addr(netif_default))) {
           return netif_default;
     }
  }

  struct netif *netif = NULL;

  /* destination IP is broadcast IP? */
  if ((src != NULL) && !ip4_addr_isany(src)) {
    /* iterate through netifs */
    for (netif = netif_list; netif != NULL; netif = netif->next) {
      /* is the netif up, does it have a link and a valid address? */
      if (netif_is_up(netif) && netif_is_link_up(netif) && !ip4_addr_isany_val(*netif_ip4_addr(netif))) {
        /* source IP matches? */
        if (ip4_addr_cmp(src, netif_ip4_addr(netif))) {
          /* return netif on which to forward IP packet */
          return netif;
        }
      }
    }
  }
  return netif;
}
```
For now everything works fine:
1. WiFi STA& Ethernet answers for ICMP requests from network
2. HTTP server avaible on both WiFi STA & Ethernet
3. SNTP & DHCP also works on both interfaces
4. MQTT client, https client connects from Default IF
....
but, I have several projects and only one of it needs such logic. So, I can't understand how to change LWIP_HOOK_IP4_ROUTE_SRC (or LWIP_HOOK_IP4_ROUTE) from project and leave LwIP component clean using cmake?

david-cermak, 2019-08-09:
The same principle applies for CMake build system as well. No need to modify the framework (no problem to use the same IDF for other projects without specific routing hooks).

Let me sketch up a simple structure of IDF project with a main component:
```
[project_dir] : CMakeLists.txt
              + [main] : CMakeLists.txt
                       + [lwip_local] + [include] : ip4_route_local.h
                                      + [src]     : ip4_route_local.c
```
1) project CMakeLists.txt (add after `project()`):
```
...
project(...project_name...)

idf_component_get_property(lwip lwip COMPONENT_LIB)
target_compile_definitions(${lwip} PRIVATE "-DLWIP_HOOK_FILENAME=\"ip4_route_local.h\"")
target_compile_options(${lwip} PRIVATE "-I${PROJECT_DIR}/main/lwip_local/include")
```
2) main CMakeLists.txt (just add ip4_route_local.c to sources):
```
idf_component_register(SRCS .... "lwip_local/src/ip4_route_local.c" ...)
```
3) ip4_route_local.h:
```
#undef LWIP_HOOK_IP4_ROUTE_SRC
#define LWIP_HOOK_IP4_ROUTE_SRC ip4_route_src_hook_local

struct netif * 
ip4_route_src_hook_local(const ip4_addr_t *src,const ip4_addr_t *dest);
```
4) ip4_route_local.c: just paste the modified definition of `ip4_route_src_hook()`
```
struct netif * 
ip4_route_src_hook(const ip4_addr_t *src,const ip4_addr_t *dest)
{
    ...
}
```

Note: This should work For IDF 4.0, please let me know if you're on a different release.

no1seman, 2019-08-09:
@david-cermak,
Great thanks for the detailed answer. For now solution is cleaner for understanding.
I've made changes to porject fully according to your instructions, everything compile successfully, but linking falling to the following error:

/home/user/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(ip4.c.obj):(.literal.ip4_route+0x4): undefined reference to `ip4_route_src_hook_local'
/home/user/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(ip4.c.obj): in function `ip4_route':
/home/user/esp/esp-mdf/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:255: undefined reference to `ip4_route_src_hook_local'
/home/user/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/lwip/liblwip.a(ip4.c.obj): in function `ip4_route_src':
/home/user/esp/esp-mdf/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:184: undefined reference to `ip4_route_src_hook_local'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

I'm using IDF version: v4.0-dev-1443-g39f090a4f, Build System: CMake, Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0

Here is my files:
[{PWD}/CMakeLists.txt]:
```
cmake_minimum_required(VERSION 3.5)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(beacon)
idf_component_get_property(lwip lwip COMPONENT_LIB)
target_compile_definitions(${lwip} PRIVATE "-DLWIP_HOOK_FILENAME=\"ip4_route_local.h\"")
target_compile_options(${lwip} PRIVATE "-I${PROJECT_DIR}/main/lwip_local/include")
```
[{PWD}/main/CMakeList.txt]:
```
idf_component_register(SRCS "app.c"
                              ...
                            "lwip_local/src/ip4_route_local.c"
                        INCLUDE_DIRS ".")
```
[{PWD}/main/lwip_local/include/ip4_route_local.h]:
```
#include "lwip/netif.h"
#undef LWIP_HOOK_IP4_ROUTE_SRC
#define LWIP_HOOK_IP4_ROUTE_SRC ip4_route_src_hook_local
struct netif *ip4_route_src_hook_local(const ip4_addr_t *src, const ip4_addr_t *dest);
```
[{PWD}/main/lwip_local/scr/ip4_route_local.c]:
```
#include "lwip/netif.h"
struct netif *ip4_route_src_hook_local(const ip4_addr_t *src, const ip4_addr_t *dest)
{
...
}
```
PS I'm sorry to bother you!

david-cermak, 2019-08-09:
Not sure what's wrong here (can reproduce the issue but not sure why the linker could not resolve this symbol). As a workaround you can for now place the definition of `ip4_route_src_hook_local()` to the app.c or some other source where `app_main()` is defined.

Will get back to you once a better solution found...

no1seman, 2019-08-10:
@david-cermak,
much thanks for idea, funally I've got it!

According to esp32.com forum many developers asking of howto change esp-idf components logic, especially LwIP. Seems that that technique must be documented on docs.espressif.com because it's not trivial.

So, here is full working exmple:
1. [{PWD}/CMakeLists.txt]:
```
cmake_minimum_required(VERSION 3.5)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(beacon)
idf_component_get_property(lwip lwip COMPONENT_LIB)
target_compile_definitions(${lwip} PRIVATE "-DLWIP_HOOK_FILENAME=\"ip4_route_local.h\"")
target_compile_options(${lwip} PRIVATE "-I${PROJECT_DIR}/main/lwip_local/include")
```
2. [{PWD}/main/CMakeList.txt]:
```
idf_component_register( SRCS    "app.c"
                                "baseutils.c"
                                "ble.c"
                                "ble_gatt.c"
                                "blink.c"
                                "bonjour.c"
                                "configure.c"
                                "ethernet.c"
                                "filesystem.c"
                                "gpio_ext.c"
                                "http.c"
                                "http_json.c"
                                "logging.c"
                                "main.c"
                                "mqtt.c"
                                "networks.c" <- ip4_route_src_hook_local here!!!
                                "ntp.c"
                                "nvs.c"
                                "power.c"
                                "ring_buffer.c"
                                "settings.c"
                                "setup.c"
                                "ssh.c"
                                "telnet.c"
                                "trex.c"
                                "utils.c"
                                "validators.c"
                                "wifi_ap.c"
                                "wifi_sta.c"
                        INCLUDE_DIRS    ".")
```
3. [{PWD}/main/lwip_local/include/ip4_route_local.h]:
```
#include "lwip/netif.h"
#undef LWIP_HOOK_IP4_ROUTE_SRC
#define LWIP_HOOK_IP4_ROUTE_SRC ip4_route_src_hook_local
struct netif *ip4_route_src_hook_local(const ip4_addr_t *src, const ip4_addr_t *dest);
```
4. [{PWD}/main/networs.c]:
```
...
#include "lwip/netif.h"
struct netif *ip4_route_src_hook_local(const ip4_addr_t *src, const ip4_addr_t *dest)
{
...// Here is custon realisation of ip4_route_src_hook logic
}
...
```

The difference from previous is that networks.c has its own header networks.h file with a number of fucntions and variables declaration that uses by other files in main dir.

PS Many-many-many-many thanks for yor help. Closing this issue.

closed 2019-08-10 by no1seman

david-cermak, 2019-08-10:
Yes, you're right. Reason for the failure is wrong link dependency.

Another option to avoid it is to add `lwip_local` sources directly to lwip project. 
I've updated the previous [post](https://github.com/espressif/esp-idf/issues/3581#issuecomment-519940878) to include ip4_route_local.c to sources of lwip component rather then main component

PS: No worries, might be helpful for others too.