espressif/esp-idf #3582: Background feature of esp_ble_gattc_open

copercini, 2019-06-03:
Looking the [docs of esp_ble_gattc_open](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gattc.html#_CPPv418esp_ble_gattc_open13esp_gatt_if_t13esp_bd_addr_t19esp_ble_addr_type_tb) there is a flag `bool is_direct` to background auto connection, when I know the mac address of the BLE device I want to connect, could I call this function and let it searching in background and when the device enters the range, auto-connect?
or in which case can I use it `is_direct` as false?

Just trying `esp_ble_gattc_open(g_gattc_if, my_mac, BLE_ADDR_TYPE_RANDOM, false);` this doesn't even try to connect even when the device is near, while `esp_ble_gattc_open(g_gattc_if, mac, BLE_ADDR_TYPE_RANDOM, true);` always works, but only the device is in the range

Weijian-Espressif, 2019-06-04:
@copercini, Sorry, IDF  doesn't  support background auto connection.

closed 2019-06-11 by Alvin1Zhang

0neblock, 2019-07-21:
Hi, is this functionality planned in a future update to esp-idf ?