espressif/esp-idf #3583: In WIFI_MODE_APSTA mode if router for STA mode drops connections all connections for AP mode will drop too

starlino, 2019-06-04:


Alvin1Zhang, 2019-06-04:
@starlino Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

negativekelvin, 2019-06-04:
How often do you call connect function to try to reconnect?

starlino, 2019-06-04:
To reproduce on Windows 10.
Create a wireless hotspot with ST / st123456 credentials
Run the program below (I am using arduino core with IDF release  3.2, I am not using an Arduino native WiFi , only native esp32-idf functions)
Notice that esp32 connects to the hotspot
Also  from windows computer connect to AP  /  ap123456 and start pining   192.168.4.1  with
ping 192.168.4.1  -t  -w 1000
Now  on windows turn off the wireless hotspot (the ST interface).
the ping window will display **Request time out.**

`Reply from 192.168.4.1: bytes=32 time=1ms TTL=255
Reply from 192.168.4.1: bytes=32 time=1ms TTL=255
Reply from 192.168.4.1: bytes=32 time=293ms TTL=255
Reply from 192.168.4.1: bytes=32 time=1ms TTL=255
Request timed out.
Request timed out.`

Somehow when the hub on ST interface goes offline it affects the AP interface. This does not always happens . This morning I disabled the IPv6 interface on windows and I was not able to reproduce this anymore.

For now I would like to disable IPv6 interface from esp32. I noticed that sometimes I am not given the  192.168.4.1 address by the AP interface but a  IPv6 address. I am not sure if this is related to my issue , but to isolate the problem I would also like to disable the IPv6 from the esp32 side (since it might be impractical to ask users to disable IPv6 on their side) . Does anybody knows how to disable IPv6 on the esp32 side so it is always forced to use IPv4 and gives me the 192.168.4.1 gateway address ?

`#define STA_SSID "ST"
#define STA_PASS "st123456"
#define AP_SSID  "AP"
#define AP_PASS  "ap123456"
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "lwip/err.h"
#include "lwip/sys.h"

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t s_wifi_event_group;

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_AP_STACONNECTED:
       log_i( "station:"MACSTR" join, AID=%d",
                 MAC2STR(event->event_info.sta_connected.mac),
                 event->event_info.sta_connected.aid);
        break;
    case SYSTEM_EVENT_AP_STADISCONNECTED:
       log_i( "station:"MACSTR"leave, AID=%d",
                 MAC2STR(event->event_info.sta_disconnected.mac),
                 event->event_info.sta_disconnected.aid);
        break;
    default:
        break;
    }
    return ESP_OK;
}

wifi_config_t wifi_config_ap;
wifi_config_t wifi_config_st;

void wifi_ap_setup(){
  log_i("");
  strcpy ((char*)wifi_config_ap.ap.ssid,AP_SSID);
  wifi_config_ap.ap.ssid_len = strlen(AP_SSID); 
  strcpy ((char*)wifi_config_ap.ap.password,AP_PASS);
  wifi_config_ap.ap.max_connection = 10;
  wifi_config_ap.ap.authmode = WIFI_AUTH_WPA_WPA2_PSK;
  ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config_ap));
}

void wifi_st_setup(){
  log_i("");
  strcpy ((char*)wifi_config_st.sta.ssid,STA_SSID);
  strcpy ((char*)wifi_config_st.sta.password,STA_PASS);
  ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config_st));
}

void wifi_init()
{
    log_i("");
    s_wifi_event_group = xEventGroupCreate();

    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
    wifi_ap_setup();
    wifi_st_setup();
  
    
    ESP_ERROR_CHECK(esp_wifi_start());

    esp_wifi_connect();
   
    log_i("done");
           
}

void setup() {
  Serial.begin(115200);
  // put your setup code here, to run once:
  wifi_init();
}

void loop(){};
`

starlino, 2019-06-04:
> 
> 
> How often do you call connect function to try to reconnect?

This usually happens after 2-3rd attempt to reconnect.

Below is slightly modified code that attempts automatic re-connect of STA interface in case of router dropped connection (10 seconds interval).

```

#define STA_SSID "ST"
#define STA_PASS "st123456"
#define AP_SSID  "AP"
#define AP_PASS  "ap123456"

#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_event_legacy.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t wifi_event_group;

const static int WIFI_ST_CONNECTED_BIT = BIT0;
const static int WIFI_ST_CONNECTING_BIT = BIT1;

static esp_err_t wifi_event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
        case SYSTEM_EVENT_STA_START:
            log_i("SYSTEM_EVENT_STA_START");
            wifi_st_up();
            break;
        case SYSTEM_EVENT_STA_GOT_IP:
            xEventGroupSetBits(wifi_event_group, WIFI_ST_CONNECTED_BIT);
            xEventGroupClearBits(wifi_event_group, WIFI_ST_CONNECTING_BIT);
            
            ESP_LOGD(TAG, "SYSTEM_EVENT_STA_GOT_IP, ip:" IPSTR ", mask:" IPSTR ", gw:" IPSTR,
            IP2STR(&event->event_info.got_ip.ip_info.ip),
            IP2STR(&event->event_info.got_ip.ip_info.netmask),
            IP2STR(&event->event_info.got_ip.ip_info.gw));            
            
            break;
        case SYSTEM_EVENT_STA_DISCONNECTED:
            log_i("SYSTEM_EVENT_STA_DISCONNECTED");
            xEventGroupClearBits(wifi_event_group, WIFI_ST_CONNECTED_BIT | WIFI_ST_CONNECTING_BIT );
            break;
      
        case SYSTEM_EVENT_AP_STACONNECTED:
           log_i( "station:"MACSTR" join, AID=%d",
                     MAC2STR(event->event_info.sta_connected.mac),
                     event->event_info.sta_connected.aid);
            break;
        case SYSTEM_EVENT_AP_STADISCONNECTED:
           log_i( "station:"MACSTR"leave, AID=%d",
                     MAC2STR(event->event_info.sta_disconnected.mac),
                     event->event_info.sta_disconnected.aid);
            break;
        default:
            break;
    }
    return ESP_OK;
}

wifi_config_t wifi_config_ap;
wifi_config_t wifi_config_st;

void wifi_ap_setup(){
  log_i("");
  strcpy ((char*)wifi_config_ap.ap.ssid,AP_SSID);
  wifi_config_ap.ap.ssid_len = strlen(AP_SSID); 
  strcpy ((char*)wifi_config_ap.ap.password,AP_PASS);
  wifi_config_ap.ap.max_connection = 10;
  wifi_config_ap.ap.authmode = WIFI_AUTH_WPA_WPA2_PSK;
  
  ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config_ap));

   
  
}

void wifi_st_setup(){
  log_i("");
  strcpy ((char*)wifi_config_st.sta.ssid,STA_SSID);
  strcpy ((char*)wifi_config_st.sta.password,STA_PASS);
  ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config_st));
}

void wifi_init()
{
    log_i("");
    wifi_event_group = xEventGroupCreate();

    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(wifi_event_handler, NULL));

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));

    wifi_ap_setup();
    wifi_st_setup();
   

    
    
    ESP_ERROR_CHECK(esp_wifi_start());

    //wifi_st_up();
   log_i("done");
           
}

#include <WiFi.h>

// Set web server port number to 80
WiFiServer server(80);

void setup() {
  Serial.begin(115200);
  // put your setup code here, to run once:
  wifi_init();
   server.begin();
}

void wifi_st_up(){
    log_i("");
    xEventGroupSetBits(wifi_event_group, WIFI_ST_CONNECTING_BIT);
    esp_wifi_connect();
}

void wifi_st_down(){
     log_i("");
       esp_wifi_disconnect();
}

uint32_t reconnect_millis = 0;
void loop(){
    while(Serial.available()){
      char c = Serial.read();
      if('S' == c) wifi_st_up();
      if('s' == c) wifi_st_down();

    }

    uint8_t event_group_bits = xEventGroupGetBits(wifi_event_group);
    if( !(event_group_bits & WIFI_ST_CONNECTED_BIT) && !(event_group_bits & WIFI_ST_CONNECTING_BIT)){
      if(! reconnect_millis){
        log_i("reconnect_millis set");
        reconnect_millis = millis() | 1;  //avoid 0 value
      }else if( millis() - reconnect_millis > 10*1000U){  //uint32_t oveflow ok
        reconnect_millis = 0;
        wifi_st_up();  //will set WIFI_ST_CONNECTING_BIT
      }
    }else if(reconnect_millis){
      log_i("reconnect_millis clear");
      reconnect_millis = 0;  //clear timer
    }
    
    
}
```

liuzfesp, 2019-07-14:
Hi @starlino how many consecutive "Request timed out." is printed when the problem happens? Is the ping recoverable when ESP32 connects to the hotspot again, or when ESP32 STA stops to connect the AP? If yes, it's not a bug, it's expected behavior. The AP will be temporarily impacted when the ESP32 STA is connecting or scanning. 

When ESP32 STA is connecting/scanning status,  the ESP32 may change to other channel and thus impacts the ESP32 AP's **receiving**. 

Let consider a very common scenario:
- ESP32 is configured to WIFI_MODE_APSTA
- ESP32 STA is connected to external AP, the channel of the external is 6
- A PC is connected to ESP32 AP, the channel is 6 since the external AP is in channel 6
- Ping from PC to ESP32 AP
- Power-off the external AP and the ESP32 STA loses WiFi connection. 
- ESP32 STA reconnects the AP. The 1st step of reconnection is to find the extern AP. It search the channel 6 firstly and fails to find the AP in channel 6 since the external AP is power-off. Then ESP32 STA tries to find (scan) the AP in other channels. For each channel different than channel 6, ESP32 STA dwells on the channel for some time (default 120ms), then go back to channel 6 to receive/transmit  data for 30ms. **However, if the PC pings the ESP32 AP when the ESP32 STA is not in channel 6, then the ESP32 AP fails to receive the packets. **

starlino, 2019-07-15:
@liuzfesp , so basically this happens because ESP32  STA  & ESP32 AP must be on same channel ?? Furthermore  this channel number is dictated by external AP ??  Would it be even hardware-possible to have AP and STA channels separate ?

liuzfesp, 2019-07-24:
So basically this happens because ESP32 STA & ESP32 AP must be on same channel ?? 
Yes.

Furthermore this channel number is dictated by external AP ?? 
Yes.

Would it be even hardware-possible to have AP and STA channels separate ?
It's not possible.

Alvin1Zhang, 2019-07-25:
@starlino Thanks for reporting the issue, feel free to reopen the ticket if your issue still exists. Thanks.

closed 2019-07-25 by Alvin1Zhang