espressif/esp-idf #3584: [ble mesh] static oob issue

zhanlej, 2019-06-04:
Hi,
I want to implement a function that only allows a specific provisioner to add devices to the ble mesh network. For example, the ble mesh device we developed can only be provisioned through the APP we developed, and other APPs cannot be provisioned.

As far as I know, static oob should be able to implement this function. For example, the device and the provisioner must set the same static oob value, so that two-way authentication can be achieved. This is just my guess, I don't know if it is correct.

So I made the following changes in the ble_mesh_node demo:
```cpp
/* Disable OOB security for SILabs Android app */
static esp_ble_mesh_prov_t provision = {
    .uuid = dev_uuid,
#if 0
    .output_size = 4,
    .output_actions = ESP_BLE_MESH_DISPLAY_NUMBER,
    .input_actions = ESP_BLE_MESH_PUSH,
    .input_size = 4,
#else
    // .output_size = 0,
    // .output_actions = 0,
#endif
    // .oob_pub_key = 1,
    .static_val = {0x01,0x02,0x03,0x04},
    .static_val_len = 4,
};
```
After compiling and flashing, I found that using nRF mesh can still directly provision the device into the Mesh network. The phenomenon of using ble_mesh_provisioner demo provision device is also the same. It feels that static oob seems to be useless. Is this the problem of my operation or your SDK? Haven't implemented static oob yet?

Campou, 2019-06-04:
Hi @zhanlej,
Thanks for the question. I had a try with the nRF Mesh App and your test code, it turns out to be the same. On the App side, it have discovered the device Static OOB ability, but the App chooses No OOB in its Provisioning Start PDU, and here is the log:
```
W (54838) BLE_MESH: Algorithm:   0x00
W (54838) BLE_MESH: Public Key:  0x00
W (54838) BLE_MESH: Auth Method: 0x00
W (54838) BLE_MESH: Auth Action: 0x00
W (54838) BLE_MESH: Auth Size:   0x00
```
So better to find some App which supports Static OOB.
Thanks.

zhanlej, 2019-06-05:
Hi @Campou 
I think it's very strange. According to my understanding, my code should set the device to only support Static OOB. If the APP does not support Static OOB, the provisioning should fail, but the result is successful.

Campou, 2019-06-05:
Hi @zhanlej,

The authentication method should be decided  by the Provisioner side in my opinion. Even if the device supports the Static OOB, Output OOB or Input OOB, but the Provisioner only supports No OOB, then i think No OOB will be used for provisioning.

Also the Provisioning Capbilities PDU only indicates that wthether the device supports Static/Output/Input OOB, no indication of whether No OOB is supported or not.

Thanks.

closed 2019-06-17 by zhanlej

ankit-thealchemist, 2020-05-15:
> Hi @zhanlej,
>
> The authentication method should be decided by the Provisioner side in my opinion. Even if the device supports the Static OOB, Output OOB or Input OOB, but the Provisioner only [...]

I think this interpretation is wrong. suppose I have some unprovisioned node, then anyone can download the nrf mesh app and provisioned my nodes by using the no oob method.
Also check this video https://www.youtube.com/watch?v=XthbU9NP0Yg, in this example the guy uses the output OOB method.

florian-ats, 2021-01-08:
> I think this interpretation is wrong. suppose I have some unprovisioned node, then anyone can download the nrf mesh app and provisioned my nodes by using the no oob method.

I agree.

I can confirm that when using static key on an ESP32 node (same example as above), I can provision it with the nRF Mesh app if I select "no OOB" during the provisioning process.

Do I miss something that should block the node provisioning by a provisioner without static key?

ankit-thealchemist, 2021-01-09:
@florian-ats please follow this for more updates...   https://github.com/espressif/esp-idf/issues/5281